olikraus/u8g2 #1110: Chinese characters get overlapped

wayne2000, 2020-02-15:
Hi Olikraus
Firstly thank you for nice lib. 
I am using u8g2 to display Chinese characters in SSD1306, and verything is fine with wqy fonts/BDFs. But I got overlap issue when switching to the new font SimFang. The ttf can be found in windows/fonts, and I generated BDF using Fontforge. With the same font SimFang, ASCII display is good.
Could you kindly let me know what mistakes could be?

wayne2000, 2020-02-15:
I may found some clues. I used "bdfconv -b 0" when generating glyph bits array, which works fine for WQY fonts but not good for SimFang fonts. Have to use "bdfconv -b 2" for SimFang. Do you know why?

Joennuh, 2020-02-15:
I think because of the half width / full width that Chinese characters use: https://en.wikipedia.org/wiki/Halfwidth_and_fullwidth_forms

olikraus, 2020-02-15:
It should be "-b 0", "-b 2" will not really help. 
Actually I had to spent a lot of work on the WQY fonts: https://github.com/olikraus/u8g2/issues/455
The width argument of the BBX statement was wrong for many glyphs: The actuall bitmap was larger than specified in the BBX statement. Maybe your font has a similar issue. 

What is the current issue with your font? Any bdfconv warnings? How does it look like?
Which unicode is affected? Can you share a bdf file?

closed 2020-06-14 by olikraus