olikraus/u8g2 #1111: Nano Every and RET012864GYPP3N00000

chicodefe, 2020-02-17:
The Display RET012864GYPP3N00000 and the drive SSD1305 is not constructor for that.

Using the following constructor: U8G2_SSD1305_128X64_ADAFRUIT_F_HW_I2C
The OLED powers up and displays graphics.

The issue is the displayed graphics appear to be offset in X and scaled incorrectly. Also rogue pixels are being displayed on screen right.

[DIL12864.PDF](https://github.com/olikraus/u8g2/files/4214264/DIL12864.PDF)

olikraus, 2020-02-17:
Any pictures?

chicodefe, 2020-02-18:
//Librerias para Arduino
#include <Arduino.h>

//Librerias para Comunicacion Serial
#include <SoftwareSerial.h>
#include <api/HardwareSerial.h>

//Librerias para I2C
#include <Wire.h>

//Librerias para Pantalla Grafica
#include <U8g2lib.h>
#include <U8x8lib.h>

//Constructor para Pantalla Grafica Oled
U8G2_SSD1305_128X64_ADAFRUIT_F_HW_I2C u8g2(U8G2_MIRROR, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ 9);

void setup() {
  //Iniciar Pantalla
  u8g2.begin();
}
void loop(void) {

  u8g2.clearBuffer();
  u8g2.setFlipMode(1);
 

  u8g2.firstPage();
  do {
 u8g2.drawFrame(0, 0, u8g2.getDisplayWidth(), u8g2.getDisplayHeight());

  
    
   //   u8g2.drawPixel(0, 0);

 u8g2.setFont(u8g2_font_4x6_tr);
 u8g2.setCursor(3, 17);
u8g2.print("Hello World!");

 u8g2.setFont(u8g2_font_profont10_tr);
 u8g2.setCursor(20, 50);
u8g2.print("Hello World!");
  } while ( u8g2.nextPage() );
  delay(1000);
}
[image]

chicodefe, 2020-02-18:
I Change the code and work fine:
  /* default_x_offset = */ 2,
  /* flipmode_x_offset = */ 2,
 to
  /* default_x_offset = */ 4,
  /* flipmode_x_offset = */ 0,

but i think is best use a new constructor for the display, some that:
U8G2_SSD1305_128X64_NONAME_F_HW_I2C

/* timing from SSD1306 */
static const u8x8_display_info_t u8x8_ssd1305_128x32_adafruit_display_info =
{
  /* chip_enable_level = */ 0,
  /* chip_disable_level = */ 1,
  
  /* post_chip_enable_wait_ns = */ 20,
  /* pre_chip_disable_wait_ns = */ 10,
  /* reset_pulse_width_ms = */ 100, 	/* SSD1306: 3 us */
  /* post_reset_wait_ms = */ 100, /* far east OLEDs need much longer setup time */
  /* sda_setup_time_ns = */ 50,		/* SSD1306: 15ns, but cycle time is 100ns, so use 100/2 */
  /* sck_pulse_width_ns = */ 50,	/* SSD1306: 20ns, but cycle time is 100ns, so use 100/2, AVR: below 70: 8 MHz, >= 70 --> 4MHz clock */
  /* sck_clock_hz = */ 4000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 40,
  /* write_pulse_width_ns = */ 150,	/* SSD1306: cycle time is 300ns, so use 300/2 = 150 */
  /* tile_width = */ 16,
  /* tile_hight = */ 4,
  /* default_x_offset = */ 4,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 128,
[image]

  /* pixel_height = */ 32
};

olikraus, 2020-02-18:
>  but i think is best use a new constructor for the display, some that:
U8G2_SSD1305_128X64_NONAME_F_HW_I2C

Which constructor did you use before?

chicodefe, 2020-02-18:
Im Using the following constructor: U8G2_SSD1305_128X64_ADAFRUIT_F_HW_I2C

olikraus, 2020-02-21:
ToDo: Create new constructor with the above mentioned settings.

olikraus, 2020-02-23:
I will keep this open until this is implemented.

olikraus, 2020-10-26:
added new constructor U8G2_SSD1305_128X64_RAYSTAR_xxx

olikraus, 2020-10-26:
Not tested. Hope it will work.

closed 2020-10-26 by olikraus