olikraus/u8g2 #1114: U8g2 8x8 FONTS: how to install other fonts

paolovr70, 2020-02-24:
Hi, l'm using the u8g2 library with SSD1306 display (HW i2C). To reduce memory usage I'm using U8x8 fonts. The example "hello world!" works perfectly. Inside the font list there are some fonts that point to font groups (of fonts), such as fntgrpx11. How to use one or more of these other fonts? (i.e. u8g2_font_helvR08_tf). So far compiling this font with Arduino IDE the message which the compiler gives me is 'u8g2_font_helvR08_tf' was not declared in this scope" . Thanks. p.

olikraus, 2020-02-24:
'u8g2_font_helvR08_tf' is for u8g2. U8x8 start with the prefix "u8x8_". The font format is different for both APIs. 

A list of all u8x8 fonts is here: https://github.com/olikraus/u8g2/wiki/fntlist8x8

You can generate a u8x8 from the bdf source with the tool bdfconv (https://github.com/olikraus/u8g2/issues/1114, not sure whether the win exe is up to date)

U8x8 is much simpler and the restrictions are:
 - The font glyphs have to fit into a grid of 8x8 pixel
 - The size is the same for each glyph, this means that variable-sized fonts (like helv08) may look ugly.

Conversion looks like this:
```
./bdfconv -y -1 -th 2 -tv 3 -v -f 2 -m '32-127' ../bdf/courB18.bdf -o u8x8_font_courb18_2x3_r.c  -n u8x8_font_courB18_2x3_r
```
`-th` and `-tv` define the size of the grid. Their arg has to be multiplied with 8: So this example will place the glyphs on a 16x24 bitmap.
`-y` defines an additional offset and `-f 2` renders u8x8 fonts

helvR08.bdf is located here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdf, but again, it might not look that good.

paolovr70, 2020-02-24:
Thank you Oliver for the fast reply. I'm looking for a scalable font (to be used from 6-7 to 20-24 pixels) with very few space between characters. (I have a lot of information to display). Which FONT do you suggest (using SSD1306 "SSD1306Wire.h" library "Arial MT Plain" gave me a good result. As your library is more flexible I would like to be able to replicate ... thanks paolo

olikraus, 2020-02-24:
I do not know "SSD1306Wire.h". This is the u8g2 project. It is not related to SSD1306Wire.h

For a small narrow font, I suggest to use u8g2 (page loop). U8x8 will never not fit that good.

Here are some font suggestions:

https://github.com/olikraus/u8g2/wiki/fntgrptulamide#nerhoe
https://github.com/olikraus/u8g2/wiki/fntgrpbitfontmaker2#lastapprenticethin
https://github.com/olikraus/u8g2/wiki/fntgrpnbp#prospero_nbp
https://github.com/olikraus/u8g2/wiki/fntgrpnbp#balthasar_regular_nbp

Or you just look by yourself: https://github.com/olikraus/u8g2/wiki/fntlistall

closed 2020-07-05 by olikraus

warmonkey, 2021-11-19:
check this out https://andrzejo.github.io
which is an u8g2 / u8x8 font editor
https://andrzejo.github.io/demo/u8x8-font-editor.html
https://andrzejo.github.io/demo/u8x8-glyph-editor.html