olikraus/u8g2 #1116: Support for MAX7219 16X16 Double Height Displays

modarsoos, 2020-02-26:
I want to make a 16x16 led matrix with your library, and use constructor like print() instead of drawstring(),
because i want to show some values on the led matrix, like numbers "speed from GPS Module" for example.

i read this article from the other library too but your library is easy to learn:
Double Height Displays
https://arduinoplusplus.wordpress.com/2017/03/15/parola-a-to-z-double-height-displays/

Best Regards,
Modar Soos

modarsoos, 2020-02-26:
we want like this zones on your lirbary:

#define HARDWARE_TYPE MD_MAX72XX::FC16_HW
#define NUM_ZONES 2
#define ZONE_SIZE 4
#define MAX_DEVICES (NUM_ZONES * ZONE_SIZE)

#define ZONE_UPPER  1
#define ZONE_LOWER  0

olikraus, 2020-02-27:
Should be doable, but I do not have time to work on this at the moment. Maybe next week.

olikraus, 2020-03-01:
I have created beta 2.28.0
Can you ran the example from here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/MAX7219_U8g2/MAX7219_U8g2.ino?

What will you see on the display? Can you attach a picture of the display content to this issue?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

modarsoos, 2020-03-01:
i test it but the display still blinking
i make this edit
//u8g2.drawStr(0,7,"U8g2");     // write something to the internal memory
u8g2.drawStr(0,7+8,"abcd");   // write something to the internal memory
 
but i realize the "abcd" string look like "ab" on one zone 8*16
and the same "ab" on the other zone 8*16

olikraus, 2020-03-01:
> i test it but the display still blinking

https://github.com/olikraus/u8g2/blob/6769d3c487b35152796a7f5c61893bc170af822e/sys/arduino/u8g2_full_buffer/MAX7219_U8g2/MAX7219_U8g2.ino#L63

I assume blinking is caused by the delay. Will the blinking frequency change with a different delay value?

There is indeed one more problem: The 16x16 display can not be supported without HUGE changes to u8g2. I thought it would be more easier, but it isn't. I am sorry to say, that u8g2 can not support MAX7219 displays higher than 8 pixel lines. :-(

olikraus, 2020-03-01:
same problem here: https://github.com/olikraus/u8g2/issues/694

modarsoos, 2020-03-01:
> There is indeed one more problem: The 16x16 display can not be supported without HUGE changes to u8g2. I thought it would be more easier, but it isn't. I am sorry to say, that u8g2 can not support MAX7219 displays higher than 8 pixel lines. :-(

Why? is there any solution>?

olikraus, 2020-03-01:
> Why? is there any solution>?

This is software. There is always a solution, but the effort is too high to change u8g2.

There is one fundamental assumption in u8g2 software architecture: U8g2 assumes, that one page of data can be addressed independently. This assumption is not true for 16x16 led displays.
  
> Is there any way to help you to make your library the best library on the earth?

u8g2 fits to a specific purpose and supports specific kind of displays. Adding displays which don't fit will increase codesize and will make the lib slower of all other usecases.

Something good will be come bad if it becomes more generic.

olikraus, 2020-07-05:
unfortunately, this can not be implemented... closing this...

closed 2020-07-05 by olikraus