olikraus/u8g2 #1118: Strange dots displaying with OLED 128x32

khoadvfirmware, 2020-02-27:
Hello Olikraus,
Firstly, Thank you very much for a great open source lib!.

I tried to use it with my OLED 128x32 (SSD1305) displaying. 
The displaying in range of 128x32 matrix seems good, but **the problem is some strange pixel at outside of 128x32 matrix is turned ON (as below image).** 
[image]

Do you have any experiences related to this problem.?
Could you help to give me your opinions?

Thank you very much!
Best regards,

olikraus, 2020-02-27:
Strange. Can you test a sh1106 128x64 driver. Will the pixel still appear?

khoadvfirmware, 2020-02-28:
Hi olikraus,

After re-check again for some specification. It seems that some data pin of SSD1305 are not be connected right. I'm confirming it again.
So i will temporary close this topic first!.
If the problem are still occur, i will update further information to you.

Thank you for your reply.!
Best regards,

closed 2020-02-28 by khoadvfirmware

mikepparks, 2020-03-25:
I'm seeing this on a 128x32 module as well using "NONAME" HW & SW drivers. Tested using the SH1106 driver as suggested, but the image is now upside-down. Artifacts are still displayed on the physical right side regardless of driver.

[image]
[image]

olikraus, 2020-03-26:
created a new issue for this