olikraus/u8g2 #1119: About PowerSave

iggymayer, 2020-02-27:
Hello, I want to use a button to achieve: the screen lights up after pressing it, and the screen sleeps after a delay. I tried to write a program, but the screen cycle flickers after uploading. What is wrong?

#include <U8g2lib.h>
#include <NTPClient.h>
#include <WiFi.h> // for WiFi shield
#include <WiFiUdp.h>

const char *ssid     = "SSID";
const char *password = "PASSWORD";
const int EN = 25;
int buttonState = 0;

U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI U8g2(U8G2_R0,/*CS*/U8X8_PIN_NONE,/*DC*/17,/*RES*/16);

WiFiUDP ntpUDP;

NTPClient timeClient(ntpUDP, "pool.ntp.org", 8*3600, 10*60*1000);

void setup()
{
  pinMode(EN,INPUT);
  
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print (".");
  }
  timeClient.begin();
  
  Serial.begin(115200);
  
  u8g2.begin();
  u8g2.setContrast(1);
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
}

void loop()
{
  buttonState = digitalRead(EN);
  if (buttonState == HIGH)
  {
    u8g2.setPowerSave(0);
    //delay(5000);
  }
  else
  {
    u8g2.setPowerSave(1);
  }
  timeClient.update();

  Serial.println(timeClient.getFormattedTime());
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_logisoso28_tn);
  u8g2.setCursor(0,22);
  u8g2.println(timeClient.getFormattedTime());
  u8g2.sendBuffer();

  delay(1000);
}

iggymayer, 2020-02-27:
```
#include <U8g2lib.h>
#include <NTPClient.h>
#include <WiFi.h> // for WiFi shield
#include <WiFiUdp.h>

const char *ssid     = "SSID";
const char *password = "PASSWORD";
const int EN = 25;
int buttonState = 0;

U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0,/*CS*/U8X8_PIN_NONE,/*DC*/17,/*RES*/16);

WiFiUDP ntpUDP;

NTPClient timeClient(ntpUDP, "pool.ntp.org", 8*3600, 10*60*1000);

void setup()
{
  pinMode(EN,INPUT);
  
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print (".");
  }
  timeClient.begin();
  
  Serial.begin(115200);
  
  u8g2.begin();
  u8g2.setContrast(1);
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
}

void loop()
{
  buttonState = digitalRead(EN);
  if (buttonState == HIGH)
  {
    u8g2.setPowerSave(0);
    //delay(5000);
  }
  else
  {
    u8g2.setPowerSave(1);
  }
  timeClient.update();

  Serial.println(timeClient.getFormattedTime());
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_logisoso28_tn);
  u8g2.setCursor(0,22);
  u8g2.println(timeClient.getFormattedTime());
  u8g2.sendBuffer();

  delay(1000);
}
```

olikraus, 2020-02-27:
Your code still writes to the display during power down. This should be avoided.

closed 2020-06-14 by olikraus