olikraus/u8g2 #1122: ESP32 Guru Meditation Error: Core  1 panic'ed (StoreProhibited).

antonmeyer, 2020-03-05:
First thanks a lot for your great lib. I had a lot of fun with it in the past and learned also from your coding concepts. I'm also a bit shy to bother you with this, but I'm also curious.

The backtrace , which was hard to catch for me, points to u8x8_DrawGlyph platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:287.

These crashes happens randomly. Meanwhile I have reduced my code in the loop to 1 call.
But stays the same.

char displaystr[20]; 
snprintf(displaystr, 17, "%sC %2d%% %4d", tempstr[stnId], newHum, diff/1000);
OLED.drawUTF8(2, stnId, displaystr);

I'm using interrupt for receiving 433 MHz messages from the ambient sensor.
The ISR looks like following
volatile unsigned long EdgeTime;
void IRAM_ATTR RF_ISR()
{
  EdgeTime = micros();
} // end of isr

And now my guesses:
I found in the u8g2lib special code for the 8266. Might be the ESP32 does not profit from it?
There are some hints to the IRAM_ATTR. I have seen some encapsulations casting the display string into const.
On the other side, I have tried to get rid of the crashes by disabling the RF_ISR interrupt before writing to the display. did not help.
So I'm a bit lost and think about to switch to the display lib from Heltec. (which I did not like too much)

During the initialization I use wire.begin for setting the i2c bus pins to 15 and 4, before calling u8x8.begin.  I got an error that CDA=0 and CLK=0 does not work. But finally it works.

I wonder, that nobody else run into the same problem.
So I wonder if the lib is ready for the ESP32, is there an issue in combination of using interrupts?

olikraus, 2020-03-05:
> I found in the u8g2lib special code for the 8266. Might be the ESP32 does not profit from it?

Any link? You can go to the src section (https://github.com/olikraus/u8g2/tree/master/csrc), click on the line number of the file and copy the url to this issue.

I probably need more details on the crash itself, but basically code should run on ESP32.

antonmeyer, 2020-03-05:
https://github.com/olikraus/u8g2/blob/5ff1f15dbc58b184bb307b74cf7933780a7fa474/csrc/u8x8_8x8.c
e.g. line 40

#if defined(ESP8266)
uint8_t u8x8_pgm_read_esp(const uint8_t * addr) 
{
   uint32_t bytes;
    bytes = *(uint32_t*)((uint32_t)addr & ~3);
    return ((uint8_t*)&bytes)[(uint32_t)addr & 3];
}
#endif

antonmeyer, 2020-03-05:
I could reproduce it by increasing write frequency to the Display and let the Interrupt run. But takes still some time.
C:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH> c:\Users\user\.platformio\packages\toolchain-xtensa32\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH\.pio\build\heltec_wifi_lora_32_V2\firmware.elf  0x7fe049a4:0x3ffb1e70 0x400d204d:0x3ffb1ea0 0x400d20c3:0x3ffb1ee0 
0x400d2109:0x3ffb1f10 0x400d19fe:0x3ffb1f30 0x400d6dfd:0x3ffb1fb0 0x400893d1:0x3ffb1fd0
0x7fe049a4: ?? ??:0
0x400d204d: u8x8_DrawGlyph at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:287
0x400d20c3: u8x8_draw_string at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:287
0x400d2109: u8x8_DrawUTF8 at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:392
0x400d19fe: U8X8::drawUTF8(unsigned char, unsigned char, char const*) at C:/Users/user/.platformio/lib/U8g2_ID942/src/U8x8lib.h:269
 (inlined by) loop() at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/main.cpp:71
0x400d6dfd: loopTask(void*) at C:\Users\user\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:19
0x400893d1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

olikraus, 2020-03-05:
I do not know much about the ESP32, but you could change
https://github.com/olikraus/u8g2/blob/5ff1f15dbc58b184bb307b74cf7933780a7fa474/csrc/u8x8_8x8.c#L40
and
https://github.com/olikraus/u8g2/blob/5ff1f15dbc58b184bb307b74cf7933780a7fa474/csrc/u8x8.h#L158

from 
"#if defined(ESP8266)" 
to 
"#if defined(ESP8266) || defined(ESP_PLATFORM) || defined(xtensa)"

in your local copy of the files. Will this help?

antonmeyer, 2020-03-05:
> from
> "#if defined(ESP8266)"
> to
> "#if defined(ESP8266) || defined(ESP_PLATFORM) || defined(xtensa)"
> in your local copy of the files. Will this help?

Looks Logical / consequent. But I could not tell if this is an improvement. the whole PROGMEM is about reducing RAM usage. And the crash might come from reading from flash at the wrong time. which is very anoying that I have to care About it at all.

olikraus, 2020-03-05:
It may look like this is related to AVR PROGMEM but it is more like a unlucky coincidence.
The above #if solves the problem that ESP8266 can not access random bytes in memory.
The PROGMEM indeed solves the AVR access to Flash in order to save RAM.
In u8g2 both problems exist for the same data structures (the fonts): The u8g2 fonts are stored in flash (to save RAM) but also additionally require byte access.

Anyhow, did you change your code and will the above patch work?

antonmeyer, 2020-03-05:
ok I added it in both files and started the test. I guess I have to wait a while. Before that I had changed from drawUTF8 to drawString. Did run 90 min stable ...
I wonder if there is a way to speed up / increase the probability of the crash. (if there is any anymore). I´ll let you know.

antonmeyer, 2020-03-05:
0x7358d4ff: ?? ??:0
0x400d2040: u8x8_DrawGlyph at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:354
0x400d20bb: u8x8_draw_string at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:354
0x400d20e9: u8x8_DrawString at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:354
0x400d19de: U8X8::drawString(unsigned char, unsigned char, char const*) at C:/Users/user/.platformio/lib/U8g2_ID942/src/U8x8lib.h:266
 (inlined by) loop() at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/main.cpp:74
0x400d6dcd: loopTask(void*) at C:\Users\user\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:19
0x400893d1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

antonmeyer, 2020-03-06:
to have more fun, google script for adding values has stoped working 23:00.
agile times. Might be a bug or API change by google.
But in general the preprocessor statements did not improve stability.

olikraus, 2020-03-06:
soo? What exactly does this mean for u8g2? The crash still happens?

antonmeyer, 2020-03-06:
yes still happens but I just recognize a change:
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited).
now the InstrFetch panics. before it was 
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Have to double check it. I could not exclude that the u8g2lib is blamed false by the backtrack.
because that one looks very chaotic:
0x40000000: ?? ??:0
0x400d200d: u8x8_DrawGlyph at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:354
0x400d13a7: add(unsigned char) at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/F007TH.h:110
0x400d1471: RFinit(unsigned char) at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/F007TH.h:163
0x400d1678: send2google(String) at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/send2GS.h:29
0x400d18ba: loop() at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/main.cpp:66
0x400d6e09: uartBegin at C:\Users\user\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-uart.c:411
0x400893d1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1

antonmeyer, 2020-03-06:
next crash blames the HTTPClient. So my current working thesis is: problem moved to a different place. I assume a relation with the "free running" PIN-interrupt. so I will revert the stress test code and make a long time observation.

0x400d328f: HTTPClient::disconnect(bool) at C:\Users\user\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient\src/HTTPClient.cpp:181
0x400d3301: HTTPClient::end() at C:\Users\user\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient\src/HTTPClient.cpp:181
0x400d44f9: HTTPClient::begin(String) at C:\Users\user\.platformio\packages\framework-arduinoespressif32\libraries\HTTPClient\src/HTTPClient.cpp:181
0x400d16bd: send2google(String) at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/send2GS.h:32 (discriminator 4)
0x400d196e: loop() at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/main.cpp:66
0x400d722d: loopTask(void*) at C:\Users\user\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:19
0x400893d1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

antonmeyer, 2020-03-06:
now the u8g2lib is back in business ;-) 
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
0x00000000: ?? ??:0
0x400d1fe0: u8x8_DrawGlyph at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:354
0x400d205b: u8x8_draw_string at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:354
0x400d2089: u8x8_DrawString at C:\Users\user\.platformio\lib\U8g2_ID942\src\clib/u8x8_8x8.c:354
0x400d1349: U8X8::drawString(unsigned char, unsigned char, char const*) at C:/Users/user/.platformio/lib/U8g2_ID942/src/U8x8lib.h:266
 (inlined by) saveReading(int, int, int) at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/F007TH.h:94
0x400d13e6: add(unsigned char) at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/F007TH.h:139
0x400d15ec: check_RF_state(unsigned char) at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/F007TH.h:267
0x400d185e: loop() at c:\Users\user\Documents\PlatformIO\Projects\Heltec_F007TH/src/main.cpp:55
0x400d7225: loopTask(void*) at C:\Users\user\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:19
0x400893d1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

antonmeyer, 2020-03-06:
I have rewritten the 433 Receiver, without Interrupts, updated your lib, added 
#if defined(ESP8266) || defined(ESP_PLATFORM) || defined(xtensa)

and now it is crashing in the httpClient … 
I should also consider Hardware issues.

olikraus, 2020-03-06:
Hmm interesting case. I can probably just look at your investigation

antonmeyer, 2020-03-07:
changing the HW results in the same bug. meanwhile everything points to the HTTPClient. I close the issue. May be anyhow the extended #if defined could help.

closed 2020-03-07 by antonmeyer

olikraus, 2020-03-07:
Thanks for the discussion. I will keep an eye on the #if topic.

antonmeyer, 2020-03-12:
Hi Oli,

I think I found the bug. It was in my adapted code for the 433 MHz Manchester decoder.
Only in case of a special receiving error condition ..grrrr

I learnt 2 lessons, do not trust copied spaghetti code and do not trust the traceback.
Stability test is running.

Sorry for disturbing you with this.

Cheers R.

olikraus, 2020-03-12:
Thanks for updating this issue.