olikraus/u8g2 #1124: NHD-C12864LZ-FSW-FBW-3V3

PedroAFlores, 2020-03-06:
Hello!
I am a beginner using STM32 and I would like to use u8g2 lib in my GLCD. I have been trying for several days use this lib with my setup but unfortunately I could not do it.
I have followed all instructions step by step.
I am using STM32F070 with ST7565R on NHD_C12864LZ_FSW_FBW_3V3 display.
Commucation via SPI (port 2) and GPIOs (CS,RES & DC) at shows:
PC0: CS
PC1: RES
PB0: A0
This is main.c code attached.

Thank you very much in advanced
[main.txt](https://github.com/olikraus/u8g2/files/4297888/main.txt)

PedroAFlores, 2020-03-06:
Hello!
I have found a wiring issue!
Contrast turned down to 100 and it seems great!
Thank you very much your hard work, creating this lib!
[image]

closed 2020-03-06 by PedroAFlores

olikraus, 2020-03-06:
:)

olikraus, 2020-03-06:
So you solved the problem?

PedroAFlores, 2020-03-07:
Hello Olikraus!
Yes I found a wiring issue after a lot of time wasting :(
I have to watch every function and take control hehe. Thank you very much, I have followed this link. It was very helpfull.
https://elastic-notes.blogspot.com/2018/10/u8g2-library-usage-with-stm32-mcu.html