olikraus/u8g2 #1125: Scroll Left to Right

malltinaa, 2020-03-08:
Hi,
Thanks for sharing this library.
I use the [ScrollingText.ino](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/ScrollingText/ScrollingText.ino) example.
How do I scroll from left to right. (I use Arabic letter)
I made a few changes but I didn't succeed.

Thanks.

olikraus, 2020-03-08:
Basically you have to increase the offset:
https://github.com/olikraus/u8g2/blob/21f5a74dff0d658a49d407fc5cd46ebdeea464f8/sys/arduino/u8g2_page_buffer/ScrollingText/ScrollingText.ino#L330

The condition and the starting point for the offset are probably also different:
https://github.com/olikraus/u8g2/blob/21f5a74dff0d658a49d407fc5cd46ebdeea464f8/sys/arduino/u8g2_page_buffer/ScrollingText/ScrollingText.ino#L331-L332

The example also assumes, that the initial offset value is 0, it might be different also:
https://github.com/olikraus/u8g2/blob/21f5a74dff0d658a49d407fc5cd46ebdeea464f8/sys/arduino/u8g2_page_buffer/ScrollingText/ScrollingText.ino#L280

malltinaa, 2020-03-09:
Thanks for the reply.
I made the following changes:

`offset+=1;`

and

`if ( (u8g2_uint_t)offset > (u8g2_uint_t)-width )`

But I can't change the starting point. The text starts moving through the middle of the screen.
How to enter text from the outside of screen?
Please see the video below.

`https://vimeo.com/user109454851/review/396395448/c0a809760e`

olikraus, 2020-03-09:
he text starts moving through the middle of the screen.

See my remark above:
You need to update line 280 or add an assignment in the setup() procedure.

malltinaa, 2020-03-09:
> > the text starts moving through the middle of the screen.
> 
> See my remark above:
> You need to update line 280 or add an assignment in the setup() procedure.

Thanks but apparently changing the `offset` value has no effect on the starting point. I made all the changes and did various tests.
I use the following code, unfortunately I'm confused and I don't know where is problem!
```
...
u8g2_uint_t offset;			// current offset for the scrolling text
u8g2_uint_t width;			// pixel width of the scrolling text (must be lesser than 128 unless U8G2_16BIT is defined
const char *text = "U8g2 ";	// scroll this text from right to left

void setup(void) {
  u8g2.begin();   
  u8g2.setFont(u8g2_font_inb30_mr);	// set the target font to calculate the pixel width
  width = u8g2.getUTF8Width(text);		// calculate the pixel width of the text 
  u8g2.setFontMode(0);		// enable transparent mode, which is faster
  offset = 0;
}

void loop(void) {
  u8g2_uint_t x; 
  u8g2.firstPage();
  do {
    // draw the scrolling text at current offset
    x = offset;
    u8g2.setFont(u8g2_font_inb30_mr);		// set the target font
    do {								// repeated drawing of the scrolling text...
      u8g2.drawUTF8(x, 30, text);			// draw the scolling text
      x += width;						// add the pixel width of the scrolling text
    } while( x < u8g2.getDisplayWidth() );		// draw again until the complete display is filled
    u8g2.setFont(u8g2_font_inb16_mr);		// draw the current pixel width
    u8g2.setCursor(0, 58);
    u8g2.print(width);					// this value must be lesser than 128 unless U8G2_16BIT is set
    
  } while ( u8g2.nextPage() );
  offset+=1;							// scroll by one pixel
  if ( (u8g2_uint_t)offset > (u8g2_uint_t)-width )	
    offset = 0;							// start over again
  delay(10);							// do some small delay
}
```
If possible, test the code.
I'm using a 128x64 SSD1306 display.

olikraus, 2020-03-09:
I try to support here, but actually your problem is a simple programming question, not a u8g2 problem. I mean, your problem/question will be the same with any other lib. I neither have the time nor the resources test/review such code. Remember: This is open source, you something for free, I do not get a single cent...

I always will be happy to receive problems/bugs/enhancement requests for u8g2.

idreamsi, 2020-03-15:
Hi @olikraus 
A scroll-related question:
How to use long texts. (E.g. 1000 character or more)
Is it enough to add `#define U8G2_16BIT` in `u8g2.h`?

olikraus, 2020-03-16:
Yes 16 bit more will help

idreamsi, 2020-03-26:
Thanks @olikraus 
Scrolling is very slow when I use long texts (for example, getting news headlines from a site's).
Is there a solution? For example, using buffers, etc.

olikraus, 2020-03-26:
> Scrolling is very slow when I use long texts and 16 bit mode 

Yes, because U8g2 will just draw all (1000) chars although most of them are outside the screen. U8g2 will try to detect this but still it will be slow.

> Is there a solution?

Yes: Draw only a subset of the text. I mean, you need to calculate the which part of the text is visible and then only draw that part (with a little bit more outside)

I once made an example for this: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/Weather/Weather.ino#L329

The assumption is, that the width of each char is equal (8 pixel). If this assumption is not true, then the code will be even more complicated.

closed 2020-07-05 by olikraus