olikraus/u8g2 #1126: Some problems occurred when using JLX256160

vowovrz, 2020-03-10:
Thanks for sharing this library.

[image]

olikraus, 2020-03-10:
Welcome.
What is the problem?

vowovrz, 2020-03-11:
Background is black.

olikraus, 2020-03-11:
Maybe change the contrast.

vowovrz, 2020-03-11:
When choosing anti-comment
U8X8_ST75256_JLX240160_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

It looks like this.  Font display is dim.
[image]

vowovrz, 2020-03-11:
Thanks , it's works，testContrast (145);

vowovrz, 2020-03-12:
thanks for the reply!
When the method getWidth () is executed, the returned width is 240, which should actually be 256, and the height is correct.

olikraus, 2020-03-12:
> When the method getWidth () is executed, the returned width is 240, which should actually be 256, and the height is correct.

You need to enable 16 bit support (see the FAQ).

closed 2020-07-05 by olikraus