olikraus/u8g2 #1128: Serial.print shows "34", U8x8.drawString displays "3423" for same string

iforce2d, 2020-03-11:
I came across a very strange problem with my SSD1306 module using the U8x8 library. 

[image]

[image]

The full code for arduino is below, see the last five lines for a basic overview. Believe it or not this is actually a minimum reproduction case, I tried for about two hours to get it as small as I could, and changing just about anything will cause the problem not to occur.

The problem seems to be related to specific values of the number being printed, and possibly also something to do with the variable types being used, or casting, or intermediate variables in calculations. That sounds very vague, but what I mean is that even though it's obvious what some values are (eg. maTotal will be 6840), setting that value explicitly makes the problem go away.

I am running this on an Arduino pro-mini equivalent, IDE version is v1.8.5

```#include <Arduino.h>
#include <U8x8lib.h>

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(U8X8_PIN_NONE);

unsigned int movingAverage[10];
unsigned int maTotal = 0;

void setup() {
  Serial.begin(9600);
  
  u8x8.begin();
  u8x8.setFont(u8x8_font_8x13B_1x2_n);
  
  memset( movingAverage, 0, 10 * sizeof(unsigned int) );
}

void loop() {
  
  static int index = 0;
  maTotal -= movingAverage[index];
  maTotal += 684;
  movingAverage[index] = 684;
  index = (index + 1) % 10;

  int avg = (int)(maTotal / 10.0f);
  float fraction = (float)avg - 684.0f;
  float a = 3.348f;
  float b = -1.0f;
  float c = 76.0f;
  float d = 0.373f;
  float e = a + ( fraction / c ) * d - 3;
  int shouldBe34 = (int)(e * 100);

  char buf[16];  
  sprintf(buf, "%d", shouldBe34 );

  Serial.println( buf );       // prints 34
  u8x8.drawString(0, 0, buf);  // displays 3423
  
}

```

olikraus, 2020-03-11:
I think the 23 is a leftover of a previous lool.
Place a "u8x8.clearDisplay();" before u8x8.drawString(...).

iforce2d, 2020-03-11:
Thanks Oli, that works. However the display flickers so much it's very hard to read, not really usable.
For this project I have just enough memory space to spare, so I think I can use the U8g2 instead, which looks pretty nice.

iforce2d, 2020-03-11:
Actually I do kinda need the memory, so I came back to this and eventually figured out where the extra numbers were coming from. In the first few loops after startup some of the values were negative (and large), this is not a problem for the project overall but I did not realize the characters would never be cleared from the display.

I also realized that to avoid the flickering, I could do something like `u8x8.drawString(6, 0, " ");` to blank out individual spaces if necessary.

Thanks for the assistance and a great library !

closed 2020-03-11 by iforce2d