olikraus/u8g2 #1129: USB needed to be connected to run off external power

patchworkalex, 2020-03-17:
Running a Teensy4.0 with an SSD1327 128x128 MIDAS display.

The Teensy gets stuck at u8g2.begin() when booting up off external power until a USB device is connected (even a portable USB charger does the trick!). The rest of my code works fine from external power alone when I remove all of the u8g2 functions so I am fairly certain that I've narrowed it down to something in this library, but from scanning though the src files I can't spot anything that would be causing u8g2.begin to become stuck in a loop.

< From reading the PJRC forum it seems in the past people have had issues with void setup waiting for serial commands before continuing. I wonder if that is a helpful clue? >

#include "FlexCAN_T4.h"
FlexCAN_T4 <CAN2, RX_SIZE_256, TX_SIZE_16> Can0;
#include <Arduino.h>
#include <U8g2lib.h>
U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI u8g2(U8G2_R2,/* cs=*/ 7, /*miso dc=*/ 9 ) ;

  u8g2.begin();
// it gets stuck here!!

olikraus, 2020-03-17:
A lib doesn't consume power, but the display does. Begin() will activate the display.

patchworkalex, 2020-03-17:
the external power is a l78l05 which can give out up to 100mA.

could this be not enough?

olikraus, 2020-03-17:
Depends on your OLED. The 78l05 may also only deliver 100mA  with good cooling. Also remember to add at least 1uF caps near the OLED power lines to support the consumption increase.

patchworkalex, 2020-04-07:
Sorry for the delayed response, I think you've solved it! 

I'm having no problems now that I've increased the power supply

:)

closed 2020-04-07 by patchworkalex