olikraus/u8g2 #1130: What about the speed differences between _1_, _2_, _F_

19rdh06, 2020-03-19:
Hi Oliver,
I'm using a ST7920 LCD display, similar as the one on your setup page (:-))
I was wondering what the speed differences were between the different setup's. You state :
_1_ is slow 
_2_ faster then _1_ (but still slow)
_F_ is fast. 
What I didn't find on your (rather excellent) pages was: How slow, how fast. ( are the speed differences something like 1 to 10 to 100?)
Secondly: I did read a lot of discussions where people complained about the RAM usage.

Would it be nice to state some info about above matters? (I think it differs from display to display, bus some indication would be nice)
I self did some testing on an ATmega328 nano.
With an already in progress project, thus some software overhead, but still getting a picture:

// using U8G2_ST7920_128X64_1_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 8);
// using :
//  unsigned long t1 = millis();
//  u8g2.firstPage();
//  do {
//    drawSomething();
//  } while ( u8g2.nextPage() );
//  unsigned long tt = millis() - t1

// tt = 60 millis for drawSomething
// total usage bytes : 15078 (49%)
// global vars : 896 (43%) (including the rest of the program)
//
// using U8G2_ST7920_128X64_2_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 8);
// tt = 51 millis for drawSomething
// total usage bytes : 15080 (49%)
// global vars : 1024 (50%) (including the rest of the program)
//
// using U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ 8);
// using : 
//  unsigned long t1 = millis();
//  u8g2.clearBuffer();
//  drawSomething();
//  u8g2.sendBuffer();
//  unsigned long tt = millis() - t1
//
// tt = 43 millis for drawSomething
// total usage bytes : 15082 (49%)
// global vars : 1792 (87%) (including the rest of the program)

Now I have an indication of RAM usage and refresh speed.
For me it became clear that: 
 Kind of lib. used  :                                                      _1_,    _2_,     _F_
1: Slow and fast are not as slow or fast as I thought: 60,      51,      43 ms per refresh
2: The RAM usage differences are more spectacular: 43%    50%   87% RAM usage
3: Bytes usage didn't differ so much.

2 questions:
Are above measurements according your own ideas, experiences?
Is it an idea to enhance your pages with this kind of info? (I could not find it anywhere on the internet, or my search skills are very poor ...)

With regards,
Rob de Haan (Netherlands)

olikraus, 2020-03-20:
Nice question, but I am not sure how to answer this. From my perspective, I addressed most of your points here: https://github.com/olikraus/u8g2/wiki/setup_tutorial#u8g2-full-buffer--page-buffer-and-u8x8-mode

The RAM usage is described along with each constructor. See the end of the bullet point line, e.g. here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-128x64_vcomh0

I also do performance measurements. But there is no systematic measure. It actually depends not only on the mode (F vs. 1/2) but also on the hardware, the display and also the u8g2 version.

Nevertheless, results are documented here for full buffer mode: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/FPS/FPS.ino

And here for page mode:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FPS/FPS.ino

Does this help?

Oliver

19rdh06, 2020-03-20:
Hi Oliver,
Thank you for your quick response.
I totally overlooked all these pages! All the info I asked for can be found on your pages. Thank you for pointing out.
With regards,
Rob

closed 2020-03-20 by 19rdh06