olikraus/u8g2 #1131: 42x65 LCD Display with ST7565D driver

jonesmg1, 2020-03-20:
I would like to use this product for some prototyping, but it seems that the dimension is not supported. 

is there a way for me to adjust the dimension of an existing constructor, or for you to release a constructor that supports this display?

https://www.crystalfontz.com/product/cfao4265attl-42x65-graphic-lcd

olikraus, 2020-03-20:
I can add a driver for this LCD with the correct dimensions. Technically those ST7565 still differ a lot.
Can you test the exiting (128x64) ST7565 driver an tell me which one has the best visual results?
I mean, the contrast does not really matter, but any text output should appear correctly.

Other question: Is there any init command sequence available? I did not found this information in the above referenced document.

jonesmg1, 2020-03-20:
I've tried the existing 128x64 and cannot tell if there is any intelligible text/graphics.  All I get are some lines that blink. 

Reading through the documentation, I have not seen anything on an init sequence.

olikraus, 2020-03-20:
> I've tried the existing 128x64 and cannot tell if there is any intelligible text/graphics.

Can you try the u8x8 GraphicsTest demo from the examples:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8x8/GraphicsTest/GraphicsTest.ino#L138

Of course you might need to update the gpio port numbers. What will you see? Is there any usefull thing visible?

jonesmg1, 2020-03-20:
I have tried enabling the constructor on line 138:

 

U8X8_ST7565_EA_DOGM128_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

 

I get no intelligible text or graphics on the display. 

 

I get a few lines that flash, but that’s all. 

 

I’ve also double checked my hardware setup:

 

Display                 Arduino Mega 2560

CS pin 1                10

RST pin 2              8

A0 pin 3                9

SCL pin 4              13

SI pin 5                  11

Vdd pin 6             3.3v

Vss pin 8               gnd

 

Michael Jones

 

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8x8/GraphicsTest/GraphicsTest.ino#L138

olikraus, 2020-03-20:
How can there be flashing lines? Can you send a picture including your setp? I think there must be something visible.

jonesmg1, 2020-03-20:
Looking at the documentation more carefully, I should have realized that the signal voltage on this Arduino is 5v and the logic on the LCD is 3.3 – not just the supply voltage.  

 

I may have toasted the LCD trying to test it without the needed level-shifter

 

I have some other 128x64 displays (also 3.3v) that I should be able to test once I locate or create a level shifter, and I will order a few more of smaller LCD’s as well.  

 

Michael Jones

 

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8x8/GraphicsTest/GraphicsTest.ino#L138

olikraus, 2020-03-20:
hm.. these LCDs often survive 5V at the input pins.

jonesmg1, 2020-03-20:
Here's what I have:  

[image]
[image]
[image]

I also have a video that I cannot include here - but emailed to you.

olikraus, 2020-03-21:
Didn't you add the external caps as depict on page 11 of the LCD datasheet? They are mandatory.

jonesmg1, 2020-03-21:
I’m embarrassed to say that I did not.. I didn’t realize they were mandatory for basic function.

 

Thank you, and I’ll get back to you when I’ve followed the directions more carefully

 

Michael Jones

olikraus, 2020-03-21:
Ok...

closed 2020-07-05 by olikraus