olikraus/u8g2 #1133: U8G2 User_Selection_List  Help

xmodpt, 2020-03-21:
Hello

I was trying to play with U8G2 User_Selection_List exemple but i can't find a way to make the dif menu option do dif things.

For exemple:

List

- Blink once
- Blink Twice

and make it do exectly that, if option 1 is selected, link LED once and if option 2 is selected, blink LED 2 times.

olikraus, 2020-03-21:
```
if ( current_selection == 1 ) {
 blink_1x();
}
if ( current_selection == 2 ) {
 blink_2x();
}
```

xmodpt, 2020-03-21:
Is that easy?!?!?! 🤪 Thank you m8

xmodpt, 2020-03-21:
SOLVED (this part)

One last question, what if i need to make a led blink and at the same time make a buzzer buzz within the same void ? at the moment it blink's 1st and then it runs the buzzer code.

Thank you

olikraus, 2020-03-21:
Create a state machine and/or learn about "blink without delay"

xmodpt, 2020-03-21:
just got it thank you very much for all the help.

closed 2020-03-21 by xmodpt