olikraus/u8g2 #1134: U8X8-library interferes with A/D-converter PCF8591 on the same I2C-bus - tricky!

HanksSaloon, 2020-03-21:
Hi. I have a SSD1306 128X64 display and 4-port A/D-converters (PCF8591) on the same I2C-bus, and they seem to interfere. As soon as something is written to the display, the data read back from the A/D-converters is wrong.

Setup: Arduino Nano, and on the I2C three PCF8591s, an EEPROM AT24C32 and a SSD1306 128X64 (The display is behind a 5V to 3v3 level converter).

Here's the simplified schematic 
[PCF8591_test.pdf](https://github.com/olikraus/u8g2/files/4363891/PCF8591_test.pdf)

The device addresses can be found in the program listing below.

`#include "Wire.h"
#include "U8x8lib.h"
#include "AT24C32_64.h"

#define An0104 (0x48) // 1st 4-port AD / PCF8571
#define An0508 (0x49) // 2nd 4-port AD / PCF8571
#define An0912 (0x4A) // 3rd 4-port AD / PCF8571

#define ExtMem (0x56)  // AT24C32 EEPROM
#define LCDaddr (0x78)

U8X8_SSD1306_128X64_NONAME_HW_I2C LCD(/* reset=*/ U8X8_PIN_NONE); // Create LCD
AT24C32_64 EXM(ExtMem); // Create external EEPROM

const int AnSet[3] = {An0104, An0508, An0912 };
byte rv0, rv1, rv2, rv3;
float volt[4];

void setup() {
  Wire.setClock(100000);
  Wire.begin();
  Serial.begin(115200);
  LCD.setI2CAddress(LCDaddr);
  LCD.begin();
  EXM.initialize();  // OPTION 2

  LCD.setFont(u8x8_font_saikyosansbold8_u); // Select font
}

void loop() {
  for (byte i=0; i<=2; i++) {
    Serial.print("Testing module nr "); Serial.println(i);
    Wire.beginTransmission(AnSet[i]);
    Wire.write(0x04);
    Wire.endTransmission();
    Wire.requestFrom(AnSet[i], 5);
    rv0=Wire.read();
    rv0=Wire.read();
    rv1=Wire.read();
    rv2=Wire.read();
    rv3=Wire.read();
 
    volt[0] = rv0 * 0.11171;
    volt[1] = rv1 * 0.0556;
    volt[2] = rv2 * 0.02272;
    volt[3] = rv3 * 0.02025;

    Serial.print(rv0); Serial.print(" "); Serial.print(volt[0]); Serial.println(" VDC");
    Serial.print(rv1); Serial.print(" "); Serial.print(volt[1]); Serial.println(" VDC");
    Serial.print(rv2); Serial.print(" "); Serial.print(volt[2]); Serial.println(" VDC");
    Serial.print(rv3); Serial.print(" "); Serial.print(volt[3]); Serial.println(" VDC");
    Serial.println();
  }
 LCD.drawString(0, 0, "TEST"); // OPTION 1

delay(5000);
}`

The output to the serial monitor looks like this:
`Testing module nr 0
243 27.15 VDC
255 14.18 VDC
149 3.39 VDC
0 0.00 VDC

Testing module nr 1
245 27.37 VDC
255 14.18 VDC
186 4.23 VDC
255 5.16 VDC

Testing module nr 2
200 22.34 VDC
213 11.84 VDC
223 5.07 VDC
175 3.54 VDC

Testing module nr 0
192 21.45 VDC
224 12.45 VDC
192 4.36 VDC
192 3.89 VDC

Testing module nr 1
192 21.45 VDC
192 10.68 VDC
192 4.36 VDC
192 3.89 VDC

Testing module nr 2
192 21.45 VDC
192 10.68 VDC
192 4.36 VDC
192 3.89 VDC
`
The first time the 3 A/D-modules are read, the reading is correct, but from the second time (and all following readings) they are incorrect, like "locked up" on certain values.

In the program listing the two options (1 and 2) are interesting here.
OPTION 1:
If this line is commented out, ALL readings are **correct**.
OPTION 2
If this line is commented out, ALL readings are **incorrect**, including the first one.
(i.e. The 

The A/D-converters are very simple, there is no dedicated driver/library involved (except for Wire).
Therefore my only candidate is the U8X8-library. Is it possible that it's mixing up the I2C-bus, or [something in] the Wire-library? However, I've tested other devices (e.g. BME280, TSL2561, PCF8575) on the same bus, and they work fine with the U8X8. The A/D-converters are the only issue I've found.

Can you please comment?

olikraus, 2020-03-21:
U8x8 will probably try to set a higher bus speed. Tell u8x8 to use 100k only by using the busclock command.

HanksSaloon, 2020-03-22:
Thanks a lot! I didn't expect that the library would "override" Wire-settings.
It did it.
I'm really annoyed i didn't try that before!
Thanks for your help - and for a great library!

closed 2020-03-22 by HanksSaloon

olikraus, 2020-03-22:
U8g2/u8x8 try to assign the max speed for the display. Users would complain otherwise...