olikraus/u8g2 #1136: I'm seeing this on a 128x32 module as well using "NONAME" HW & SW drivers. Tested using the SH1106 driver as suggested, but the image is now upside-down. Artifacts are still displayed on the physical right side regardless of driver.

olikraus, 2020-03-26:
I'm seeing this on a 128x32 module as well using "NONAME" HW & SW drivers. Tested using the SH1106 driver as suggested, but the image is now upside-down. Artifacts are still displayed on the physical right side regardless of driver.

[image]
[image]

_Originally posted by @mikepparks in https://github.com/olikraus/u8g2/issues/1118#issuecomment-603962671_

olikraus, 2020-03-26:
Can you provide the datasheet for this display?

mikepparks, 2020-03-26:
https://www.waveshare.com/wiki/1.3inch_OLED_HAT

olikraus, 2020-03-26:
So you are say neither SSD1305 128X32_NONAME (https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1305-128x32_noname) nor SSD1305 128X32_ADAFRUIT (https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1305-128x32_adafruit) will work?

mikepparks, 2020-03-26:
It appears that this display will work as expected via the ADAFRUIT driver. The pixels on the edge are not visible, and the alignment appears to be correct. Using the NONAME driver, the width of the pixels displaying on the edge appear to be directly related to the number of pixels that the coordinates are being shifted to the left in normal orientation.

olikraus, 2020-03-28:
ok, then this sounds the problem is solved, right?

mikepparks, 2020-03-28:
If I use the other driver, yes. Thanks.

closed 2020-07-05 by olikraus