olikraus/u8g2 #1137: First line empty / last line white dots

lltodoll, 2020-03-29:
Hello,

I have a question:

My display shows on the left side (first column) nothing and on the right side (last column) white dots. What I am doing wrong? Or is it a bug? 

`
#include <Arduino.h>
#include <U8x8lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(U8X8_PIN_NONE); 

void setup(void)
{
  u8x8.begin();
}

void loop(void)
{
  u8x8.setFont(u8x8_font_amstrad_cpc_extended_r);
  u8x8.drawString(0, 0, "ABCDEFGHIJKLMNOP");
  u8x8.drawString(0, 1, "BCDEFGHIJKLMNOPQ");
  u8x8.drawString(0, 2, "CDEFGHIJKLMNOPQR");
  delay(100000);
  u8x8.clearDisplay();
}
`

Thanks for reply
[image]

olikraus, 2020-03-29:
It could be a SH1106 display controller instead of the SSD1306.

lltodoll, 2020-03-29:
Yes that was it. Thanks a lot :)

closed 2020-03-29 by lltodoll

v1993, 2020-03-30:
@lltodoll By the way

```
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
```

most likely can be omitted.

olikraus, 2020-03-30:
Yes... I think it was once required in an older version of the Arduino IDE.

lltodoll, 2020-03-30:
ah ok thx