olikraus/u8g2 #1138: U8G2_SSD1327_MIDAS_128X128_F_HW_I2C  setI2CAddress not working

kovachjk, 2020-03-30:
The waveshare 1.5 mono OLED mono has an option of 0x3C or 0X3D for an the I2C address.  It only works is I do not call u8g2.setI2CAddress()  or call u8g2.setI2CAddress(0x78).   The address 0x78 is for my MakerHawk 1.5 inch OLED.  
The  WaveShare will not work with the address 0X3C or 0X3D only OX78.  The MakerHawk will not work with any address.
They both used to work as expected. Was there a recent update?
I cannot follow the u8g2 libraries to see where the I2C address is being set.  Plus I tried to expose the getI2CAddress to see what the default is.(I think it's 255) but I could not.
Does it make any difference in the order of calling:
u8g2.begin();
u8g2.setI2CAddress(0x78);/

Is there any other place where I should be setting the I2C address?

olikraus, 2020-03-30:
In u8g2 you have to multiply the address with two. So for 0x3c you need to use 0x78. For 0x3d you need to use 0x7a.
The call to setI2CAddress has to happen before calling .begin().

Both questions are answered in the manual: https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress ;-)

kovachjk, 2020-04-01:
I am hiding my face in shame.  I should have read the manual
I used an example for the vendor of the OLED I had and did not pay attention to the difference.
When I changed vendors and changed the address and it did not work I started looking at the code. I did not even think of the manual.
Thank God for your manual, your code is way above my pay grade. ;-)
Thank you for all your hard work, and your continued support, many people would be lost without it.

closed 2020-04-01 by kovachjk