olikraus/u8g2 #1140: Add font to U8g2

luxxtek, 2020-03-31:
Is it possible to add [this font](https://github.com/keshikan/DSEG) to the library.
If so, what is the workflow?
Any help would be appreciated.

olikraus, 2020-03-31:
The central tool is bdfconv which converts .bdf into the internal u8g2 format. If you can provide .bdf then I can create the u8g2 fonts.

luxxtek, 2020-03-31:
I contacted the font creator, and was advised to use [this tool](http://hukka.ncn.fi/?fony).
It works after installing the font on a windows machine, importing as TFT and exporting as .bdf. I I hope this bdf file will work for you. 
How do I send this bdf file to you?

v1993, 2020-03-31:
You can drag it over comment window to add as attachment. But make sure to zip it first as GitHub limit possible extensions.

luxxtek, 2020-03-31:
Included a zip that contains the original true type font, the bdf file and a license text.
The font creator aks to include this license text in the map of the font
I don't know if i converted correctly, if not please try how to convert in .bdf  format using the mentioned tool. This Font has may versions, once you know the settings are correct i will convert all versions

[font test github.zip](https://github.com/olikraus/u8g2/files/4410057/font.test.github.zip)

olikraus, 2020-03-31:
I have renamed your bdf file to "dseg7_classic_regular.bdf". If you need multiple files (and/or if you want to see it in u8g2 once) we should define a shortcut for it, like dseg7cr or something like this.

This is the u8g2 conversion command:
```
bdfconv -v -f 1 -m '32-65000' dseg7_classic_regular.bdf -o u8g2_font_dseg7_classic_regular_all.c  -n u8g2_font_dseg7_classic_regular_all -d ../bdf/helvR12.bdf
```
This is how it looks like:

[image]

Not sure about the chars >127. Is this intended? You need to tell, which glyphs are required.

This is the code, which could be included in your .ino file:
```
/*
  Fontname: DSEG7 Classic
  Copyright: 
  Glyphs: 224/256
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_dseg7_classic_regular_all[2116] U8G2_FONT_SECTION("u8g2_font_dseg7_classic_regular_all") = 
  "\340\0\3\3\4\4\4\4\5\10\13\0\0\13\0\7\0\1B\2\220\10' \5\0\210\22!\5\0\210"
  "\31\42\10#\330\23\21I\0#\13d\230\64Y*\21J\12\0$\15t\210\64\222\210$\70\221\214\42"
  "\0%\16g\230\67\222P$\211<\213$E\6&\14e\230U\261H,&\21I\10'\6!\330\22"
  "\2(\12r\210\63)\222P$\0)\10r\210\23\241H\36*\10\63\310\63\21J\0+\11D\250U"
  "\261\10)\2,\6!\211\22\2-\6\25\332\31\5.\5\0\210\20/\11b\230\62)\222\24\0\60\15"
  "\267\211\31\207\250\326\70 \252\365\20\61\7\261\217\31\225\2\62\14\267\211\71\346\134*\341\314\26\0\63\13"
  "\266\212\31\326\254\346h\352\1\64\14\267\211\31Q]#u@\70\7\65\15\267\211\31\226pv@\35\20"
  "\316b\66\15\267\211\31\226pv@%\252\365\20\67\13\267\211\31\207\250\326\70\70\17\70\15\267\211\31\207"
  "\250\326H%\252\365\20\71\15\267\211\31\207\250\326H\35\20\316b:\11b\250\22r\200\4\0;\7Q"
  "\211\22!\1<\10D\250u\21E\1=\7$\270\25\207\0>\11D\250\25A\31\11\0\77\13d\230"
  "\64\222P,\24K\1@\22x\210X\264`(\62\221\204\42\222PDR\12\3A\15\267\211\31\207\250"
  "\326H%\252k\0B\15\267\211\31\341\334\1\225\250\326C\0C\10f\211\71\326\254\6D\13\267\211\331"
  "y\251D\265\36\2E\12\266\211\31\327\314\326\254\6F\12\266\211\31\327\314\326\134\1G\15\267\211\31\226"
  "pvpT\353!\0H\14\267\211\31\341\334\1\225\250\256\1I\6Q\217\31\5J\13\267\211\331y\7"
  "D\265\36\2K\15\267\211\31\226pv@%\252k\0L\13\266\211\31\321\334\201\321\254\6M\15\267\211"
  "\31\207\250\326\70 \252k\0N\11g\211\71\225\250\256\1O\12g\211\71\225\250\326C\0P\15\267\211"
  "\31\207\250\326H%\234\63\0Q\15\267\211\31\207\250\326H\35\20\316\1R\10f\211\71\326\134\1S\15"
  "\267\211\31\341\334\1u@\70\213\1T\12\266\211\31\321\234\255Y\15U\11W\211\31Q\255\207\0V\15"
  "\267\211\31Q]\343\200\250\326C\0W\15\267\211\31Q]#\225\250\326C\0X\14\267\211\31Q]#"
  "\225\250\256\1Y\15\267\211\31Q]#u@\70\213\1Z\14\267\211\71\346\334\1\341\314\26\0[\6q"
  "\211\22\7\134\11b\230\22\221\244H\2]\7r\210\22\222|^\10\63\310\64\241I\0_\6\25\212\31"
  "\5`\6\22\350\23\2a\15\267\211\31\207\250\326H%\252k\0b\15\267\211\31\341\334\1\225\250\326C"
  "\0c\10f\211\71\326\254\6d\13\267\211\331y\251D\265\36\2e\12\266\211\31\327\314\326\254\6f\12"
  "\266\211\31\327\314\326\134\1g\15\267\211\31\226pvpT\353!\0h\14\267\211\31\341\334\1\225\250\256"
  "\1i\6Q\217\31\5j\13\267\211\331y\7D\265\36\2k\15\267\211\31\226pv@%\252k\0l"
  "\13\266\211\31\321\334\201\321\254\6m\15\267\211\31\207\250\326\70 \252k\0n\11g\211\71\225\250\256\1"
  "o\12g\211\71\225\250\326C\0p\15\267\211\31\207\250\326H%\234\63\0q\15\267\211\31\207\250\326H"
  "\35\20\316\1r\10f\211\71\326\134\1s\15\267\211\31\341\334\1u@\70\213\1t\12\266\211\31\321\234"
  "\255Y\15u\11W\211\31Q\255\207\0v\15\267\211\31Q]\343\200\250\326C\0w\15\267\211\31Q]"
  "#\225\250\326C\0x\14\267\211\31Q]#\225\250\256\1y\15\267\211\31Q]#u@\70\213\1z"
  "\14\267\211\71\346\334\1\341\314\26\0{\10r\210\63)\223\14|\6q\211\22\7}\10r\210\23\241H"
  "\36~\7$\330\65\244\0\177\7b\230\22\207\2\200\5\0\210\20\201\5\0\210\20\202\5\0\210\20\203\5"
  "\0\210\20\204\5\0\210\20\205\5\0\210\20\206\5\0\210\20\207\5\0\210\20\210\5\0\210\20\211\5\0\210"
  "\20\212\5\0\210\20\213\5\0\210\20\214\5\0\210\20\215\5\0\210\20\216\5\0\210\20\217\5\0\210\20\220"
  "\5\0\210\20\221\5\0\210\20\222\5\0\210\20\223\5\0\210\20\224\5\0\210\20\225\5\0\210\20\226\5\0"
  "\210\20\227\5\0\210\20\230\5\0\210\20\231\5\0\210\20\232\5\0\210\20\233\5\0\210\20\234\5\0\210\20"
  "\235\5\0\210\20\236\5\0\210\20\237\5\0\210\20\240\5\0\210\22\241\6Q\211\22\5\242\13d\210T)"
  "\222L\223\20\0\243\12d\230\64\244Hh\24\42\244\11D\250\24\21IDT\245\14d\230\24\241H\212"
  "$b\212\0\246\7q\211\22#\1\247\16t\210\64\242HD\22\212$EB\1\250\6\22\350\23\2\251"
  "\16f\230\66\224\210d\66\223H\42\24\0\252\10C\270\23\242C\0\253\11C\231\24\221\211$\1\254\7"
  "$\270\25\264\0\255\6\22\270\23\2\256\15f\230\66\224\244I\244\62I\241\0\257\6\24\370\24\4\260\13"
  "g\331\31\207\250\326H\5\0\261\11T\230U\261\10)B\262\7\63\310\23\242\2\263\10\63\310\23\242\310"
  "\0\264\6\21\351\23\1\265\11S\211\24\21\225\22\0\266\11t\210\64\225I\376\1\267\6\21\271\22\1\270"
  "\6\21\211\23\1\271\7\62\310\63\223\0\272\7C\270\63)\7\273\11C\231\24\222\210d\2\274\20g\230"
  "\67\261\210(\26\212Fb\21Qh\0\275\17g\230\67\261\210(\26\212\316B\222\320\0\276\17g\230\27"
  "\262P\322$\32\211ED\241\1\277\12T\210T\261PZ(\0\300\15u\230\65\302\240(\22\232\304d"
  "\1\301\15u\230U\301\240(\22\232\304d\1\302\15u\230U\261\231(\22\232\304d\1\303\15u\230\65"
  "\263\240(\22\232\304d\1\304\15u\230\65iAQ$\64\211\311\2\305\15u\230\65\302\240(\22\232\304"
  "d\1\306\16h\230x\264H\64\22\14\215h\61\2\307\15u\210V\242\220\60\32\212\320\42\0\310\11t"
  "\231\25\242Z\254F\311\11t\231\65\241Z\254F\312\11t\231\65\241Z\254F\313\12t\231\25\221H-"
[... cut ...]
  "\22\242H\26\0\360\13d\230\64)\242\210H\24\31\361\11c\231\24\63\212J\0\362\11d\230\64b\222"
  "(\62\363\11d\230Ta\222(\62\364\12d\230T)\21\222(\62\365\11d\230\64S\222(\62\366\12"
  "d\230\64\251$Qd\0\367\11\64\250U\21R\4\0\370\7D\230\64,\5\371\11c\231\24\302\210\12"
  "\1\372\11c\231\64\301\210\12\1\373\10c\231\64)Z\10\374\12c\231\24\221XD\205\0\375\13t\210"
  "T\321PD\244\24\2\376\12s\211\24\241\20ER\2\377\13t\210\64\211\241\210H)\4\0\0\0\4"
  "\377\377\0";
```

The font name for the setFont command would be `u8g2_font_dseg7_classic_regular_all`

olikraus, 2020-03-31:
I played around a little bit with the converter. I restricted the bdf conversion to those glyphs which look like 7-seg chars (`-m '48-57,65-92,97-122'`). I also enforce monospace font (`-b 2`)

```
bdfconv -v -f 1 -b 2 -m '48-57,65-92,97-122' dseg7_classic_regular.bdf -o u8g2_font_dseg7_cr.c  -n u8g2_font_dseg7_cr -d ../bdf/helvR12.bdf
```

This is the result:

[image]

And this is the code:
```
/*
  Fontname: DSEG7 Classic
  Copyright: 
  Glyphs: 64/256
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_dseg7_cr[1110] U8G2_FONT_SECTION("u8g2_font_dseg7_cr") = 
  "@\2\4\3\4\4\1\1\5\11\13\0\0\13\0\7\0\0\252\2{\4\71\60\27\271\347\70\211\12\211\12"
  "\211\12\211\212\215\12\211\12\211\12\211\12\271\0\61\13\271\347\13\314=H`.\0\62\14\271g\61\314M"
  "M`\206&\0\63\14\271g\61\314Me`\66\26\0\64\23\271\347\210\12\211\12\211\12\211\12\211\212\251"
  "\14\314\5\0\65\15\271\347\260\11\314\262\62\60\33\13\0\66\23\271\347\260\11\314\262&*$*$*$"
  "*\344\2\0\67\22\271\347\70\211\12\211\12\211\12\211\212\7\11\314\5\0\70\30\271\347\70\211\12\211\12\211"
  "\12\211\212\251\211\12\211\12\211\12\211\12\271\0\71\23\271\347\70\211\12\211\12\211\12\211\212\251\14\314\306\2"
  "\0A\31\271\347\70\211\12\211\12\211\12\211\212\251\211\12\211\12\211\12\211\12\211\212\0B\22\271\347\10\314"
  "eMTHTHTHT\310\5\0C\13\271\347\307\244&\60C\23\0D\22\271\347\13\314MMT"
  "HTHTHT\310\5\0E\15\271\347\260\11\314\262&\60C\23\0F\14\271\347\260\11\314\262&\60"
  "w\0G\23\271\347\260\11\314\36$*$*$*$*\344\2\0H\23\271\347\10\314eMTHT"
  "HTHTHT\4\0I\11\271\347\307E`.\0J\21\271\347\13\314mTHTHTHT\310"
  "\5\0K\24\271\347\260\11\314\262&*$*$*$*$*\2\0L\14\271\347\10\314=H`\206"
  "&\0M\30\271\347\70\211\12\211\12\211\12\211\212\215\12\211\12\211\12\211\12\211\212\0N\22\271\347\307\244"
  "&*$*$*$*$*\2\0O\21\271\347\307\244&*$*$*$*\344\2\0P\22\271"
  "\347\70\211\12\211\12\211\12\211\212\251\11\314\35\0Q\22\271\347\70\211\12\211\12\211\12\211\212\251\14\314\5"
  "\0R\12\271\347\307\244&\60w\0S\14\271\347\10\314ee`\66\26\0T\14\271\347\10\314eM`"
  "\206&\0U\20\271\347\307\64*$*$*$*\344\2\0V\30\271\347\210\12\211\12\211\12\211\12\211"
  "\212\215\12\211\12\211\12\211\12\271\0W\31\271\347\210\12\211\12\211\12\211\12\211\212\251\211\12\211\12\211\12"
  "\211\12\271\0X\32\271\347\210\12\211\12\211\12\211\12\211\212\251\211\12\211\12\211\12\211\12\211\212\0Y\24"
  "\271\347\210\12\211\12\211\12\211\12\211\212\251\14\314\306\2\0Z\13\271g\61\314m`\206&\0[\11\271"
  "\347G\27\230\357\0\134\14\271\347G\26\230\62\60=\4\0a\31\271\347\70\211\12\211\12\211\12\211\212\251"
  "\211\12\211\12\211\12\211\12\211\212\0b\22\271\347\10\314eMTHTHTHT\310\5\0c\13\271"
  "\347\307\244&\60C\23\0d\22\271\347\13\314MMTHTHTHT\310\5\0e\15\271\347\260\11"
  "\314\262&\60C\23\0f\14\271\347\260\11\314\262&\60w\0g\23\271\347\260\11\314\36$*$*$"
  "*$*\344\2\0h\23\271\347\10\314eMTHTHTHTHT\4\0i\11\271\347\307E`"
  ".\0j\21\271\347\13\314mTHTHTHT\310\5\0k\24\271\347\260\11\314\262&*$*$"
  "*$*$*\2\0l\14\271\347\10\314=H`\206&\0m\30\271\347\70\211\12\211\12\211\12\211\212"
  "\215\12\211\12\211\12\211\12\211\212\0n\22\271\347\307\244&*$*$*$*$*\2\0o\21\271"
  "\347\307\244&*$*$*$*\344\2\0p\22\271\347\70\211\12\211\12\211\12\211\212\251\11\314\35\0"
  "q\22\271\347\70\211\12\211\12\211\12\211\212\251\14\314\5\0r\12\271\347\307\244&\60w\0s\14\271\347"
  "\10\314ee`\66\26\0t\14\271\347\10\314eM`\206&\0u\20\271\347\307\64*$*$*$"
  "*\344\2\0v\30\271\347\210\12\211\12\211\12\211\12\211\212\215\12\211\12\211\12\211\12\271\0w\31\271\347"
  "\210\12\211\12\211\12\211\12\211\212\251\211\12\211\12\211\12\211\12\271\0x\32\271\347\210\12\211\12\211\12\211"
  "\12\211\212\251\211\12\211\12\211\12\211\12\211\212\0y\24\271\347\210\12\211\12\211\12\211\12\211\212\251\14\314"
  "\306\2\0z\13\271g\61\314m`\206&\0\0\0\0\4\377\377\0";
```

olikraus, 2020-03-31:
Just for information: bdfconv handling is mentioned in the u8g2 FAQ https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L233

luxxtek, 2020-03-31:
Thank you for you effort.
I'm not a font specialist and a little overwhelmed by all this information.
So is it correct that the font will not be added to default library fonts but that I will be able to use it?

luxxtek, 2020-03-31:
Thank you so much for your effort and help. I got it woking on my a NodeMCU  I2C OLED 128x32 display.
It is not yet clear to me what the correct workflow is but I will try to figure out how I can do it on my own.
[image]

olikraus, 2020-03-31:
Sieht gut aus :-)

Ah, yes I can add the fonts to u8g2 if you want... I mean, it is not really required for you project, you can just paste the code into your ino file, but I guess you did this already.

You can add the other bdf files here and I will do the conversion.
Maybe you can remove all the chars from the BDF fonts which are not needed. I mean, there are some strange chars in the bdf of your zip file which are not like 7-segment types.

luxxtek, 2020-04-01:
Could you be so kind to convert this one as well?
I cleaned up, hope I didn't touch any vital parts ...
[font test github 2.zip](https://github.com/olikraus/u8g2/files/4413342/font.test.github.2.zip)

olikraus, 2020-04-01:
This is the result. It looks a little bit odd, because some segments are rendered with 1 pixel other with 2 pixel. Maybe you should select a different point size for the vector to bitmap conversion.

[image]

```
/*
  Fontname: DSEG14 Classic
  Copyright: 
  Glyphs: 224/256
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_dseg14_cr[4206] U8G2_FONT_SECTION("u8g2_font_dseg14_cr") = 
  "\340\2\4\2\5\5\1\1\6\20\23\0\377\22\377\20\1\3\177\7\231\20Q \11p\6\77\377\377G\0"
  "!\11p\6\77\377\377G\0\42\27p\6\77&\17\320(\65J\215R\243\324(\65\312P\236\377\15\0"
  "#\36p\6\77\337\15\264\231l&\233\311f\262\231l&\233\311f\262\231l&\233\311\16\244\0$-"
  "p\6\317\340 \225\30\14d\32\245F\251Qj\224\32\245F\31\312\16\16\242!\245F\251Qj\224\32"
  "\245F\251\221\15\16R\203\203\14\0%.p\6\77&\17X\345V\271A(\67\310\350\24\31\235\42\23"
  "\314#\6\7y\200.\23\321e\24\62\215B&J\310R\11Y*!\327c\0&-p\6\317\340 "
  "\67\30\10S\311TR\224\324H\63\322L\236\225.\23\321e\24\62\215B&J\310R\11Y*!\33"
  "\34\244\6\7\31\0'\14p\6\77\245\334\347y\376\67\0(\30p\6\77q\36\220\7\304\315\363\334\344"
  "\1r= \17\310\3\362t\0)\30p\6\77Y\36\220\7\310\365\200< \317M\36\20\67\317\3\362"
  "\324\0**p\6\77YD\221\214(\222\12ER\61\210&\6\321\304:;\70\210\246\23\333\304 \251"
  "\30$\25\212dD\221\214(\362t\0+\21p\6\77\245\334\347\331\301A\64\17\220\373\236\22,\17p"
  "\6\77\377\27y@\334<\17\310S\3-\14p\6\77\377xp\220\347\37\3.\11p\6\77\377\377G"
  "\0/\25p\6\77q\36\220\7\304\315\363\314\363\200\270y\36\220\247\6\60\64p\6\317\340 \225\30\34"
  "\204d\11\221,!\222%D*\205H\245\20\251\42\242\244\36\225\24)b\42EL\64\210\211\6\61\321"
  "N\264\23%\6\7\251\301A\6\61\15p\6\77Z\356\357\11\344\376\36\3\62\32p\6\317\340 \67\70"
  "\210\373\253\301A*\17\220\373<\61\30\350\6\7\31\0\63\30p\6\317\340 \67\70\210\373\253\301A\36"
  " \367\263\301Ajp\220\1\64\33p\6\77&)\22\212\204\42\241H(\22\212\204\242\244jp\220\7"
  "\310\375=\6\65\33p\6\317\340 \225\30\14dr\237\347\21\203\203<@\356g\203\203\324\340 \3\66"
  "%p\6\317\340 \225\30\14dr\237\347\21\203\203TR$\24\11EB\221P$\24\11E\211\301A"
  "jp\220\1\67\35p\6\317\340 \225\30\34\204\204\42\241H(\22\212\204\42\241(\251'\220\373{\14"
  "\0\70.p\6\317\340 \225\30\34\204\204\42\241H(\22\212\204\42\241(\251\32\34\244\222\42\241H("
  "\22\212\204\42\241H(J\14\16R\203\203\14\0\71$p\6\317\340 \225\30\34\204\204\42\241H(\22"
  "\212\204\42\241(\251\32\34\344\1r\77\33\34\244\6\7\31\0:\17p\6\77\317\344z>\312\3\344z"
  "\236\0;\36p\6\77\337\15\264\231l&\233\311f\262\231l&\233\311f\262\231l&\233\311\16\244\0"
  "<\32p\6\77q\36\220\7\304\315\363\314\363\200\270y\36\220\7\34\350\6\7\31\0=\21p\6\77\377"
  "xp\220\347\347\203\201np\220\1>\37p\6\77Y\36\220\7\310\365\200< \317]\36 \327\3\362"
  "\200< \71\30\350\6\7\31\0\77\37p\6\317\340 \225\30\34\204\204\42\241H(\22\212\204\42\241("
  ")\35H\363\0\271\357)\1@-p\6\317\340 \225\30\34\204\204\42\241H(\22\212\204\42\241()"
  "\35\250B!\33#\215\221\306Hc\244\61\322\30%\6\7\251\301A\6A*p\6\317\340 \225\30\34"
  "\204\204\42\241H(\22\212\204\42\241(\251\32\34\244\222\42\241H(\22\212\204\42\241H(J\352\61\0"
  "B+p\6\317\340 \67\70Hj\224\32\245F\251Qj\224\32eH:\220\206\224\32\245F\251Qj"
  "\224\32\245F\66\70H\15\16\62\0C\32p\6\317\340 \225\30\14dr\237\347)\362\0\271\317\23\203"
  "\201np\220\1D*p\6\317\340 \67\70Hj\224\32\245F\251Qj\224\32eH\17\15)\65J"
  "\215R\243\324(\65J\215lp\220\32\34d\0E\35p\6\317\340 \225\30\14dr\237\347\21\203\203"
  "T\36 \367yb\60\320\15\16\62\0F\30p\6\317\340 \225\30\14dr\237\347\21\203\203T\36 "
  "\367y\36\16G$p\6\317\340 \225\30\14dr\237\347a\3UR$\24\11EB\221P$\24\11"
  "E\211\301Ajp\220\1H&p\6\77&)\22\212\204\42\241H(\22\212\204\242\244jp\220J\212"
  "\204\42\241H(\22\212\204\42\241(\251\307\0I\30p\6\317\340 \67\30h\345>\317S\344\1r\337"
  "\16\6\272\301A\6J\34p\6\77Z\356\357QI\221P$\24\11EB\221P$\24%\6\7\251\301"
  "A\6K$p\6\77&\17\220\345d\71YN\245S\351T\301<b \315\3TA\225N\245\223\345"
  "d\71Y.\17\7L\27p\6\77&\17\220\373<O\221\7\310}\236\30\14t\203\203\14\0M.p"
  "\6\77&)Z%D\253\204h\20J\210\6\31\205H\221Q\210\24\231\210(\251G\205B\66F\32#"
  "\215\221\306Hc\244\61J\352\61\0N-p\6\77&)\332\211v\242AL\64\210\211\24\61\221\42&"
  "J\352QI\221*\42R)D*\205H\226\20\311\22\42YB\224\324c\0O,p\6\317\340 \225"
  "\30\34\204\204\42\241H(\22\212\204\42\241(\251G%EB\221P$\24\11EB\221P\224\30\34\244"
  "\6\7\31\0P!p\6\317\340 \225\30\34\204\204\42\241H(\22\212\204\42\241(\251\32\34\244\362\0"
  "\271\317\363p\0Q\61p\6\317\340 \225\30\34\204\204\42\241H(\22\212\204\42\241(\251G%E\252"
  "\210H\245\20\251\24\42YB$K\210d\11Qbp\220\32\34d\0R)p\6\317\340 \225\30\34"
  "\204\204\42\241H(\22\212\204\42\241(\251\32\34\244\362\0UP\245S\351d\71YN\226\313\303\1S"
  "*p\6\317\340 \225\30\14d\353\203\360 \254\10+\302y\304\340 \17\320F\264\12\255B\234\20'"
  "\304\11\331\340 \65\70\310\0T\24p\6\317\340 \67\30h\345>\317S\344\1r\337S\2U(p"
  "\6\77&)\22\212\204\42\241H(\22\212\204\242\244\36\225\24\11EB\221P$\24\11EBQbp"
  "\220\32\34d\0V!p\6\77&\17\220\345d\71YN\245S\351T\301<E\36\240\10+\302\203\360"
  " \274^\347\341\0W.p\6\77&)\322\30i\214\64F\32#\215\221\306\250H\217J\212\24\231\210"
  "H\221Q\210\6\31\205h\20J\210V\11\321*!J\352\61\0X\36p\6\77Y*\231J\212\222\32"
  "iF\232\311\63\317d\63J\215R\224L%Sy:\0Y\27p\6\77Y*\231J\212\222\32iF"
  "\232\311\63\310\3\344\276\247\4Z\36p\6\317\340 \67\30\310\363\200< n\236g\236\7\304\315\363\200"
  "<\340@\67\70\310\0[\36p\6\77\337\15\264\231l&\233\311f\262\231l&\233\311f\262\231l&"
[... 51 lines cut ...]
[... cut ...]
  "\352\11p\6\77\377\377G\0\353\11p\6\77\377\377G\0\354\11p\6\77\377\377G\0\355\11p\6\77"
  "\377\377G\0\356\11p\6\77\377\377G\0\357\11p\6\77\377\377G\0\360\11p\6\77\377\377G\0\361"
  "\11p\6\77\377\377G\0\362\11p\6\77\377\377G\0\363\11p\6\77\377\377G\0\364\11p\6\77\377"
  "\377G\0\365\11p\6\77\377\377G\0\366\11p\6\77\377\377G\0\367\11p\6\77\377\377G\0\370\11"
  "p\6\77\377\377G\0\371\11p\6\77\377\377G\0\372\11p\6\77\377\377G\0\373\11p\6\77\377\377"
  "G\0\374\11p\6\77\377\377G\0\375\11p\6\77\377\377G\0\376\11p\6\77\377\377G\0\377\11p"
  "\6\77\377\377G\0\0\0\0\4\377\377\0";

```

luxxtek, 2020-04-01:
It's OK for me, looks a little strange but it'swhat I needed. 
[image]

olikraus, 2020-04-01:
nice :-)

luxxtek, 2020-04-01:
Great library (helpdesk included!)
Thank you for your effort.

closed 2020-07-05 by olikraus