olikraus/u8g2 #1141: HW SPI on NHD-2.7-12864WDW3

goulart-paul, 2020-04-01:
I am trying to get HW SPI to work on a Newhaven SSD1322 128x64 display ([datasheet](https://www.mouser.co.uk/datasheet/2/291/NHD-2.7-12864WDW3-1129822.pdf)) via an Arduino Nano 33 IoT ([docs](https://store.arduino.cc/arduino-nano-33-iot)).

There are two (possibly related) issues: 

1.  When I use SW SPI, I can write text to the display but the update rate is extremely slow (sub 1 Hz).    Constructor is : `U8G2_SSD1322_NHD_128X64_F_4W_SW_SPI`.

2. When I use HW SPI, I get a blank display.  Constructor is `U8G2_SSD1322_NHD_128X64_F_4W_HW_SPI`.

I also tried 3 wire HW SPI and also got a blank display (not in example below).

I have checked many times and am reasonably certain that I have the the connections to the correct pins on the boards.   

Full example: 
```
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

#define PIN_SCLK      13
#define PIN_SDIN      11
#define PIN_CS        12
#define PIN_DC         6
#define PIN_RES        9

// Works
U8G2_SSD1322_NHD_128X64_F_4W_SW_SPI u8g2(U8G2_R0,PIN_SCLK,PIN_SDIN,PIN_CS,PIN_DC,PIN_RES);

// Does not work (display non-responsive)
//U8G2_SSD1322_NHD_128X64_F_4W_HW_SPI u8g2(U8G2_R0,PIN_CS,PIN_DC,PIN_RES);

void setup(void) {

  pinMode(PIN_RES, OUTPUT);
  digitalWrite(PIN_RES, LOW);
  delay (200);
  digitalWrite(PIN_RES, HIGH);
  delay (200);
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB08_tr);
    u8g2.drawStr(0,20,"Hello World!");
  } while ( u8g2.nextPage() );
}
```

olikraus, 2020-04-02:
Thanks for creating an issue here. 
I assume, that the CS pin must not be 12. If I remember correctly, Pin 12 is the MISO pin of the SPI hardware subsystem. As soon as the SPI hardware is activated, Pin 12 becomes the MISO pin and can not be used for other functiosn. This means, you need to select a different pin as CS pin.

goulart-paul, 2020-04-02:
Changing CS from 12 to 10 solved the problem immediately.    The refresh rate also dropped from ~1100ms to about 35ms.   

Thank you for such a fast reply!

closed 2020-04-02 by goulart-paul

olikraus, 2020-04-02:
I am happy to help here. Let me know if there are other issues.

Just a final note: This problem is specific to your microcontroller (AVR Controller) on your board. It not a bug in u8g2 and would have affected any other SPI device.

loscst, 2024-11-27:
Bonjour,
Je rencontre le même problème mais je suis sur une carte NUCLEOF103RB.
Si vous avez une idée de pourquoi ca ne flnctionne pas, je suis preneur 

#include <Arduino.h>
#include <U8x8lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
 #define PIN_CLK PA5
#define PIN_DATA PA7
#define PIN_CS PA4
#define PIN_DC PA6
#define PIN_RESET PC7

U8X8_SSD1322_NHD_128X64_4W_SW_SPI u8x8(/* clock=*/ PIN_CLK, /* data=*/ PIN_DATA, /* cs=*/ PIN_CS, /* dc=*/ PIN_DC, /* reset=*/ PIN_RESET);
//U8X8_SSD1322_NHD_128X64_4W_HW_SPI u8x8(/* cs=*/ PIN_CS, /* dc=*/ PIN_DC, /* reset=*/ PIN_RESET);

Je souhaite utiliser HW_SPI