olikraus/u8g2 #1143: blocking behaviour of u8g2.nextPage()

dnlwgnd, 2020-04-02:
Hi,

I am using u8g2 on an Arduino Pro Mini and so far I am very impressed. Thank you!

I now ran into a somewhat advanced performance issue in that I found that nextPage() does block until the buffer has been sent completely and take too long in my application. I use HW SPI on a 128x64 Oled display, but still it blocks longer thank I want.

In my usecase I want to receive the entire buffer via a Serial connection (RPi on the other end) because using some of the draw functions (esp. for putting Strings) are even more too slow. I require my main loop to loop at least about every 2ms, since I need to retrieve sensor values from a magenetometer. Unfortunately I must use the display in a paged mode, i can not spare the memory for the full buffer.
Now I manged to send the data I want over the Serial connection, write it to the buffer and call nextPage() to send it to the display. The problem now occures when at the same time as the first page gets sent to the display the next page already arrives on the Serial. 

```C
  while(Serial.available() && pos_in_buffer<buffer_size){
    //reading up to buffer size bytes from Serial into buffer
    *(u8g2.getBufferPtr()+pos_in_buffer++) = Serial.read();
  }
  if(pos_in_buffer == buffer_size){
    //buffer has been filled, send it
    if (!u8g2.nextPage()){
      //all pages have been written
      u8g2.firstPage();
    }
    pos_in_buffer = 0;
  }
```

I would require a nextPage() (or sendBuffer()) function that returns control to the main loop after each single byte was put to the SPI bus, so that I have the chance to read further bytes that arrive over Serial. Currently i loose data since the Serial buffer overflows while nextPage() sends data to the display. Do you have an idea?

Any hints appreciated,
Daniel

olikraus, 2020-04-03:
I would personally say, you have hit Arduino limitations. In your case I assume, you need to read serial data in the background via interrupts. 
In fact in an none-Arduino Environment one would trigger an interrupt once a byte is received from the serial bus. The interrupt service routine would then store the byte in a suitable buffer, taking care on overflow conditions.
Then another parallel process would read from that buffer, taking care on underflow conditions.

Arduino simplifies things and it is not really made for such problems like you described.

dnlwgnd, 2020-04-04:
Thank you for the comment. I slowly come to that realization also. In the meantime i was able to mitigate the issue a little bit by using a slower Serial speed.
I browsed your code a bit, at came upon the concept of tiles. That might be the best unit to send in one go. However I do not quite get the magic happening, especially the link to the underlying u8x8 part. Would it be feasible to extend the nextPage() funktion so that it only sends a tile instead a complete page? Or I thought, maybe I can put the needed bytes to write the display memory myself on the SPI bus?
Anyways, big thanks for your time.
Daniel.

olikraus, 2020-04-04:
> Would it be feasible to extend the nextPage() funktion so that it only sends a tile instead a complete page? 

Hmm.. the benefit is not really clear to me execpt for slowing down everything.

The code behind nextPage is here:

https://github.com/olikraus/u8g2/blob/3d8d1e814481827ecf25af3f57e6b90a8a47bd1e/csrc/u8g2_buffer.c#L116-L133

You can do the sending by yourself.

Refresh is not required in your case and the clear buffer command can be done separatly.

The only remaining update ths the current tile row. 

Most of the functions are also described in the manual.

> However I do not quite get the magic happening, especially the link to the underlying u8x8 part. 

The link are those tiles ;-) The u8g2 buffer is assumed to be a list of tiles and sent to by u8x8 draw tile function. The link is exactly this line: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_buffer.c#L63

dnlwgnd, 2020-04-15:
Thank you for those detail. I will try to make something from it.

closed 2020-04-15 by dnlwgnd