olikraus/u8g2 #1145: Partial refresh on WaveShare 1.54 e-paper, BUSY tracking

trollcop, 2020-04-04:
I've noticed that in the `u8x8_d_ssd1607_ws_200x200` driver, partial / quick refresh sequence is used to update display with `u8x8_d_ssd1607_ws_to_refresh_seq`.

This is all great, however there's commented out UX8_DLY instructions which, if readded, allow larger regions to be refreshed.

I was getting no display updates when drawing larger changes and wondered what was going on, until i hooked up logic analyzer and saw that LCD busy signal kept staying high, while next draw/update instruction was already pushed to panel. With the 250 delays the drawing (usually) succeeds (except for full-screen update, of course), but becomes slow (of course) if just a small region is updated.

Anyway, my current solution is to idle-loop after u8g2_SendBuffer() waiting for BSY flag to clear and yielding time to other tasks to run.

This brings me to the question of why I wrote all this,

1) Is there a mechanism in the command sequences to wait for some GPIO to change state (i.e. hold for BSY pin)
2) Why other panels in the 1607 driver are using very large refresh_seq controls with full LUT upload etc. This one leads me to believe anything other than _ws_ variant of 1607 driver will do entire screen redraw and flicker/etc on each update, is that right?

olikraus, 2020-04-04:
U8g2 was not build for e-paper devices. I tried to support e-paper devices, but i relized, that there are too many differences to graphic OLEDs and LCD. Additionally, there are other excellent libraries with better support: See for example the libs from @ZinggJM based on the adafruit gfx lib. Those libs are also supported by https://github.com/olikraus/U8g2_for_Adafruit_GFX, which means, that you can use all the U8g2 fonts.

So the answers are:
1. No, busy pin is not supported.
2. Yes it does flicker. It some point of time, I stopped working on e-paper and did not do any further optimization.

trollcop, 2020-04-04:
I actually *like* u8g2 C-interface as opposed to all those half-C++ junks you linked.
Noted re points 1+2, I guess I'll deal with my workaround for now, and if I end up upgrading for a different display for production, hack in partial refresh the same way it's done with _ws_ variant.

closed 2020-04-04 by trollcop

olikraus, 2020-04-04:
> I actually like u8g2 C-interface

I am an old C programmer. I also wanted to use u8g2 in my own RAM/ROM limited projects without Arduino IDE (e.g. here: https://drolliblog.wordpress.com/2017/11/02/stm32l031x6-eval-board/)

So the backbone of u8g2/u8x8 is C only including the inheritance of u8g2 from u8x8 and virtual member functions implemented as callback fn-pointer in plain C. All this with a thin layer of C++ for the Arduino people.

In fact I usually separate the (slow) user interaction from the sensor/actor handling.
Reading button inputs and display output is done in some kind of a main loop, while all the sensor things run in background (interrupt service procedures): 
https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/clock/main.c#L937

In fact I like the arm systick as a regular trigger for my background procedures:
https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/clock/main.c#L197