olikraus/u8g2 #1146: u8g2_d_memory.c allocating 128kbytes in ZI/.data/.bss

trollcop, 2020-04-04:
This file looks autogenerated, but I wonder if there's a better way to do what it's doing. Currently, on non-gcc compiler (ARMCC, just FYI), building this file (even in -O3/max optimizations mode) allocates ALL the static buf[] arrays in ram, resulting in ~128k of ram being used during link, and failing on MCUs with less than that.

My solution has been to #if 0 everything except the allocators I need (u8g2_m_25_25_* in my case), and then the build/link succeeds, but I wonder if there's a better way to handle this without depending on a gcc-ism - that it will clear all unused / unlinked statics because the function isn't used/linked.

olikraus, 2020-04-04:
U8g2 was written for Arduino IDE, which inturn depends on the linker garbage collection.

olikraus, 2020-04-04:
I think the --feedback option should do the same for armcc.

trollcop, 2020-04-10:
That doesn't really address my initial comment that there is probably a better way to do this, somehow.

olikraus, 2020-04-10:
> This file looks autogenerated, but I wonder if there's a better way to do what it's doing.

There are always pro and cons. What would be a better approach? Dynamic memory allocation? But then you need code for "malloc()" and friends. But malloc/sbrk will eat up flash memory and also RAM overhead is required.

The current approach moves the memory allocation problem to the linker. Which is the best what I can think of: No extra code and no RAM overhead is required.

closed 2020-07-05 by olikraus