olikraus/u8g2 #1147: what am i doing wrong?

xmodpt, 2020-04-04:
Hello all

I have 2 seperat issues

First:
I have made a simple meu using the "SelectionList" example but i cant make my changes work.
that meens, execute an event when an option is selected.
Code atatched:
`/*

*/

#include <Arduino.h>
#include <U8g2lib.h>
#include <AccelStepper.h>
AccelStepper stepper; // Defaults to AccelStepper::FULL4WIRE (4 pins) on 2, 3, 4, 5

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
// 10000ul =   10 sec
// 30250ul =   30 sec
// 60500ul =   1 min
// 302500ul =  5 min
// 605000ul = 10 min
// 907500ul = 15 min
#define COUNTDOWN_TIME  10000ul // Set timer
#define pinRele 6        //Set relay Pin  
#define nivelRele HIGH   //Relay logic

char
    szString[20];
byte
    mins, secs;
       
unsigned long
    timeTemp,
    timeNow,
    timeStart,
    timeElapsed,
    timeLeft;   

// U8G2 Lib display config
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
// End of constructor list

void setup(void) {
   stepper.setMaxSpeed(1000);
   stepper.setSpeed(50);  
 // RELAY SETTINGS 
     pinMode(pinRele, OUTPUT);
     digitalWrite(pinRele, !nivelRele);
   // Rotary encoder or buttons
  u8g2.begin(/*Select=*/ A2, /*Right/Next=*/ A4, /*Left/Prev=*/ A5, /*Up=*/ A0, /*Down=*/ A1, /*Home/Cancel=*/ A3); // Arduboy DevKit
// Font Type config 
  u8g2.setFont(u8g2_font_6x12_tr);
timeStart = millis();
    mins = 1;
    secs = 1;
}

const char *string_list = 
  "05 min\n"
  "10 min\n"
  "15 min\n"
  "\n"
  "About";

uint8_t current_selection = 1;

void loop(void) {

  current_selection = u8g2.userInterfaceSelectionList(
    "Curing Station V1.0",
    current_selection, 
    string_list);

//  if ( current_selection == 0 ) {
//    u8g2.userInterfaceMessage(
//	"Selection:", 
//	"curing for 05 min?",
//	"",
//	" ok \n cancel ");
//  
//  }
  if ( current_selection == 1 ) {
    u8g2.userInterfaceMessage(
  "Selection:", 
  "curing for 5 min?",
  "",
  " ok \n cancel " ), relay_action();

  }
  
  if ( current_selection == 2 ) {
    u8g2.userInterfaceMessage(
  "Selection:", 
  "curing for 10 min?",
  "",
  " ok \n cancel ");
  }
  if ( current_selection == 3 ) {
    u8g2.userInterfaceMessage(
  "Selection:", 
  "curing for 15 min?",
  "",
  " ok \n cancel ");
   DoCountdown();
   relay_action();
    }
    if ( current_selection == 5 ) {
    u8g2.userInterfaceMessage(
  "About:", 
  "Curing system",
  "design by Xmodpt",
  "ver: 1.0");
    }
}

/*   VOIDS LIST */
void DoCountdown()
{
    static unsigned long
        lastTimeNow = 0;
    static byte
        lastsecs = 1;
       
    timeNow = millis();
    timeElapsed = timeNow - timeStart;
   
    if( mins == 0 && secs == 0 )
        return;
       
    timeLeft = COUNTDOWN_TIME - timeElapsed;

    mins = (byte)(timeLeft / 60000ul);
    timeTemp = timeLeft - (mins * 60000);
    secs = (byte)(timeTemp / 1000ul);
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
//        display.clearDisplay();
//        display.setCursor( 30, 20 );
//        sprintf( szString, "DONE" );
//        display.print( szString );
        Serial.println( szString );       
    }
    else if( secs != lastsecs )
    {
        lastsecs = secs;
//        display.setCursor( 20, 20 );
//        sprintf( szString, "%02d:%02d", mins, secs );
        Serial.println( szString );  
//        display.clearDisplay();
//        display.print( szString );
     }
//   display.display();
}
void stepper_motor()
{  
   if ( mins == 0 && secs == 0 ) { 
    
       stepper.stop();
      
   } else {
       stepper.runSpeed();
   }
}
void relay_action(){
  
  if ( mins == 0 && secs == 0 ) { 
    
      digitalWrite(pinRele, nivelRele);
      
   } else {
      digitalWrite(pinRele, !nivelRele);
   }
  
  }`

Second issue:

I have a countdown timer that works great with the Adafruit lib but i cant seem to make it work using the U8G2 lib ( same as the above menu).
When i make the changes that i "think" ;) that are needed... i get a black/blanc display 
Code attached:

`/****

******/ 

#include <Adafruit_SH1106.h>
#include <Adafruit_GFX.h>
#include <AccelStepper.h>

AccelStepper stepper; // Defaults to AccelStepper::FULL4WIRE (4 pins) on 2, 3, 4, 5

#define OLED_ADDR       0x3C //Oled address
#define OLED_RESET -1 // Oled reset
    Adafruit_SH1106 display(OLED_RESET); //Oled config type
// 10000ul =   10 sec
// 30250ul =   30 sec
// 60500ul =   1 min
// 302500ul =  5 min
// 605000ul = 10 min
// 907500ul = 15 min
#define COUNTDOWN_TIME  10000ul // Set timer
#define pinRele 6        //Set relay Pin  
#define nivelRele HIGH   //Relay logic

char
    szString[20];
byte
    mins, secs;
       
unsigned long
    timeTemp,
    timeNow,
    timeStart,
    timeElapsed,
    timeLeft;   

void setup()
{  
   stepper.setMaxSpeed(1000);
   stepper.setSpeed(50);  
 // RELAY SETTINGS 
     pinMode(pinRele, OUTPUT);
     digitalWrite(pinRele, !nivelRele);
 // initialize and clear display
    display.begin(SH1106_SWITCHCAPVCC, OLED_ADDR);
    display.clearDisplay();
    display.setTextSize(3);
    display.setTextColor(WHITE);    
     
   Serial.begin(9600);
    timeStart = millis();
    mins = 1;
    secs = 1;
}
void loop()
{  
   DoCountdown();
   stepper_motor();
   relay_action();
}
void stepper_motor()
{  
   if ( mins == 0 && secs == 0 ) { 
    
       stepper.stop();
      
   } else {
       stepper.runSpeed();
   }
}
void relay_action(){
  
  if ( mins == 0 && secs == 0 ) { 
    
      digitalWrite(pinRele, nivelRele);
      
   } else {
      digitalWrite(pinRele, !nivelRele);
   }
  
  }

void DoCountdown()
{
    static unsigned long
        lastTimeNow = 0;
    static byte
        lastsecs = 1;
       
    timeNow = millis();
    timeElapsed = timeNow - timeStart;
   
    if( mins == 0 && secs == 0 )
        return;
       
    timeLeft = COUNTDOWN_TIME - timeElapsed;

    mins = (byte)(timeLeft / 60000ul);
    timeTemp = timeLeft - (mins * 60000);
    secs = (byte)(timeTemp / 1000ul);
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
        display.clearDisplay();
        display.setCursor( 30, 20 );
        sprintf( szString, "DONE" );
        display.print( szString );
        Serial.println( szString );       
    }
    else if( secs != lastsecs )
    {
        lastsecs = secs;
        display.setCursor( 20, 20 );
        sprintf( szString, "%02d:%02d", mins, secs );
        Serial.println( szString );  
        display.clearDisplay();
        display.print( szString );
     }
   display.display();
}`

Can anyone tell me what the hell am i doing wrong?
The objective is to merge the 2 codes together to work as 1.

xmodpt, 2020-04-04:
Second issue:

I have a countdown timer that works great with the Adafruit lib but i cant seem to make it work using the U8G2 lib ( same as the above menu).
When i make the changes that i "think" ;) that are needed... i get a black/blanc display 
Code attached:

```/****

******/ 

#include <Adafruit_SH1106.h>
#include <Adafruit_GFX.h>
#include <AccelStepper.h>

AccelStepper stepper; // Defaults to AccelStepper::FULL4WIRE (4 pins) on 2, 3, 4, 5

#define OLED_ADDR       0x3C //Oled address
#define OLED_RESET -1 // Oled reset
    Adafruit_SH1106 display(OLED_RESET); //Oled config type
// 10000ul =   10 sec
// 30250ul =   30 sec
// 60500ul =   1 min
// 302500ul =  5 min
// 605000ul = 10 min
// 907500ul = 15 min
#define COUNTDOWN_TIME  10000ul // Set timer
#define pinRele 6        //Set relay Pin  
#define nivelRele HIGH   //Relay logic

char
    szString[20];
byte
    mins, secs;
       
unsigned long
    timeTemp,
    timeNow,
    timeStart,
    timeElapsed,
    timeLeft;   

void setup()
{  
   stepper.setMaxSpeed(1000);
   stepper.setSpeed(50);  
 // RELAY SETTINGS 
     pinMode(pinRele, OUTPUT);
     digitalWrite(pinRele, !nivelRele);
 // initialize and clear display
    display.begin(SH1106_SWITCHCAPVCC, OLED_ADDR);
    display.clearDisplay();
    display.setTextSize(3);
    display.setTextColor(WHITE);    
     
   Serial.begin(9600);
    timeStart = millis();
    mins = 1;
    secs = 1;
}
void loop()
{  
   DoCountdown();
   stepper_motor();
   relay_action();
}
void stepper_motor()
{  
   if ( mins == 0 && secs == 0 ) { 
    
       stepper.stop();
      
   } else {
       stepper.runSpeed();
   }
}
void relay_action(){
  
  if ( mins == 0 && secs == 0 ) { 
    
      digitalWrite(pinRele, nivelRele);
      
   } else {
      digitalWrite(pinRele, !nivelRele);
   }
  
  }

void DoCountdown()
{
    static unsigned long
        lastTimeNow = 0;
    static byte
        lastsecs = 1;
       
    timeNow = millis();
    timeElapsed = timeNow - timeStart;
   
    if( mins == 0 && secs == 0 )
        return;
       
    timeLeft = COUNTDOWN_TIME - timeElapsed;

    mins = (byte)(timeLeft / 60000ul);
    timeTemp = timeLeft - (mins * 60000);
    secs = (byte)(timeTemp / 1000ul);
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
        display.clearDisplay();
        display.setCursor( 30, 20 );
        sprintf( szString, "DONE" );
        display.print( szString );
        Serial.println( szString );       
    }
    else if( secs != lastsecs )
    {
        lastsecs = secs;
        display.setCursor( 20, 20 );
        sprintf( szString, "%02d:%02d", mins, secs );
        Serial.println( szString );  
        display.clearDisplay();
        display.print( szString );
     }
   display.display();
}```

Can anyone tell me what the hell am i doing wrong?
The objective is to merge the 2 codes together to work as 1.

xmodpt, 2020-04-04:
Can anyone tell me what the hell am i doing wrong?
The objective is to merge the 2 codes together to work as 1.

olikraus, 2020-04-05:
First: I suggest to use `U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);` instead of `U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);` (F instead of 1), this is more compatible with Adafruit GFX.

Second: 
It does not make sense to remove the Adadruit `//   display.display();` without replacing this command with the corresponding u8g2 command. Use https://github.com/olikraus/u8g2/wiki/u8g2reference#sendbuffer

xmodpt, 2020-04-05:
hello m8 thank you for the help but i am still stuck

on my second issue i still cant seem to get any type of image on my oled.
If i use the default adafruit sh1106 + adafruite gfx.... i get image and countdown as needed

if i change te code to use u8g2 and make the changes.... i cant get any image, just a blanck screen.

````
/****

******/ 

/**** ADAFRUIT LIB ******/
//#include <Adafruit_SH1106.h>
//#include <Adafruit_GFX.h>
/*************************/

/****** U8G2 LIB *********/
#include <U8g2lib.h>
/***** U8G2 settings *****/
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
/*************************/
#include <AccelStepper.h> // stepper motor lib 

AccelStepper stepper; // Defaults to AccelStepper::FULL4WIRE (4 pins) on 2, 3, 4, 5
/**** U8G2 Display settings *****/
//U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
/********************************/

/** ADAFRUITE Display settings **/
//#define OLED_ADDR       0x3C //Oled address
//#define OLED_RESET -1 // Oled reset
//    Adafruit_SH1106 display(OLED_RESET); //Oled config type
/*********************************/

/****** Time config list ********/
// 10000ul =   10 sec
// 30250ul =   30 sec
// 60500ul =   1 min
// 302500ul =  5 min
// 605000ul = 10 min
// 907500ul = 15 min
#define COUNTDOWN_TIME  10000ul // Set timer
#define pinRele 6        //Set relay Pin  
#define nivelRele HIGH   //Relay logic

char
    szString[20];
byte
    mins, secs;
       
unsigned long
    timeTemp,
    timeNow,
    timeStart,
    timeElapsed,
    timeLeft;   

void setup()
{  
   stepper.setMaxSpeed(1000);
   stepper.setSpeed(50);  
 // RELAY SETTINGS 
     pinMode(pinRele, OUTPUT);
     digitalWrite(pinRele, !nivelRele);
 // initialize and clear display
    u8g2.begin(/*Select=*/ A2, /*Right/Next=*/ A4, /*Left/Prev=*/ A5, /*Up=*/ A0, /*Down=*/ A1, /*Home/Cancel=*/ A3); // Arduboy DevKit
    u8g2.setFont(u8g2_font_6x12_tr);
    u8g2.clearDisplay();
  
   Serial.begin(9600);
    timeStart = millis();
    mins = 1;
    secs = 1;
}
void loop()
{  
   DoCountdown();
   stepper_motor();
   relay_action();
}

/****** VOIDS ********/
void stepper_motor()
{  
   if ( mins == 0 && secs == 0 ) { 
    
       stepper.stop();
      
   } else {
       stepper.runSpeed();
   }
}
void relay_action(){
  
  if ( mins == 0 && secs == 0 ) { 
    
      digitalWrite(pinRele, nivelRele);
      
   } else {
      digitalWrite(pinRele, !nivelRele);
   }
  
  }

void DoCountdown()
{
    static unsigned long
        lastTimeNow = 0;
    static byte
        lastsecs = 1;
       
    timeNow = millis();
    timeElapsed = timeNow - timeStart;
   
    if( mins == 0 && secs == 0 )
        return;
       
    timeLeft = COUNTDOWN_TIME - timeElapsed;

    mins = (byte)(timeLeft / 60000ul);
    timeTemp = timeLeft - (mins * 60000);
    secs = (byte)(timeTemp / 1000ul);
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
        u8g2.clearDisplay();
        u8g2.setCursor( 30, 20 );
        sprintf( szString, "DONE" );
        u8g2.print( szString );
        Serial.println( szString );       
    }
    else if( secs != lastsecs )
    {
        lastsecs = secs;
        u8g2.setCursor( 10, 10 );
        sprintf( szString, "%02d:%02d", mins, secs );
        Serial.println( szString );  
        u8g2.clearDisplay();
        u8g2.print( szString );
     }
   u8g2.clearBuffer();
   //display.display();
}
````

olikraus, 2020-04-05:
Again: use sendBuffer!

xmodpt, 2020-04-05:
i have tryed on the loop

````
void loop()
{  
  u8g2.sendBuffer();
   DoCountdown();
   stepper_motor();
   relay_action();
   u8g2.clearBuffer();
} 
````

No image

and in the void counter and still no image

````
void DoCountdown()
{
   u8g2.clearBuffer();
    static unsigned long
        lastTimeNow = 0;
    static byte
        lastsecs = 1;
       
    timeNow = millis();
    timeElapsed = timeNow - timeStart;
   
    if( mins == 0 && secs == 0 )
        return;
       
    timeLeft = COUNTDOWN_TIME - timeElapsed;

    mins = (byte)(timeLeft / 60000ul);
    timeTemp = timeLeft - (mins * 60000);
    secs = (byte)(timeTemp / 1000ul);
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
        u8g2.clearDisplay();
        u8g2.setCursor( 30, 20 );
        sprintf( szString, "DONE" );
        u8g2.print( szString );
        Serial.println( szString );       
    }
    else if( secs != lastsecs )
    {
        lastsecs = secs;
        u8g2.setCursor( 10, 10 );
        sprintf( szString, "%02d:%02d", mins, secs );
        Serial.println( szString );  
        u8g2.clearDisplay();
        u8g2.print( szString );
     }
   
   u8g2.sendBuffer();
   //display.display();
}
````

xmodpt, 2020-04-05:
and i have tryed the seperated , one at a time and both together and still nothing :(

As noob as i can be :(

xmodpt, 2020-04-05:
Got it to show image by changing the display type to "F" but it blinks for every counting sec
Expl:  show 00:05 -> blinks -> 00:04 -> blinks -> 00:03 -> blinks...etc

xmodpt, 2020-04-05:
GOT IT !!! :)

i had an u8g2.clearBuffer(); that was wrong.

it works now !!!!

in  regards to the "SelectionList" how can i get code to execute when an option is selected ?

xmodpt, 2020-04-05:
now me again :|

when i go to me code of the SelectionList and change the display from "1" to "F" the list get's "cutted" and the menus go all funny . so it works greate with the "1" version

in the countdown timer if i use the "F" option, it look's great but if i use the "1" option i get NO image....

im sum
I have 2 codes that work greate seperated and when the are combined... none works at all.

olikraus, 2020-04-05:
> in regards to the "SelectionList" how can i get code to execute when an option is selected ?

Just take the return value and act on that result:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/SelectionList/SelectionList.ino

> when i go to me code of the SelectionList and change the display from "1" to "F" the list get's "cutted" and the menus go all funny . so it works greate with the "1" version

I do not know your code, but the same example above demonstrates, that F constructor will work nicely.

The example are also there in the Arduino IDE. Please test by yourself.

xmodpt, 2020-04-05:
in regards to the to the display i solved it.... had a "," @ the wrong place- SOLVED
in regards to the selection 
i have 3 voids
DoCountdown()
stepper_motor()
relay_action()
 
and it i select 5min(on the menu) it should start DoCountdown() and the code for the countdown but i can seem to find out how to get it to work or were to put/call it.
Note: regular menu uses _u8g2.setFont(u8g2_font_6x12_tr);_ and the countdown uses _u8g2.setFont(u8g2_font_helvB24_tf );_

```
/*

*/

#include <Arduino.h>
#include <U8g2lib.h>
#include <AccelStepper.h>
AccelStepper stepper; // Defaults to AccelStepper::FULL4WIRE (4 pins) on 2, 3, 4, 5

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
// 10000ul =   10 sec
// 30250ul =   30 sec
// 60500ul =   1 min
// 302500ul =  5 min
// 605000ul = 10 min
// 907500ul = 15 min
#define COUNTDOWN_TIME  10000ul // Set timer
#define pinRele 6        //Set relay Pin  
#define nivelRele HIGH   //Relay logic

char
    szString[20];
byte
    mins, secs;
       
unsigned long
    timeTemp,
    timeNow,
    timeStart,
    timeElapsed,
    timeLeft;   

// U8G2 Lib display config
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
// End of constructor list

void setup(void) {
   stepper.setMaxSpeed(1000);
   stepper.setSpeed(50);  
 // RELAY SETTINGS 
     pinMode(pinRele, OUTPUT);
     digitalWrite(pinRele, !nivelRele);
   // Rotary encoder or buttons
  u8g2.begin(/*Select=*/ A2, /*Right/Next=*/ A4, /*Left/Prev=*/ A5, /*Up=*/ A0, /*Down=*/ A1, /*Home/Cancel=*/ A3); // Arduboy DevKit
// Font Type config 
  u8g2.setFont(u8g2_font_6x12_tr);
    timeStart = millis();
    mins = 1;
    secs = 1;
  u8g2.clearDisplay();  
}

const char *string_list = 
  "05 min\n"
  "10 min\n"
  "15 min\n"
  "\n"
  "About";

uint8_t current_selection = 1;

void loop(void) {

  current_selection = u8g2.userInterfaceSelectionList(
    "Curing Station V1.0",
    current_selection, 
    string_list);

  if ( current_selection == 1 ) {
    u8g2.userInterfaceMessage(
  "Selection:", 
  "curing for 5 min?",
  "",
  " ok \n cancel " );
}
  
  if ( current_selection == 2 ) {
    u8g2.userInterfaceMessage(
  "Selection:", 
  "curing for 10 min?",
  "",
  " ok \n cancel ");
  }
  if ( current_selection == 3 ) {
    u8g2.userInterfaceMessage(
  "Selection:", 
  "curing for 15 min?",
  "",
  " ok \n cancel ");
    }
    if ( current_selection == 5 ) {
    u8g2.userInterfaceMessage(
  "About:", 
  "Curing system",
  "design by Xmodpt",
  "ver: 1.0");
    }
}
/****** VOIDS ********/
void stepper_motor()
{  
   if ( mins == 0 && secs == 0 ) { 
    
       stepper.stop();
      
   } else {
       stepper.runSpeed();
   }
}
void relay_action(){
  
  if ( mins == 0 && secs == 0 ) { 
    
      digitalWrite(pinRele, nivelRele);
      
   } else {
      digitalWrite(pinRele, !nivelRele);
   }
  
  }

void DoCountdown()
{
   u8g2.setFont(u8g2_font_helvB24_tf );
   //u8g2.clearBuffer();
    static unsigned long
        lastTimeNow = 0;
    static byte
        lastsecs = 1;
       
    timeNow = millis();
    timeElapsed = timeNow - timeStart;
   
    if( mins == 0 && secs == 0 )
        return;
       
    timeLeft = COUNTDOWN_TIME - timeElapsed;

    mins = (byte)(timeLeft / 60000ul);
    timeTemp = timeLeft - (mins * 60000);
    secs = (byte)(timeTemp / 1000ul);
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
        u8g2.clearDisplay();
        u8g2.setCursor( 15, 45 );
        sprintf( szString, "DONE" );
        u8g2.print( szString );
        Serial.println( szString );       
    }
    else if( secs != lastsecs )
    {
        lastsecs = secs;
        u8g2.setCursor( 20,45 );
        sprintf( szString, "%02d:%02d", mins, secs );
        Serial.println( szString );  
        u8g2.clearDisplay();
        u8g2.print( szString );
     }
   
   u8g2.sendBuffer();
 }
````

olikraus, 2020-04-05:
What about this:
``` C
if ( current_selection == 1 ) 
{
   DoCountdown()
}
```

xmodpt, 2020-04-05:
it doesn't work

if i select 5m it blinks something in the back and retutns to the menu

olikraus, 2020-04-05:
Does the u8g2 example work? If yes, then it seems to be a problem of your code.

xmodpt, 2020-04-05:
of that i have no dought.
both codes work great... if executedon their won.
When i combine them toguether ... then i have the problem and i can't seem to figure it out.

Menu.ino with the merge code
```
/*

*/

#include <Arduino.h>
#include <U8g2lib.h>
#include <AccelStepper.h>
AccelStepper stepper; // Defaults to AccelStepper::FULL4WIRE (4 pins) on 2, 3, 4, 5

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
// 10000ul =   10 sec
// 30250ul =   30 sec
// 60500ul =   1 min
// 302500ul =  5 min
// 605000ul = 10 min
// 907500ul = 15 min
#define COUNTDOWN_TIME  10000ul // Set timer
//#define pinRele 6        //Set relay Pin  
//#define nivelRele HIGH   //Relay logic

char
    szString[20];
byte
    mins, secs;
       
unsigned long
    timeTemp,
    timeNow,
    timeStart,
    timeElapsed,
    timeLeft;   

// U8G2 Lib display config
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
// End of constructor list

const char *string_list = 
  "05 min\n"
  "10 min\n"
  "15 min\n"
  "\n"
  "About";

uint8_t current_selection = 1;

void setup(void) {
 //  stepper.setMaxSpeed(1000);
 //  stepper.setSpeed(50);  
 // RELAY SETTINGS 
//     pinMode(pinRele, OUTPUT);
     
   // Rotary encoder or buttons
    u8g2.begin(/*Select=*/ A2, /*Right/Next=*/ A4, /*Left/Prev=*/ A5, /*Up=*/ A0, /*Down=*/ A1, /*Home/Cancel=*/ A3); // Arduboy DevKit
  // Font Type config 
    u8g2.setFont(u8g2_font_6x12_tr);
    u8g2.clearDisplay();
    Serial.begin(9600); 
    timeStart = millis();
    mins = 1;
[... 202 lines cut ...]
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
        u8g2.clearDisplay();
        u8g2.setCursor( 15, 45 );
        sprintf( szString, "DONE" );
        u8g2.print( szString );
        Serial.println( szString );       
    }
    else if( secs != lastsecs )
    {
        lastsecs = secs;
        u8g2.setCursor( 20,45 );
        sprintf( szString, "%02d:%02d", mins, secs );
        Serial.println( szString ); 
        u8g2.clearBuffer(); 
        //u8g2.clearDisplay();
        u8g2.print( szString );
     }
   
   u8g2.sendBuffer();
   //display.display();
}
```

xmodpt, 2020-04-05:
by the looks of it i must be missing some kind of pause or delay before returning back to the menu.

the code runs, 1 selec 5 min , it blinks with the counter once and it returns to the menu

olikraus, 2020-04-05:
Yes, I assume you need to wait until your counting is finished.

xmodpt, 2020-04-05:
yes but how ?!?!?! :(
if i get a delay in there... it stops and goes back to the main menu

olikraus, 2020-04-05:
Use a finite state machine (ps://en.wikipedia.org/wiki/Finite-state_machine) with two states: One for the UI and another state for counting.

xmodpt, 2020-04-05:
Now you got me.... I don't have a clue of what that is or how to do it...

olikraus, 2020-04-05:
Well... it is not a u8g2 problem... it is a programming method. Often used in embedded programming.

1. Introduce a state variable: `int state;`
2. Define a constant value for each state, say
```
#define UI 0
#define COUNTER 1
```
3. Build your program

``` C
loop()
{
  switch( state )
  {
    case UI:
      break;
    case COUNTER:
      break;
  }
}
```

Now you can jump between the states, by assigning a different value to `state`. This is a little bit boring with two states, but in your case, you would just do a `state = COUNTER;` to jump from UI to COUNTER. If course you will do this only, if user has selected the counter:

``` C
loop()
{
  switch( state )
  {
    case UI:
      ...
      if ( condition_for_couinting_is_true )
        state = COUNTER;
      break;
      ...
    case COUNTER:
      ...
      if ( counting_is_done )
        state = UI;
      ...
      break;
  }
}
```

Now you can think about additional questions. Do you need to be able to abort counting?
Maybe if the user presses a button?  

``` C
loop()
{
  switch( state )
  {
    case UI:
      ...
      if ( condition_for_couinting_is_true )
        state = COUNTER;
      break;
      ...
    case COUNTER:
      ...
      if ( user_has_pressed_a_button )
        state = UI;
      if ( counting_is_done )
        state = UI;
      ...
      break;
  }
}
```

You get the idea? So if nothing happens, then you do not need to assign anything to "state". If something important happens, you change "state" and the software jumps to a different place to do different things.

Here is one more idea: Maybe you want to wait for 2 seconds until you start counting.
Let us introduce a third state: "WAITCNT". Then it will look like this:

``` C
long untilms;
loop()
{
  switch( state )
  {
    case UI:
      ...
      if ( condition_for_couinting_is_true ) {
        state = WAITCNT;
        untilms = millis() + 2000; // calculate target time for the next state
        print_to_screen_that_the_counter_starts_soon();
      }
      break;
      ...
    case WAITCNT:
      if ( millis() > untilms )
        state = COUNTER
      break;
    case COUNTER:
      ...
      if ( user_has_pressed_a_button )
        state = UI;
      if ( counting_is_done )
        state = UI;
      ...
      break;
  }
}
```

Can you see how easy it is to wait for 2 seconds until your real action starts?

Hope you got the idea. Students usually spend years on this...

xmodpt, 2020-04-05:
WOW ... ok ... sorry about my english... but WTF.... it does make sence.
so i can use the "switch( state )" func to solve this problem... now lets see how to implement it.

one last question
Using U8g2 am getting 93% user warning. is there a better way to do it ?

Thank you allot
again .... you are a great help

olikraus, 2020-04-05:
Maybe one more thing. Usually you have two things: Setup + Doing. So for your counter, you need to do kind of a setup and then you have to increment/decrement your counter until your end condition is reached:

The setup is usually done in the previous state and the actual action (the doing) is done in the current state:

```
long untilms;
loop()
{
  switch( state )
  {
    case UI:
      ...
      if ( condition_for_counting_is_true ) {
        state = WAITCNT;
        untilms = millis() + 2000; // calculate target time for the next state
        print_to_screen_that_the_counter_starts_soon();
      }
      break;
      ...
    case WAITCNT:
      if ( millis() > untilms ) {
        state = COUNTER;
        setup_counter();
      }
      break;
    case COUNTER:
      do_count();
      if ( user_has_pressed_a_button )
        state = UI;
      if ( counting_is_done )
        state = UI;
      ...
      break;
  }
}
```

olikraus, 2020-04-05:
> Using U8g2 am getting 93% user warning. is there a better way to do it ?

use the "1" constructor instead of "F" :-D

But this requires to learn about the first/next loop (picture loop).

xmodpt, 2020-04-05:
again 

TAHNK YOU !!! 👍

closed 2020-04-05 by xmodpt