olikraus/u8g2 #1149: U8G2 class could have virtual destructor

RotateAt60MPH, 2020-04-06:
Hello!

Nice job on this library.  One small suggestion.  Could there be a virtual destructor added on the U8G2 class?  It should be a simple addition to U8g2lib.h:

virtual ~U8G2(void);

This will allow those who to interact with the library to use a generic U8G2 * instead of having a) to define the library instance only in a global, and b) it makes it easier if you change displays since the calling code only deals with &U8G2.

From what I can tell, all your display-specific derived classes just call the base constructor, so why not have the ability to generic the display, like using an interface?

Thanks!

olikraus, 2020-04-06:
All you points are already possible. Nevertheless, I think we can add the virtual destruct as long as the impact on the code size is not so huge.

RotateAt60MPH, 2020-04-06:
And I guess this could apply to U8X8, as well?  :)

You mention that all points are already possible.  They are possible, but the compiler warning will always be there.

olikraus, 2020-04-06:
yes...