olikraus/u8g2 #1150: ESP32 u8g2.sendBuffer()  conflicts IRAM_ATTR onTimer()

blotfi, 2020-04-07:
Dear all
if I put the code

`    u8g2.clearBuffer();  
    u8g2.setCursor(0, 32);
    u8g2.print("U=");   
    u8g2.print(Umot);   
    u8g2.sendBuffer();`
in a 20 ms Timer interrupt  ISR
IRAM_ATTR onTimer()
Nothing is displayed on the OLED display :
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // (rotation, [reset [, clock, data]])

and if I put it in the loop(), it works, but from time to time the display freeze

can you tell me if the I2C or any internal routine called by u8g2 interfer with ISR Timer
thanks

olikraus, 2020-04-07:
U8g2 does not use any specific isr and only calls standard Arduino procedures.

blotfi, 2020-04-07:
do you have an idea of the time to execute a 
clearBuffer()  and  sendBuffer()
on an ESP32 for an OLED 128x64 with SSD1306 and HW I2C?
So to see if the ISR is overrun

olikraus, 2020-04-08:
Hmm.. if we assume 400kHz
Then the time would be
1/400000*128*64/8 = 25ms

there will be some overhead and also this is split into transfers of not more than 32 bytes (due to the Arduino IDE), so it might be 50ms.

blotfi, 2020-04-08:
Thanks for the estimation, it can help many users, I am sure
So this why it can't fit in an ISR of 20 ms
But if I put it in the main loop and it becomes interrupted while being sending the buffer, (in an I2C transfer for example), this will corrupt the display
It works like 95% of time but not 100% 
Any idea how to fix this?

olikraus, 2020-04-08:
You could stop ist during transmission.

blotfi, 2020-04-08:
no, I can't stop the ISR, it is mandatory for sampling and control that the ISR period is respected.
but if I can split the OLED Buffer transfert in many pieces, it will solve the issue
exemple sending the buffer in 4 or 8 pieces and not at once (the sampling period for the display is 1 second so I have plenty of 20 ms period to do it)
do you have a workaround in this way?

olikraus, 2020-04-09:
Here are some more suggestions: For testing you could switch to SW emulated I2C: U8G2_SSD1306_128X64_NONAME_F_SW_I2C.

If this works, then I assume there is something wrong with the I2C implementation in your Arduino IDE.

There are several issues with ISR and ESP32: https://github.com/espressif/arduino-esp32/issues
Maybe there is still some issue. It also seems, that there are some constraints in using ISRs and Wire.

All in all, I do not know what I can do here. As mentioned, U8g2 does not use interrupts and just calls functions from Wire.h in your case.

blotfi, 2020-04-10:
Yes indeed, if I use SW I2C it works better (of course only in the main loop, not in the ISR since it takes a lot of time)
but I don't have the sporadic resets of the ESP32
Thanks

closed 2020-04-10 by blotfi