olikraus/u8g2 #1151: Request: JLX128128G-338

32teeth, 2020-04-08:
Looking for some help connecting this COG (Chip on Glass) 128 x 128 Display

* [Datasheet](http://www.jlxlcd.cn/upload/file/2017912183136.pdf)
* [Product Page](https://www.aliexpress.com/item/32817070261.html?spm=a2g0s.9042311.0.0.7d734c4d4KvRYI)

I have 4-wire SPI version

Would love to get this up and running but seems i can't figure out the naming convention of pins if i use the LiquidCrystal library.

Any insight would be appreciated

[image]

olikraus, 2020-04-08:
I will try to reactivate the code. Hope this device will work for you: #164

olikraus, 2020-04-08:
Constructor should be U8X8_UC1617_JLX128128_4W_SW_SPI as an example (all other constructors are also available)

Maybe you can test this and send me some pics (if there is something visible)

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

tunaftlmk, 2020-04-09:
Hello
I use the library too but XBM sample code is working but the picture I created in my bitmap is not visible.

32teeth, 2020-04-09:
Will test and report back
Thank you

olikraus, 2020-04-09:
@tunaftlmk Moved your question to #1152

32teeth, 2020-04-09:
@olikraus 

Do you have an [sample code](https://github.com/olikraus/u8g2/issues/1151#issuecomment-611136294) for this?

Am i meant to create **U8X8_UC1617_JLX128128_4W_SW_SPI** or does it exist?

olikraus, 2020-04-09:
It should exist in the beta release. You can use any u8x8 example and add the constructor there.

32teeth, 2020-04-09:
This is what i have
[image]

olikraus, 2020-04-09:
yes, looks like SPI. Example code could be this (see also the u8x8 examples in the Arduino IDE).
The pins need to be defined according to your wiring.

``` C++
//U8X8_UC1617_JLX128128_2ND_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8X8_UC1617_JLX128128_2ND_4W_HW_SPI u8x8(/* cs=*/ 6, /* dc=*/ 4, /* reset=*/ 12);
void setup(void)
{ 
  u8x8.begin();
  u8x8.setPowerSave(0); 
}

void loop(void)
{
  u8x8.setFont(u8x8_font_chroma48medium8_r);
  u8x8.drawString(0,0,"Hello World!");
  delay(2000);
}
```

olikraus, 2020-04-09:
The "dc" pin in U8g2 refers to A0(RS) pin on your display.

tunaftlmk, 2020-04-10:
Does Gımp produce G code?

olikraus <notifications@github.com>, 9 Nis 2020 Per, 23:34 tarihinde şunu
yazdı:

olikraus, 2020-04-10:
Deleted and answered here: https://github.com/olikraus/u8g2/issues/1152

olikraus, 2020-04-15:
One issue could be the missing level shifter. Gpio output of the Arduino is 5v, but the display only accepts 3.3v. no sure, whether this will work.

32teeth, 2020-04-15:
Does the ```8x8``` code work?

olikraus, 2020-04-15:
u8g2 is build on top of u8x8. So the answer is: I do not know, but if u8x8 works, then also u8g2 will work. Indeed if the SW does not work, then the u8x8 needs to be fixed. But the problem is: We do not know exactly whether it's a HW or SW problem.

This is my current summary:
Your  hardware might not work, because of two reasons:
- There is no level shifter between the 5V Arduino and the 3.3V display (the display might be destroyed already)
- Your hardware is assembled on a solderless breadboard. Such a protoboard is very unrelibale and my cause wiring problems, which are very hard to detect.

On the other side the software never got tested. I think I have a similar display in my lab, and as far as I remember I also was not able to activate it. Whether I had a SW problem or HW problem is also unknown.  Due to missing time, I have up the display in 2017.

Here is my plan: I will try to find the display in my lab. I will do another attempt to make it work. 
What I can not tell is any timeline on this. Although lock down, I still have to do a lot of different other things... so u8g2 (as a spare time project) has to wait...

olikraus, 2020-04-18:
Ok, found the displays in my lab...

[image]

olikraus, 2020-04-18:
I made some progress, but whatever code was there is very much incomplete :-(

olikraus, 2020-04-19:
Good news: Software should work now.
Constructor prefix: U8X8_UC1617_JLX128128 (same as above)
I have created u8g2 release 2.28.4

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2020-04-19:
[image]

The natural orientation is rotated by 90 degree. So use U8G2_R1 or U8G2_R3 if required.

tunaftlmk, 2020-04-19:
> [image:

thank you, i will be back soon

olikraus <notifications@github.com>, 19 Nis 2020 Paz, 10:41 tarihinde şunu
yazdı:

32teeth, 2020-04-19:
Amazing work @olikraus 
any chance you can share your wiring?

olikraus, 2020-04-19:
[image]

All five signals (clk, data, cs, dc and reset) are routed via an 74HC4050 (the IC in the middle).
The pinout is the same for all the JLX display, so I reused my old protoboard here. The only difference is, that D5..D7 must not be connected (this is different for other JLX displays). Because of this, I have added to switches.

32teeth, 2020-04-19:
Thank you
Seems that the one thing i am missing is the 74HC4050 level shifter
From the image it just seems you are using them straight forward (judging from color)

olikraus, 2020-04-20:
Yes, Arduino outputs are inputs of the 4050. Output of the 4050 go to the display.

32teeth, 2020-04-21:
Okay, awesome.
I have gotten it to work, only need to set rotation

OOC, is there a U8g2 version as well as u8x8

olikraus, 2020-04-21:
Yes, all constructor calls are there. Not sure whether I updated the wiki.

32teeth, 2020-04-21:
> Yes, all constructor calls are there. Not sure whether I updated the wiki.

This should also work on **Adafruit Feather Huzzah** _(ESP32 4MB)_
The U8g2 compiles to big for standard _328_ or _32u4_

32teeth, 2020-04-22:
Got it going on a 4MB ESP32

[image]

Thank you very much for your help!

olikraus, 2020-04-25:
Nice, thanks for sharing the picture.

closed 2020-04-25 by olikraus