olikraus/u8g2 #1152: Hello

olikraus, 2020-04-09:
Hello
I use the library too but XBM sample code is working but the picture I created in my bitmap is not visible.I wonder why it may be the problem

_Originally posted by @tunaftlmk in https://github.com/olikraus/u8g2/issues/1151#issuecomment-611419827_

olikraus, 2020-04-09:
@tunaftlmk, I assume this is a different problem. Please describe your issue here.

tunaftlmk, 2020-04-09:
Hello
I use the u8g2.drawXBMP function in the library. but I cannot show the picture in bitmap format from my lcd.assistan.
file
0x00, 0x01, 0x00, 0x00, 0x00, 0x7E, 0x7F, 0x00, 0x01, 0xFE, 0x1F, 0x80, 0x03, 0xFE, 0x07, 0xC0,
0x07, 0x8E, 0x01, 0xE0, 0x07, 0x82, 0x01, 0xE0, 0x07, 0xF0, 0x07, 0xE0, 0x07, 0xFC, 0x1F, 0xE0,
0x07, 0xF0, 0x07, 0xE0, 0x07, 0x82, 0x01, 0xE0, 0x07, 0x8E, 0x01, 0xE0, 0x03, 0xFE, 0x07, 0xC0,
0x01, 0xFE, 0x1F, 0x80, 0x00, 0xFE, 0x7F, 0x00, 0x00, 0x01, 0x00, 0x00, 
in this way
32x15 in size 
I cannot show this picture on the screen.
The picture in the sample code looks fine, but the picture I'm converting does not look right.
I paid attention to dimensions and function spelling.

olikraus, 2020-04-09:
I do not know lcd.assistan. The format should be XBM (as saved by the tool gimp). How does your code look like?

tunaftlmk, 2020-04-09:
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif

U8G2_SSD1327_WS_128X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /*
data=*/ 11, /* cs=*/ 10, /* dc=*/ 7, /* reset=*/ 8);

const unsigned char tuna[] U8X8_PROGMEM = {
0x00, 0x01, 0x00, 0x00, 0x00, 0x7E, 0x7F, 0x00, 0x01, 0xFE, 0x1F, 0x80,
0x03, 0xFE, 0x07, 0xC0,
0x07, 0x8E, 0x01, 0xE0, 0x07, 0x82, 0x01, 0xE0, 0x07, 0xF0, 0x07, 0xE0,
0x07, 0xFC, 0x1F, 0xE0,
0x07, 0xF0, 0x07, 0xE0, 0x07, 0x82, 0x01, 0xE0, 0x07, 0x8E, 0x01, 0xE0,
0x03, 0xFE, 0x07, 0xC0,
0x01, 0xFE, 0x1F, 0x80, 0x00, 0xFE, 0x7F, 0x00, 0x00, 0x01, 0x00, 0x00,
};

void setup(void) {

  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.drawXBMP(20,20, 32, 15, tuna);
  } while ( u8g2.nextPage() );
  //delay(1000);
}
this is the code
when i use another library the picture is smooth but your library does not
look right
And the memory of the Arduino is filling up very quickly. What should I do
for this?

olikraus <notifications@github.com>, 9 Nis 2020 Per, 18:32 tarihinde şunu
yazdı:

tunaftlmk, 2020-04-09:
I returned from the mail, teacher
I threw out the code and how the screen looks

olikraus, 2020-04-09:
How does the picture of the screen look like and how does the original picture look?

olikraus, 2020-04-10:
Please do not simply reply to e-mails. Or at least reply to the correct e-mails. I will delete your reply in the other thread.

> Does Gımp produce G code?

Why is this relevant? It produces XBM format. Thats important here.

tunaftlmk, 2020-04-10:
How does it produce the XBM format.
Is there an example for this

olikraus, 2020-04-10:
> How does it produce the XBM format.
> Is there an example for this?

https://sandhansblog.wordpress.com/2017/04/16/interfacing-displaying-a-custom-graphic-on-an-0-96-i2c-oled/

tunaftlmk, 2020-04-10:
thank you.
I solved the problem

tunaftlmk, 2020-04-10:
I have one more problem.
Arduino memory is filling up very quickly.
what can i do from the library for that?

olikraus, 2020-04-10:
If I remember correctly, the u8g2 faq includes some tips.

closed 2020-04-18 by olikraus