olikraus/u8g2 #1153: Poor 4-Wire SPI performance on 256x64 SSD1322 & Teensy 3.5

QuadeHale, 2020-04-10:
I have scoured the internet to no avail to try and figure out why i am getting sub-30 frames per second with a full framebuffer on a Teensy 3.5.
Code is as follows:
```c++
#include <SPI.h>
#include <U8g2lib.h>
#include <U8x8lib.h>

#define logo_width 256
#define logo_height 58
static const unsigned char logo_bits[] = {...}; // about half the pixels are on

#define DATA_PIN 11
#define CLOCK_PIN 27
#define RESET_PIN 24
#define CHIP_SELECT_PIN 10
#define MODE_PIN 25

U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI dispHW(U8G2_R2, CHIP_SELECT_PIN, MODE_PIN, RESET_PIN); 

void setup() {
  SPI.setMOSI(11); // i am using non-default pins
  SPI.setSCK(27); // ^^
  pinMode(13, OUTPUT);
  digitalWrite(13, HIGH);
  dispHW.begin();
}

void loop() {
  static uint8_t x = 0;
  static uint8_t y = 0;
  
  digitalWrite(13, !digitalRead(13));
  dispHW.clearBuffer();
  dispHW.drawXBM(x, y, logo_width, logo_height, logo_bits);
  dispHW.sendBuffer();
  x++;
  if (x > 254) {
    y = (y + 4) % 64;
    x = 0;
  }
}
```

I've seen some stuff in the source about setting the base SPI clock speed to 10MHz but *even that* is substantially faster than what should be necessary to push 2048 bytes down the pipe at least 60 times a second, but the Teensy 3.5 has a native SPI speed of 30MHz.
Is the lag in the XMB -> buffer? I have tried it with and without PROGMEM - and it seems to make no difference. I notice the same problem if I swap the XMB for some text - the more text is on the screen the longer it takes to render. As the text slowly moves off the screen, the framerate picks up. 
Other than the framerate, every function works properly. I have checked the other issues flagged by GitHub and none of them have answers. I am at a loss, any guidance would be appreciated.

olikraus, 2020-04-10:
> to push 2048 bytes down the pipe

Actually it is four times more than this, because the SSD1322 is a graylevel display.

From my own measures (https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/FPS/FPS.ino#L209) I never reached more than 14 FPS

QuadeHale, 2020-04-10:
Okay - but that's at 10MHz; if we are at 30MHz shouldn't that mean 40+ FPS worst-case scenario?
I've tried changing the clock speed in the U8x8 function that sets up the SSD1322 but that seemed to have no effect.

olikraus, 2020-04-11:
Did you use the setBusClock procedure? But I think the SPI speed of the display itself is limited.

QuadeHale, 2020-04-11:
I just put my scope on the SCK to watch and see; when set to `30000000` the bit rate improves accordingly (up to that limit). The 'packet' rate moves from ~1.4MHz to ~1.8MHz.

However, the space between the data packets does not; I believe this to be the source of the performance issues. From the last high of one packet to the first high of the next is 320nS; this value does not change with bus speed. The whole clock cycle should work down to 100nS according to the [spec sheet](https://www.newhavendisplay.com/resources_dataFiles/datasheets/OLEDs/SSD1322.pdf) (page 53). I have tried messing with the timing values in `u8x8_d_ssd1322.c` but to no avail.

I have tried digging through the source further to find where the delays are inserted, but frankly there is a dizzying amount of code here!

olikraus, 2020-04-11:
The transfer of the SPI data happens here: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1322.c#L207

This will end up here:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L880

QuadeHale, 2020-04-11:
So where does the pause on the clock line come from? I would have thought the clock would stay at 30MHz for the entire transaction; instead we have this 320ns gap after every 8 bits sent.

olikraus, 2020-04-11:
As mentioned:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L880

This will transfer only one byte at a time. Unfortunately there is no proper block transfer method provided by the Arduino HAL, so U8g2 must transfer each byte separatly.

olikraus, 2020-04-11:
Just to complete this: There is a block transfer method, but unfortunately it is destructive an can not be used in u8g2:
https://arduino.stackexchange.com/questions/56419/spi-transferbuffer-size-does-not-send-data-from-buffer

QuadeHale, 2020-04-11:
The Teensy is able to do a block write, as seen here:
[https://forum.pjrc.com/threads/59091-Teensy-3-6-Max-SPI-transfer-speed](https://forum.pjrc.com/threads/59091-Teensy-3-6-Max-SPI-transfer-speed).
I admit this is getting a bit above my head - I assume I would have to implement this function somewhere around [https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L880](https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L880), right?

olikraus, 2020-04-11:
Yes, exactly. If your local Arduino HAL has a better block transfer, then this would be the right place to use ist.

QuadeHale, 2020-04-11:
I have managed to remove the gap by uncommenting
[https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L875](https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L875), commenting out the `data=...` down to the end of the `while` block and based on the Teensy SPI library:
`void SPIClass::transfer(const void * buf, void * retbuf, size_t count)` I changed the code to:
```
void *nullPtr;
SPI.transfer((uint8_t *)arg_ptr, nullPtr, arg_int);
```
and it's functioning correctly. It is not any faster, but the gaps between bytes have disappeared.
Now on the scope when I zoom out time-wise, I see a 26uS cavern of time where nothing is happening on the SCK line. The entire data transmission takes 9.7uS and then a 26uS pause. Any idea what the source of that pause is?

olikraus, 2020-04-11:
> The entire data transmission takes 9.7uS

How many bytes do you see here? As mentioned earlier, u8g2 breaks down the transfer into block, each should be 32 bytes: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1322.c#L207

Between those 32-bytes blocks, there will be a gap where the next 32 bytes are calculated: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1322.c#L123

This procedure does the recalculation for the SSD1322. Basically each bit (0 or 1) is mapped to either the value 0 or 15 (b1111). Maybe this procedure is responsible for the slow down.

QuadeHale, 2020-04-11:
I see 32 bytes, as you predicted. 
I am flabberghasted that converting four bytes in to 32 with some simple bitwise math is taking that long, though.
I am at a loss. I was hoping to at least get a consistent 30fps out of this thing, especially because the math shows I should be able to get much more. I am tempted to take what I've learned here and write a wrapper for the driver, replacing U8x8 and using a full 65k buffer and I could just shove the whole thing down the pipe always -  I've got the space to do so on this MCU.

olikraus, 2020-04-11:
U8g2/U8x8 had been written with monochrome devices in mind. The SSD1322 does not really fit here, especially U8g2/U8x8 do not support graylevel.

olikraus, 2020-04-18:
> replacing U8x8 and using a full 65k buffer and I could just shove the whole thing down the pipe always - I've got the space to do so on this MCU.

No doubt a good idea. U8g2 was written to be portable and a base for further development. It is somehow universal but can not replace a special written high speed driver for exactly one controller.

will close this issue...

closed 2020-04-18 by olikraus

JamesGKent, 2020-05-19:
don't know how much of a speed difference it would make, but could you not buffer it locally like in the I2C examples for other platforms and then use the block transfer function as it would only overwrite the local buffer that we don't care about?

olikraus, 2020-05-22:
Actually, the I2C buffer is introduced by the Arduino HAL. The opposite is true: U8g2 tries to transfer blocks with the maximum size for SPI. Only for I2C those huge blocks have to be cut down to smaller blocks.