olikraus/u8g2 #1156: Vertical Scrolling loop doesn't work

mattiaapoz, 2020-04-11:
Hi, thanks for your work! I'm trying to create a vertical scrolling text. The text is variable (received from Android) and I used substring to display correctly on the OLED. Now I need to scroll it up. I created this code that correctly shows the rows on the OLED but doesn't scroll, I used offset for decreasing the value at each cylce, from the Serial monitor I see that the cycle works but the text won't move! Can you help me? 

That's the code:
`for(int offset = 52; offset >= 0; offset--){
      for(z=0; z<str_len; z+=21){
        substr = "";
        substr = text.substring(pointer, pointer+21);
        u8g2.setCursor(2, 11+y+offset);
        u8g2.print(substr);
        u8g2.sendBuffer();
        pointer+=21;
        y += 11;
      }
      delay(100);
      Serial.println(offset);
    }`

Thanks :)

olikraus, 2020-04-11:
I would suggest to greatly simplify your code and add much more comments. It is better to start with simple examples. 
Also remember to use sendBuffer() only once you are done with the drawing for one offset (it does not matter if you do it more often, but it is a waste of time)
Additionally you probably need clearBuffer() somewhere.

BTW: I think the major problem in your could is "y", not beeing reset.

mattiaapoz, 2020-04-11:
`#include <Arduino.h>
#include <Wire.h>
#include <U8g2lib.h>
#include <ESP8266WiFi.h> //Libraries

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); //Create OLED object

String text = "This is a long text to check if the substring and the scrolling works pretty well!"; //String to display
String substr = ""; //Substring

int str_len = text.length()+1; //Get str_len
int pointer = 0; //Pointer for start and end substring
int y = 0; //Y Coordinate
  
void setup() {
  Serial.begin(9600); //Begin serial
  u8g2.begin(); //Display begin
  u8g2.clearBuffer(); //Clear display buffer
  u8g2.setFont(u8g2_font_ncenB08_tr); //Set text font
}

void loop() {
    for(int offset = 52; offset >= 0; offset--){ //Create a cycle that decrease Y by 1
      for(int z=0; z<str_len; z+=21){ //Cylce for substring creation
        substr = ""; //Clear the substring
        substr = text.substring(pointer, pointer+21); //Fill substr variable from pointer to pointer +21
        u8g2.setCursor(2, 11+y+offset); //Set the cursor for printing the substr 11 is the row height, Y is the coordinate for that substr and offset is the variable
        u8g2.print(substr); //Print the substr
        pointer+=21; //Increment the pointer to get the next substr
        y += 11; //Add 11 for move the next row below the ones before
      }
      u8g2.sendBuffer(); //send the buffer
      delay(100); //Delay for the scrolling
      Serial.println(offset); //Check the offset value from serial monitor
    }
}`

That's the full code, if I decrease manually the offset all the substring move up or down and the if you watch the Serial monitor you can see that the offset goes down and back to 52 but the the text doesn't move on the display.

olikraus, 2020-04-11:
For proper code format, use three `

> That's the full code, if I decrease manually the offset all the substring move up or down and the if you watch the Serial monitor you can see that the offset goes down and back to 52 but the the text doesn't move on the display.

It does move, but only below the visible screen.

I quote myself:

> BTW: I think the major problem in your could is "y", not beeing reset.

mattiaapoz, 2020-04-11:
> BTW: I think the major problem in your could is "y", not beeing reset.

So where do you think I have to do that? 

> For proper code format, use three `

Sorry for this!

olikraus, 2020-04-11:
> So where do you think I have to do that?

My initial idea was, that, if you add the comments, you will see your mistake by yourself.
I suggest to do more debugging on your end. 
What about this: Print out the value of "y". Maybe this will give you an idea what is wrong with your code.
If you see the values, of "y", then ask yourself, what this will mean for
```
u8g2.setCursor(2, 11+y+offset)
```

mattiaapoz, 2020-04-11:
> If you see the values, of "y", then ask yourself, what this will mean for

I just checked through serial monitor and I can't really understand why o.o

olikraus, 2020-04-11:
why what? Excause for asking, but I just had a look at your code. My assumption is, that y will become very large because you never reset "y".

mattiaapoz, 2020-04-11:
> why what? Excause for asking, but I just had a look at your code. My assumption is, that y will become very large because you never reset "y".

I'm taking a look to the for cycle condition

mattiaapoz, 2020-04-11:
> why what? Excause for asking, but I just had a look at your code. My assumption is, that y will become very large because you never reset "y".

Yes, Y get higher and higher but I can't understand why, if I create the same cycle with one line text it works well!

olikraus, 2020-04-11:
> Yes, Y get higher and higher but I can't understand why

Because you never reset "y"

mattiaapoz, 2020-04-11:
> Because you never reset "y"

That's true but I need this "y" for the multiline visualization. Do you think ScrollingText.ino example is fit-able to my needs?

olikraus, 2020-04-11:
I do not know your needs.

> That's true but I need this "y" for the multiline visualization. 

Sure, but you still need to clear the variable when you output the text for the next time.

This leads me to my very first comment: "I would suggest to greatly simplify your code".0

What I mean is this: You should start to output and scroll a single line of text. Once this works, go ahead with multiple lines of text.

mattiaapoz, 2020-04-11:
Thanks for your suggest, I'll try tomorrow to follow another way! 
My need is just get a multiline text from a single string, and display them on the OLED through scrolling

olikraus, 2020-04-11:
> just

Doing software development is never simple. No doubt, doing multiline output is not simple at all.  It is instead a complicated task. I work in the area of software quality, and i can definitly say, that even one line of c code can be extremly complicated (I do not mean complex pointer expressions, but simple low pass filter for example). 

Make thinks simple. Start with simple things. Break down your problem into simple tasks. Solve one task after the other. ;-)

mattiaapoz, 2020-04-12:
> Make thinks simple. Start with simple things. Break down your problem into simple tasks. Solve one task after the other. ;-)

Thanks to this! I just restarted with all the loops and I found the problem: it was the pointer!
If you want I can make an examples for everyone need it! 
My sketch divide a variable lenght string into substrings and show them in the OLED scrolling in order to visualize all of them!

closed 2020-04-18 by olikraus