olikraus/u8g2 #1157: u8g2.userInterfaceSelectionList() does not works

cano-stack, 2020-04-11:
Hi,
I wanted to program a menu and wanted to use the function u8g2.userInterfaceSelectionList() 

That's not gonna work. Is the function only run once because the "HI" is not displayed ?

Greetings from Germany

`
#include <Arduino.h>
#include <U8g2lib.h>

U8G2_ST7565_64128N_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 12, /* data=*/ 10, /* cs=*/ 8, /* dc=*/ 11, /* reset=*/ 9);

#define up 5
#define down 6
#define select 7

void setup() {

  Serial.begin(115200);

  pinMode(up, INPUT);
  pinMode(down, INPUT);
  pinMode(select, INPUT);

  u8g2.begin();

  u8g2.setFont(u8g2_font_6x12_tr);
}

const char *string_list =
  "Altocumulus\n"
  "Altostratus\n"
  "Cirrocumulus\n"
  "Cirrostratus\n"
  "Cirrus\n"
  "Cumulonimbus\n"
  "Cumulus\n"
  "Nimbostratus\n"
  "Stratocumulus\n"
  "Stratus";

uint8_t current_selection = 2;

void loop() {

  bool Bla1 = !digitalRead(up);
  bool Bla2 = !digitalRead(down);
  bool Bla3 = !digitalRead(select);

  Serial.println(current_selection);

  if (Bla1)
    current_selection++;

  if (Bla2)
    current_selection--;

  u8g2.userInterfaceSelectionList(
    "Cloud Types",
    current_selection,
    string_list) ;

  Serial.println("HI"); // is not displayed
}
`

olikraus, 2020-04-11:
userInterfaceSelectionList() will wait for your input. 

Additionally you should tell which buttons to use: https://github.com/olikraus/u8g2/wiki/u8g2reference#begin

I also suggest to start with one of the existing examples (see the Example menu in your Arduino IDE).
1. Ensure that your display works
2. Use one of the menu examples to test whether you buttons work. Remember to use aktive low buttons.

Also note, that u8g2 does not include a full features menu system. Better usr https://github.com/neu-rah/ArduinoMenu which supports u8g2.

cano-stack, 2020-04-12:
ah okay thanks
I will test the example of the Arduino IDE 

greetings

closed 2020-04-18 by olikraus