olikraus/u8g2 #1159: ST7567A 64x32 LCD

762SPR, 2020-04-13:
Hello, I have a display that seems to be the same as in issue 657 that I am trying to get working with this library without any luck so far. With issue 657 it appears that there was an edit made to add functionality for this display, but it never got tested... I am trying to use it on an ESP32 devkit 4 and can do any testing needed to get the functionality working. I am not sure if it is normal for displays, but even with a simple I2C scanner sketch I am unable to see the display (I CAN see other I2C devices) I am scanning with a for loop doing Wire.beginTransmission(address); Wire.endTransmission(); on addresses 0-127, which always returns 2 on endTransmission.

This should be the correct initialization code according to the mfg, it is just a little over my head to implement it...

```
void iic_start(void)
{
    SI=1;
    SCL=1;
    delayus();
    SI=0;
    delayus();
    SCL=0;
}
void iic_stop(void)
{
    SI=0;
    SCL=1;
    delayus();
    SI=1;
    delayus();
    SCL=1;  
}

void check_ACK(void)
{
    SCL = 1;
    _nop_();
     _nop_(); 
    SCL = 0;
    _nop_();
     _nop_(); 
}

void write_com(uchar para)
{
    iic_start();
    sendbyte(0x7e);  //SA0=1,SA1=1
    check_ACK();
    sendbyte(0x80);  //1 A0 000000
    check_ACK();
    sendbyte(para);
    check_ACK();
    iic_stop();
}

void init(void)
{   
        RES=0;
        delay(20);
        RES=1;
        delay(200);
        
     //   write_com(0xe2);         //Reset
     //   delay(40);

        write_com(0xff);
        write_com(0x6B);      //High Power Mode ON
        write_com(0xfe);

        write_com(0xff);      //Enhance Power Mode
        write_com(0x72);
        write_com(0xfe);
               
        write_com(0xd6);         //set 1/33 duty
      //  write_com(0x93);         //set 1/5 bias
        write_com(0x91);         //set 1/7 bias
      //  write_com(0x9c);         //set 175 Hz
        write_com(0x9e);         //set 300 Hz

        write_com(0xae);        //Dispaly Off

        write_com(0xa2);
        write_com(0xa0);
        write_com(0xc8);

        write_com(0xf8);
        write_com(0x01);
        
        write_com(0x26);        //set RR 5.0 ;
        write_com(0x81);        //Electronic Volume Mode Set
      //  write_com(0x27);        //Electronic Volume Register Set
        write_com(0x2B);          //11.2V
      
        write_com(0x2c);        //Booster ON
        write_com(0x2e);        //Regulator ON
        write_com(0x2f);        //Follower ON
        delay(10);
        
        write_com(0xaf);        //Dispaly On
        delay(10);
}
```

Thanks for any help!

olikraus, 2020-04-13:
Any datasheet of the display?

The display should appear on the i2c bus. It doesn't make sense to continue without seeing the display on the bus.

Did you configure the display for i2c?

762SPR, 2020-04-13:
The datasheet is the same for issue 657, sorry I thought I mentioned it was the ST7567A controller. Here is a direct link [](http://www.artronic.com.pl/pdf/pl/ST7567A.pdf)

I don't know how to configure the display for i2c, I believe that is its only mode, but I am also not seeing it on the bus so I don't know.

According to the pinout (also the same as 657, link here: [](http://www.artronic.com.pl/o_produkcie.php?id=1532?) ) there are only SCL and SDA wires to communicate so I am fairly certain it is i2c

olikraus, 2020-04-13:
How did you setup the display? You need a proper reset circuit and the two caps for the step up converter.

762SPR, 2020-04-13:
I have a 1uf cap on 1-2 and the same on 3-4. I have a 10k pullup on the reset pin, and 2.2k pullups on the data pins (3.3v from the esp32). Is the reset used for initialization?

Thanks

olikraus, 2020-04-13:
> Is the reset used for initialization?

Usually you must apply a proper reset to the display. It will not work otherwise.
Without having calculated the exact delay, but with your 10k pullup on the reset, you probably need a 100uF cap from the reset to GND. In u8g2/u8x8 you can also feed the reset from an existing GPIO pin. In this case, u8g2/u8x8 will ensure, that the device receives a proper reset pulse.

All in all: The display will not respond on I2C before a proper reset has happend. There is a full chapter on this topic in the linked datasheet for the ST7567: "[reset] ... is essential before operating".

olikraus, 2020-04-13:
One more point: Additionall place a 100uF between GND and power supply of your display.

762SPR, 2020-04-13:
Ok thanks. I didn't know the hardware reset had to be triggered before the i2c would acknowledge. With a u2g8 constructor, do I still need to do the hardware reset procedure, or is it done automatically? I had previously tried using the constructor:
`U8G2_ST7567_64X32_F_HW_I2C u8g2(); `
but that didn't seem to work either, perhaps because there was no decoupling capacitor? I will double check everything in the morning and see if I can't get it working with the u8g2 lib again. I had assumed since the other issue never seemed to get tested that it was still unresolved and went straight to trying to find the display on the bus by itself without the assistance of the library, however I may have missed something. I'll report back, thanks for the help so far!

olikraus, 2020-04-13:
> With a u2g8 constructor, do I still need to do the hardware reset procedure, or is it done automatically? 

You can sacrifice a GPIO to let u8g2 do the reset. But having the hardware reset circuit usually is the better solution.

762SPR, 2020-04-13:
OK finally some progress! I made up a completely new breakout board and this time when I plugged it in to the esp32 I still had the old I2C scanner code running and surprisingly even without the reset sequence, the display is now registering on the scan (0x3f). I am using a different display sample so maybe the other one was just a bad unit or I had somehow messed something up...

Unfortunately, that's where my success ends. I have tried the u8g2 library's helloWorld, logo, and graphics test examples both assigning pin 5 to reset in the constructor and leaving it null and resetting manually with this code:
```
void setup(void) {
  pinMode(5, OUTPUT);
  
  digitalWrite(5, LOW);
  delay(20);
  digitalWrite(5, HIGH);
  delay(200);

  u8g2.begin();
}
```

No luck with any of the example applications, but perhaps I am not properly using them? I'm using this constructor and have tried using pin 5 and PIN_NONE:
`U8G2_ST7567_64X32_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); `

I have also tried just flashing the screen by changing all pixels with no luck:
```
void loop() {
  byte error;
  Wire.beginTransmission(0x3F);
Wire.write(0x00);
Wire.write(0xA5); //all pixels on
error = Wire.endTransmission() ;
Serial.print("Error:");
Serial.println(error);
Serial.println("All ON");
delay(2000);
Wire.beginTransmission(0x3F);
Wire.write(0x00);
Wire.write(0xA4); //all pixels off
error = Wire.endTransmission() ;
Serial.print("Error:");
Serial.println(error);
Serial.println("All OFF");
delay(2000);
}
```

This is AFTER doing the reset sequence, then the init code from my first post. Nothing changes on the screen, but error is returning 0 which should indicate success.

Any ideas? Thanks!

olikraus, 2020-04-13:
Best is to use the software emulated I2C constructor instead of the HW I2C to exclude any issues with your board. Then test the display with one of the existing U8g2 examples from the Arduino IDE. For example with the U8g2 Logo.
Another issue might be this: Maybe the default I2C address is wrong in u8g2, so you should also set derived I2C address (setI2CAddress command in U8g2).

762SPR, 2020-04-13:
Still no luck... I'm beginning to wonder if I am missing something extremely basic? Just getting a blank screen, but 0x3f appears to be the correct address as it acknowledges write commands. I have tried with both u8g2's reset defined (pin 5) and doing it manually...

This is from the logo example
```
U8G2_ST7567_64X32_F_SW_I2C u8g2(U8G2_R0, 18, 19/*, 5*/);

void setup(void) {
  pinMode(5, OUTPUT);
  pinMode(18, OUTPUT);
  pinMode(19, OUTPUT); // do these need to be set?
  
  digitalWrite(5, LOW);
  delay(20);
  digitalWrite(5, HIGH);
  delay(200);

  u8g2.setI2CAddress(0x3F);
  u8g2.begin();
}
```

olikraus, 2020-04-14:
At least you must multiply the address with 2 (see the u8g2 manual)

762SPR, 2020-04-14:
Ah HA! I tried the address 0x7E... no change, but then I moved to a different angle and caught the slightest hint of a picture on the display! It turns out the contrast was all the way down! I set it to 255 and FINALLY got a picture showing! (I'm curious why the address needs to be multiplied by 2?)

There are a couple small issues, but first I tested a few different settings and here are my findings:

-The address DOES need to be set. after finding out about the contrast I tried commenting out the address and it didn't work so the default for this display is wrong. Perhaps change the default to 0x7E?
-Both HW and SW I2C appear to be working, as well using the built in reset pin functionality with your library.
-Contrast is low by default, perhaps automatically set it to 255 or 127 in the initialization?
-There are some artifacts on the right side (U8G2_R0), as well as everything is shifted as many pixels as the artifact area to the left. If I were to guess I would say the start address for the pixels is offset. If I set the orientation to U8G2_R2, the artifacts remain on the same side of the physical screen, they do not rotate with the orientation. I have attached an image showing the artifacts in R0 orientation. If you would like other tests or pictures just let me know!

[image]

Thank you for your help and patience so far! I was about to shelf this project, but now I am encouraged!

olikraus, 2020-04-14:
Looks good. I know, it can be hard to display something on an LCD/OLED.

>  (I'm curious why the address needs to be multiplied by 2?)

The I2C address is not so clearly defined. The first byte transfered to the I2C device contains the 7 bit address in the upper part of the 8 bits. This is where the multiplication with 2 comes from.
I decided to avoid this calculation and ask the users of u8g2/u8x8 to privde the already multipled address. Well, finally, it is a none-ending discussion...

> Perhaps change the default to 0x7E?

I decided to use the lowest address provided by th econtroller. This is actually 0x78 (see the datasheet on page 23). This controller actually provides four different I2C addresses and your display LCD vendor deviated has choosen the highest address. There there was a 1:4 chance to hit the right address for u8g2/u8x8. Well, thats why there is an option to set the address manually.

But I think I can create a new constructor for your display, with fixed constrast and differnt default for the I2C address.

 > There are some artifacts on the right side (U8G2_R0),

Actually the ST7565 controller contains a 132x65 bitmap matrix. Which area is mapped to the active LCD area is again defined by the display vendor. The existing constructor was made for a different display with different properties (different constrast, different i2c address, different wiring)

We now need to find out the correct configuration for your LCD memory mapping.

Excellent and detailed picture by the way!

Please locate file u8x8_d_st7567.c on your harddisk.
https://github.com/olikraus/u8g2/blob/5acf0c652d1ab7fe54d897e809bbb8905f32eca7/csrc/u8x8_d_st7567.c#L597-L598

Can you modify the above metioned lines? Best is to use the u8g2 flip mode example. It will cycle through all R0, R1, R2, R3 modes. In all the modes there must be a closed frame exactly near the border of the display. 
The challange now is to adjust default_x_offset (R0 and R1) and flipmode_x_offset (R2 and R3) so that this fits for your display.  I personally assume that you need to change both values from 32 to 36.

Can you test this? Thanks a lot!

762SPR, 2020-04-14:
Great! I was hoping it would be a simple configuration issue. I think making a new constructor will be the best route once we get the changes sorted out. I just don't know enough about LCDs to start messing with it without any direction... I will try that tonight and report back the results!

After posting last night I was tinkering with writing to the display directly and learned a little about how it works so what you said seems to make sense, but again I'm no expert so we will see tonight. Thanks for all the help!

762SPR, 2020-04-15:
Great news! It's looking just about perfect! You were spot on about 36 for the default X offset. For some reason, the flip mode should still be 32 though. I also think based on my testing last night that the technical bottom of the screen is where the ribbon cable comes out of. When I did:

```
write_com(0x40); // start line
  write_com(0xb0); // page
  write_com(0x12);//Set Higer Column Address
  write_com(0x00); //Set Lower Column Address

// inside loop:

write_data(0xAA);// alternating dashed line
```

It started drawing in the upper left corner when oriented with the cable facing the bottom. In application, it doesn't make much difference because you can just give it a different rotation in the constructor, but I figured I'd pass that along anyway if you end up making a special constructor for it

The LCD unit itself is called a Lummax HEM6432-03 if that helps you name it. To my eye, about 225 seems good for the contrast, 255 starts bleeding through the dark pixels just a little bit. I2C address should be 0x7E with the multiplication applied.

Other than that, it's just about perfect! Thank you so much for your help getting all this working, and also for making and publishing the library in the first place! I have learned quite a bit about bare bones LCD modules from this project, so I'm really happy about that, but I have also learned that I would have had to put a LOT of time and effort into figuring out and writing all of these functions myself, so I am very happy that you have already done all the work for me!

[image]
[image]

olikraus, 2020-04-18:
I have created a new constructor "U8G2_ST7567_HEM6432..."
It includes the new x offset and a different I2C default address (hopefully). Maybe you can test the same.

I have created a new beta release 2.28.2.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2020-04-18:
Thanks for your work on this. Nice pictures!.

762SPR, 2020-04-19:
It looks great! I tried 
`U8G2_ST7567_HEM6432_1_HW_I2C  u8g2 (U8G2_R0, 5);`
Inside of the flipMode demo and it worked flawlessly! II may have discovered a small bug though. When ran the first test, the display seemed quite dim and I thought I was mistaken with my 225 suggestion as I noticed you changed that in the constructor, so I set it to 255 in the constructor and it looked good, but not as washed out as I remembered 255 before. To prove I wasn't going crazy I went and set the constructor back to 225, and then compared it to setting the contrast to 255 via 
`u8g2.setContrast(225);`
As you can see from the pictures, the brightness is significantly brighter when set via setContrast vs sending
`U8X8_CA(0x081, 225),`
in the constructor. Perhaps there is a math error?

Besides that it looks great, I'm quite happy with it and how quickly you got it fixed up for me so thank you again!
[image]
[image]

olikraus, 2020-04-19:
> Perhaps there is a math error?

Yes, it is actually 225 / 4 because only 6 bits are accepted by the controller.
I have updated the repo and created beta 2.28.5

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

dankatzman, 2020-05-14:
Hello. I am new to this LCD and try to activate it as well. no successes yet! 
I red the discussion above. Very informative!!!
I have downloaded the master.zip file. Opened the flipMode demo but did not find a line with this text:
U8G2_ST7567_HEM6432_1_HW_I2C u8g2 (U8G2_R0, 5);
What am i missing here. should such line appears in this demo file?
Thanks for helping.

762SPR, 2020-05-14:
No, that is inside of the source code, but you will have to dig for it. If you have the latest version, it should be inside just paste that line into your code amd it should work!

Have you tried that and it's still not working, or was that what you were missing? If you look close with my pictures you should be able to see the electronic circuit as well, don't forget about the capacitors and the pull up resistors (I used smd components, they are tough to see, but there!)

dankatzman, 2020-05-14:
hi. i do have the latest version. i downloaded the library today from the link https://github.com/olikraus/U8g2_Arduino/archive/master.zip

however i was able to activate the LCD with Arduino UNO through the I2C with the following code:

#include <Arduino.h>
#include <U8x8lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif

U8X8_ST7567_HEM6432_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); // constructor
void setup(void)
{
	u8x8.begin();
	u8x8.setFont(u8x8_font_8x13B_1x2_r);
	u8x8.clear();
	u8x8.setFlipMode(0);
	u8x8.setInverseFont(0);
	u8x8.drawString(1, 1, "12:34");
}

void loop(void)
{
}
and got the following image
[image]
[image]

dankatzman, 2020-05-14:
as you can see i was not using the constructor that was commandeered
U8G2_ST7567_HEM6432_1_HW_I2C u8g2 (U8G2_R0, 5);
when i used it the program did not compile.

however, what i got is good step but i need to get a bigger font. when I use othe fornt option they are either too big or too small. what is the option to get bigger font that will enable priting 12:34 at about 90% of the LCD width?
thanks for helping me

762SPR, 2020-05-14:
It may be because you are including u8g8, and not u8g2. Try the u8g2 libraries and see if that works. Or post the compile error if it doesn't. Also, I'm surprised it is working without the reset pin defined...

dankatzman, 2020-05-17:
Hi
You are right. In included u8g2lib instead of u8x8lib and it gave me many good size fonts for my project.
I made my wiring based on this article:
https://ep.com.pl/podzespoly/11799-modul-miniaturowego-wyswietlacza-graficznego-dla-mikrokontrolera
 as you can see, the font size is large to the maximum
[image]
as for the rest. I don't know what to asnwer.
thanks for your guide.
dan

olikraus, 2020-05-22:
Excause for my rare presence on this project.
Thanks for testing. So the constructor will work, right?
Can we close this issue?

dankatzman, 2020-05-23:
Hi
The constructor that I am using U8X8_ST7567_HEM6432_HW_I2C  works good for me.
There are few constructors for HEM6432. 
I don't know why but it is good that one of them works for me. 
Thank you very much
Dan

olikraus, 2020-05-23:
Great, thanks for the feedback

closed 2020-05-31 by olikraus

catalin7, 2025-07-29:
> > HELP PLEASE, WHERE DID YOU BUY THIS lcd ?
> 
> At least I don't own the device nor did I buy it. Did you try the SW_I2C constructor?

I would if I know where to buy it , thank you

olikraus, 2025-07-30:
Even I would like to know where to buy it..