olikraus/u8g2 #1160: png2bin does not work

yogri, 2020-04-13:
Hi there.
I would like to convert image files to .bin files so that they can be read from the SD card. Unfortunately I cannot start the png2bin.exe on Windows 10. Only a black window (console) opens and closes again immediately.
I would be happy if you could help me.
Thank you.

AnHardt, 2020-04-13:
> I cannot start the png2bin.exe on Windows 10. Only a black window (console) opens and closes again immediately.

`png2bin.exe` is a command line program. Use it from a console.

>I would like to convert image files to .bin files so that they can be read from the SD card. 

`png2bin.exe` produces files formated like https://github.com/olikraus/u8g2/blob/master/tools/png2bin/u8g2_logo_97x51.xbm. That format is thought to be used directly in a c-programs .h-file. That's far from ideal for reading from a file during run time. A less ASCII format would be much better.

olikraus, 2020-04-13:
png2bin does not really generate a true XBM. Instead it generates a XBM-line-binary which can be stored on a memory card. Then the decode algorithm in drawFile will read, decode and draw the original image: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino#L373

yogri, 2020-04-13:
AnHardt, thank you very much. Now I can create bin files.
Unfortunately the .bin files are not displayed correctly. I use the example LoadFromSD.ino to read the file.
I reduced the pictures with paint and saved them as png. It is also displayed correctly in the png2bin ASCII preview. But only a noise can be seen on the display.
What am I doing wrong?

olikraus, 2020-04-18:
Is this solved? Maybe post some code here.

yogri, 2020-04-19:
It works now. I cannot create working png files with MS Paint. I have now installed Gimp and now everything works fine.
I have one last question. My SSD1327 display is currently working in I2C mode. So I can only display monochrome pictures. Can the png2bin converter generate images that can be displayed in grayscale or in color in SPI mode? If so, then I re-solder my display to SPI accordingly.

olikraus, 2020-04-19:
U8g2 does not support grayscale. U8g2 is a pure black and white library. This is not a question of I2C or SPI.

yogri, 2020-04-19:
OK, too bad. But the pictures look good too.
Thanks for the quick support.

closed 2020-04-19 by yogri

JasonWong08, 2021-01-24:
How to generate a true XBM files from png file in windows 7 OS?

olikraus, 2021-01-25:
is this question related to u8g2?