olikraus/u8g2 #1162: How to DrawStr with encoding 127 to 255 with char/uint8_t string(array)

jieweiyang, 2020-04-15:
I have created some custom icons in FONY, export as BDF, and converted with bdfconv.
For special and convenient purpose, i created the invert chars in between 127 to 255.
Can I use DrawStr directly to draw them with char/uint8_t array?

for example as below in C:
```
uint8_t L5[] = "123456";
L5[0]+=112;
L5[1]+=112;
u8g2_DrawStr(&u8g2, 0, 15, L5);
```

Thanks.

olikraus, 2020-04-15:
Yes, this should work.

jieweiyang, 2020-04-15:
Actually it doesn't. All turn into blank.
I tried modify the font to move those icons fall under 97-122 which are lower case area, it can be draw correctly.

jieweiyang, 2020-04-15:
attached with bdfconv command.

bdfconv -v DED.bdf -b 0 -f 1 -n ded -o u8g2_font_ded.c

Parse File DED.bdf: 256 glyph(s) found
Map: map_cmd_list='*'
Map: exclude=0 from=32/$20 to=255/$ff map=32/$20
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=-2, w=8, h=11
CalculateMaxBBX: Encodings x=32, y=46, w=32, h=161
bf_CalculateMinMaxDWidth: dx_min=8, dx_max=8
bf_CalculateMinMaxDWidth: x_min=0, x_max=3
bf_CalculateMinMaxDWidth: Monospaced font.
bf_CalculateMaxBitFieldSize: bbx.x=3, bbx.y=2, bbx.w=4, bbx.h=4, dwidth=5
RLE Compress: best zero bits 3, one bits 4, total bit size 13248
RLE Compress: Font code generation, selected glyphs=224, total glyphs=256
RLE Compress: ASCII gylphs=224, Unicode glyphs=0
RLE Compress: Glyphs per unicode lookup table entry=100
RLE Compress: Unicode lookup table len=0, written entries=0
RLE Compress: Unicode lookup table first entry: delta=4, encoding=65535
RLE Compress: Unicode lookup table last entry: delta=5123, encoding=0
RLE Compress: 'A' pos = 247, 'a' pos = 586
RLE Compress: Font size 1687
bf_WriteU8G2CByFilename: Write file 'u8g2_font_ded.c'

olikraus, 2020-04-15:
Can you attach/paste the font here?

jieweiyang, 2020-04-16:
Here's the BDF file.

https://www.dropbox.com/s/lhwodzod46t492s/F16_DED.bdf?dl=0

olikraus, 2020-04-18:
hmm... can you instead add the DED.bdf file? You can also just copy/past the content here.

olikraus, 2020-04-18:
or even better the generated "u8g2_font_ded.c"

jieweiyang, 2020-04-18:
`/*
  Fontname: F16DED
  Copyright: Created with Fony 1.4.7
  Glyphs: 224/256
  BBX Build Mode: 0
*/
const uint8_t f16ded[1699] U8G2_FONT_SECTION("f16ded") = 
  "\340\0\3\4\4\4\3\2\5\10\13\0\376\10\377\0\0\0\367\2J\6\206 \5\0\24\3!\5\0\24"
  "\3\42\5\0\24\3#\5\0\24\3$\5\0\24\3%\5\0\24\3&\5\0\24\3'\5\0\24\3("
  "\5\0\24\3)\5\0\24\3*\15u\26+J\224\221\222\221\230\202\0+\5\0\24\3,\5\0\24\3"
  "-\5\0\24\3.\6\42\17\203\0/\5\0\24\3\60\15\226\15K\214D\210N\42dB\0\61\12\224"
  "\16+HdH.\10\62\14\226\15\207\4IPDE\220\1\63\15\226\15\203QLl\224\10\205\4\0"
  "\64\14\226\15CD~a\23\224\2\0\65\14\226\15\303\261\304\220\10\205\4\0\66\15\226\15\207\4\261\4"
  "\211\10\205\4\0\67\12\226\15\203Q\232\214\322\0\70\16\226\15\207\4\11\205D\210\10\205\4\0\71\16\226"
  "\15\207\4\211\10\245P\244\204\4\0:\5\0\24\3;\5\0\24\3<\5\0\24\3=\5\0\24\3>"
  "\5\0\24\3\77\5\0\24\3@\5\0\24\3A\15\226\15K\214\4\211\350!\210H\0B\15\226\15\243"
  "\4\351\244D\210\250\5\0C\14\226\15\207\4\211P&\24\22\0D\13\226\15\243\250D\376\213\12\0E"
  "\13\226\15\303\221\210HP\220\1F\13\226\15\303\221\210HPF\0G\14\226\15\207\4\211P\244\22J"
  "\1H\13\226\15C\210\322C\20M\2I\11\226\15\203IPN\14J\13\226\15\203IPNDc\0"
  "K\17\226\15C\210\4\205h\214\250D\244H\0L\10\226\15CP>\62M\13\226\15#p\350!\210"
  "N\2N\15\226\15#\214\250\244\204\244\22\265\0O\15\226\15K\214D\210N\42dB\0P\15\226\15"
  "\243\4\211\350\204HP\42\0Q\13\226\15\207\4\211\26$R\1R\17\226\15\243\4\211\350\244D(D"
  "\210H\0S\15\226\15\207\4\261\250P\20\205\4\0T\11\226\15\203IP>\1U\12\226\15C\210>"
  "\241\220\0V\15\226\15C\210N\42AA\324\204\0W\13\226\15C\210N\17A\203\1X\21\226\15C"
  "\210H$(\210L\214(\212\20\221\0Y\16\226\15C\210H$(\210LPN\0Z\11\226\15\203Q"
  "L\216\14[\5\0\24\3\134\5\0\24\3]\5\0\24\3^\5\0\24\3_\5\0\24\3`\5\0\24"
  "\3a\15\226\15K\214\4\211\350!\210H\0b\15\226\15\243\4\351\244D\210\250\5\0c\14\226\15\207"
  "\4\211P&\24\22\0d\13\226\15\243\250D\376\213\12\0e\13\226\15\303\221\210HP\220\1f\13\226"
  "\15\303\221\210HPF\0g\14\226\15\207\4\211P\244\22J\1h\13\226\15C\210\322C\20M\2i"
  "\11\226\15\203IPN\14j\13\226\15\203IPNDc\0k\17\226\15C\210\4\205h\214\250D\244"
  "H\0l\10\226\15CP>\62m\13\226\15#p\350!\210N\2n\15\226\15#\214\250\244\204\244\22"
  "\265\0o\15\226\15K\214D\210N\42dB\0p\15\226\15\243\4\211\350\204HP\42\0q\13\226\15"
  "\207\4\211\26$R\1r\17\226\15\243\4\211\350\244D(D\210H\0s\15\226\15\207\4\261\250P\20"
  "\205\4\0t\11\226\15\203IP>\1u\12\226\15C\210>\241\220\0v\15\226\15C\210N\42AA"
  "\324\204\0w\13\226\15C\210N\17A\203\1x\21\226\15C\210H$(\210L\214(\212\20\221\0y"
  "\16\226\15C\210H$(\210LPN\0z\11\226\15\203QL\216\14{\5\0\24\3|\5\0\24\3"
  "}\5\0\24\3~\5\0\24\3\177\5\0\24\3\200\5\0\24\3\201\5\0\24\3\202\5\0\24\3\203\5"
  "\0\24\3\204\7\270\4\343\37\32\205\5\0\24\3\206\5\0\24\3\207\5\0\24\3\210\5\0\24\3\211\5"
  "\0\24\3\212\5\0\24\3\213\5\0\24\3\214\5\0\24\3\215\5\0\24\3\216\21\270\4\343\241\244$\12"
  "\31)\31I\224\222\7\2\217\5\0\24\3\220\5\0\24\3\221\5\0\24\3\222\5\0\24\3\223\5\0\24"
  "\3\224\11\270\4\343\17GF\13\225\5\0\24\3\226\15\270\4c\251pH\376i\260h\1\227\15\270\4"
  "\203\305\250\314\310\246BB\5\230\20\270\4CqXH\310hTt\310X\70\1\231\15\270\4#Y\330"
  "\250\332\221\314\203\12\232\22\270\4#)h(h(h(hX\270\310\222\2\233\15\270\4#Y\222\351"
  "\260\221\314\203\12\234\16\270\4CqX\310tXH\316\203\12\235\14\270\4#Y\330\310\250&K\14\236"
  "\16\270\4CqXH\362\340\220\234\7\25\237\16\270\4CqXH\316\243F\243\203\12\240\5\0\24\3"
  "\241\5\0\24\3\242\5\0\24\3\243\5\0\24\3\244\5\0\24\3\245\5\0\24\3\246\5\0\24\3\247\5"
  "\0\24\3\250\5\0\24\3\251\5\0\24\3\252\5\0\24\3\253\5\0\24\3\254\5\0\24\3\255\5\0\24"
  "\3\256\5\0\24\3\257\5\0\24\3\260\5\0\24\3\261\5\0\24\3\262\5\0\24\3\263\5\0\24\3\264"
  "\5\0\24\3\265\5\0\24\3\266\5\0\24\3\267\5\0\24\3\270\5\0\24\3\271\5\0\24\3\272\5\0"
  "\24\3\273\5\0\24\3\274\5\0\24\3\275\5\0\24\3\276\5\0\24\3\277\5\0\24\3\300\5\0\24\3"
  "\301\5\0\24\3\302\5\0\24\3\303\5\0\24\3\304\5\0\24\3\305\5\0\24\3\306\5\0\24\3\307\5"
  "\0\24\3\310\5\0\24\3\311\5\0\24\3\312\5\0\24\3\313\5\0\24\3\314\5\0\24\3\315\5\0\24"
  "\3\316\5\0\24\3\317\5\0\24\3\320\5\0\24\3\321\5\0\24\3\322\5\0\24\3\323\5\0\24\3\324"
  "\5\0\24\3\325\5\0\24\3\326\5\0\24\3\327\5\0\24\3\330\5\0\24\3\331\5\0\24\3\332\5\0"
  "\24\3\333\5\0\24\3\334\5\0\24\3\335\5\0\24\3\336\5\0\24\3\337\5\0\24\3\340\5\0\24\3"
  "\341\5\0\24\3\342\5\0\24\3\343\5\0\24\3\344\5\0\24\3\345\5\0\24\3\346\5\0\24\3\347\5"
  "\0\24\3\350\5\0\24\3\351\5\0\24\3\352\5\0\24\3\353\5\0\24\3\354\5\0\24\3\355\5\0\24"
  "\3\356\5\0\24\3\357\5\0\24\3\360\5\0\24\3\361\5\0\24\3\362\5\0\24\3\363\5\0\24\3\364"
  "\5\0\24\3\365\5\0\24\3\366\5\0\24\3\367\5\0\24\3\370\5\0\24\3\371\5\0\24\3\372\5\0"
  "\24\3\373\5\0\24\3\374\5\0\24\3\375\5\0\24\3\376\5\0\24\3\377\5\0\24\3\0\0\0\4\377"
  "\377\0";
`

olikraus, 2020-04-18:
Hm... somehow the conversion did fail. From where did you get bdfconv tool?

The correct generated code is this
```
const uint8_t ded[1687] U8G2_FONT_SECTION("ded") = 
  "\340\0\3\4\4\4\3\2\5\10\13\0\376\10\377\0\0\0\367\2J\6z \5\0\24\3!\5\0\24"
  "\3\42\5\0\24\3#\5\0\24\3$\5\0\24\3%\5\0\24\3&\5\0\24\3'\5\0\24\3("
  "\5\0\24\3)\5\0\24\3*\15u\26+J\224\221\222\221\230\202\0+\5\0\24\3,\5\0\24\3"
  "-\5\0\24\3.\6\42\17\203\0/\5\0\24\3\60\15\226\15K\214D\210N\42dB\0\61\12\224"
  "\16+HdH.\10\62\14\226\15\207\4IPDE\220\1\63\15\226\15\203QLl\224\10\205\4\0"
  "\64\14\226\15CD~a\23\224\2\0\65\14\226\15\303\261\304\220\10\205\4\0\66\15\226\15\207\4\261\4"
  "\211\10\205\4\0\67\12\226\15\203Q\232\214\322\0\70\16\226\15\207\4\11\205D\210\10\205\4\0\71\16\226"
  "\15\207\4\211\10\245P\244\204\4\0:\5\0\24\3;\5\0\24\3<\5\0\24\3=\5\0\24\3>"
  "\5\0\24\3\77\5\0\24\3@\5\0\24\3A\15\226\15K\214\4\211\350!\210H\0B\15\226\15\243"
  "\4\351\244D\210\250\5\0C\14\226\15\207\4\211P&\24\22\0D\13\226\15\243\250D\376\213\12\0E"
  "\13\226\15\303\221\210HP\220\1F\13\226\15\303\221\210HPF\0G\14\226\15\207\4\211P\244\22J"
  "\1H\13\226\15C\210\322C\20M\2I\11\226\15\203IPN\14J\13\226\15\203IPNDc\0"
  "K\17\226\15C\210\4\205h\214\250D\244H\0L\10\226\15CP>\62M\13\226\15#p\350!\210"
  "N\2N\15\226\15#\214\250\244\204\244\22\265\0O\15\226\15K\214D\210N\42dB\0P\15\226\15"
  "\243\4\211\350\204HP\42\0Q\13\226\15\207\4\211\26$R\1R\17\226\15\243\4\211\350\244D(D"
  "\210H\0S\15\226\15\207\4\261\250P\20\205\4\0T\11\226\15\203IP>\1U\12\226\15C\210>"
  "\241\220\0V\15\226\15C\210N\42AA\324\204\0W\13\226\15C\210N\17A\203\1X\21\226\15C"
  "\210H$(\210L\214(\212\20\221\0Y\16\226\15C\210H$(\210LPN\0Z\11\226\15\203Q"
  "L\216\14[\5\0\24\3\134\5\0\24\3]\5\0\24\3^\5\0\24\3_\5\0\24\3`\5\0\24"
  "\3a\15\226\15K\214\4\211\350!\210H\0b\15\226\15\243\4\351\244D\210\250\5\0c\14\226\15\207"
  "\4\211P&\24\22\0d\13\226\15\243\250D\376\213\12\0e\13\226\15\303\221\210HP\220\1f\13\226"
  "\15\303\221\210HPF\0g\14\226\15\207\4\211P\244\22J\1h\13\226\15C\210\322C\20M\2i"
  "\11\226\15\203IPN\14j\13\226\15\203IPNDc\0k\17\226\15C\210\4\205h\214\250D\244"
  "H\0l\10\226\15CP>\62m\13\226\15#p\350!\210N\2n\15\226\15#\214\250\244\204\244\22"
  "\265\0o\15\226\15K\214D\210N\42dB\0p\15\226\15\243\4\211\350\204HP\42\0q\13\226\15"
  "\207\4\211\26$R\1r\17\226\15\243\4\211\350\244D(D\210H\0s\15\226\15\207\4\261\250P\20"
  "\205\4\0t\11\226\15\203IP>\1u\12\226\15C\210>\241\220\0v\15\226\15C\210N\42AA"
  "\324\204\0w\13\226\15C\210N\17A\203\1x\21\226\15C\210H$(\210L\214(\212\20\221\0y"
  "\16\226\15C\210H$(\210LPN\0z\11\226\15\203QL\216\14{\5\0\24\3|\5\0\24\3"
  "}\5\0\24\3~\5\0\24\3\177\5\0\24\3\200\5\0\24\3\201\5\0\24\3\202\5\0\24\3\203\5"
  "\0\24\3\204\5\0\24\3\205\5\0\24\3\206\5\0\24\3\207\5\0\24\3\210\5\0\24\3\211\5\0\24"
  "\3\212\5\0\24\3\213\5\0\24\3\214\5\0\24\3\215\5\0\24\3\216\5\0\24\3\217\5\0\24\3\220"
  "\5\0\24\3\221\5\0\24\3\222\5\0\24\3\223\5\0\24\3\224\5\0\24\3\225\5\0\24\3\226\5\0"
  "\24\3\227\5\0\24\3\230\5\0\24\3\231\5\0\24\3\232\5\0\24\3\233\5\0\24\3\234\5\0\24\3"
  "\235\5\0\24\3\236\5\0\24\3\237\5\0\24\3\240\5\0\24\3\241\11\270\4\343\17GF\13\242\7\270"
  "\4\343\37\32\243\15\270\4c\251pH\376i\260h\1\244\15\270\4\203\305\250\314\310\246BB\5\245\20"
  "\270\4CqXH\310hTt\310X\70\1\246\15\270\4#Y\330\250\332\221\314\203\12\247\22\270\4#"
  ")h(h(h(hX\270\310\222\2\250\15\270\4#Y\222\351\260\221\314\203\12\251\16\270\4Cq"
  "X\310tXH\316\203\12\252\14\270\4#Y\330\310\250&K\14\253\16\270\4CqXH\362\340\220\234"
  "\7\25\254\16\270\4CqXH\316\243F\243\203\12\255\5\0\24\3\256\5\0\24\3\257\5\0\24\3\260"
  "\5\0\24\3\261\5\0\24\3\262\5\0\24\3\263\5\0\24\3\264\5\0\24\3\265\5\0\24\3\266\5\0"
  "\24\3\267\5\0\24\3\270\5\0\24\3\271\5\0\24\3\272\5\0\24\3\273\5\0\24\3\274\5\0\24\3"
  "\275\5\0\24\3\276\5\0\24\3\277\5\0\24\3\300\5\0\24\3\301\5\0\24\3\302\5\0\24\3\303\5"
  "\0\24\3\304\5\0\24\3\305\5\0\24\3\306\5\0\24\3\307\5\0\24\3\310\5\0\24\3\311\5\0\24"
  "\3\312\5\0\24\3\313\5\0\24\3\314\5\0\24\3\315\5\0\24\3\316\5\0\24\3\317\5\0\24\3\320"
  "\5\0\24\3\321\5\0\24\3\322\5\0\24\3\323\5\0\24\3\324\5\0\24\3\325\5\0\24\3\326\5\0"
  "\24\3\327\5\0\24\3\330\5\0\24\3\331\5\0\24\3\332\5\0\24\3\333\5\0\24\3\334\5\0\24\3"
  "\335\5\0\24\3\336\5\0\24\3\337\5\0\24\3\340\5\0\24\3\341\5\0\24\3\342\5\0\24\3\343\5"
  "\0\24\3\344\5\0\24\3\345\5\0\24\3\346\5\0\24\3\347\5\0\24\3\350\5\0\24\3\351\5\0\24"
  "\3\352\5\0\24\3\353\5\0\24\3\354\5\0\24\3\355\5\0\24\3\356\5\0\24\3\357\5\0\24\3\360"
  "\5\0\24\3\361\5\0\24\3\362\5\0\24\3\363\5\0\24\3\364\5\0\24\3\365\5\0\24\3\366\5\0"
  "\24\3\367\5\0\24\3\370\5\0\24\3\371\5\0\24\3\372\5\0\24\3\373\5\0\24\3\374\5\0\24\3"
  "\375\5\0\24\3\376\5\0\24\3\377\5\0\24\3\0\0\0\4\377\377\0";
```

Also note, that your offset is 115 and not 112 for the numbers.

jieweiyang, 2020-04-18:
i got it from "u8g2/tools/font/bdfconv/".

may also share the full command line for your conversion ?

Thanks.

olikraus, 2020-04-18:
I have used exactly the same commandline options, but maybe the windows exe is outdated. I have used the linux binary directly from the latest source.

jieweiyang, 2020-04-19:
will the windows exe be updated? i can see most of them are 2yrs/14months ago.
Or where can get the linux binary?
Thanks.

olikraus, 2020-04-19:
> will the windows exe be updated? 

I do not have access to my windows machine at the moment 

> Or where can get the linux binary?

This is simple: Just run make in the bdfconv dir to create the linux binary. 
https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv
There are no dependencies. It is just plain c code.

olikraus, 2020-04-19:
But actually it should work. The exe is more recent than the files. Not sure what is wrong with the exe :-/

BTW: I have used this commandline:
https://github.com/olikraus/u8g2/blob/442db0b87dbb358992c972012171b285120744c6/tools/font/bdfconv/Makefile#L56

jieweiyang, 2020-04-21:
i ran it in linux, now it works fine. Thanks for the info.

closed 2020-04-21 by jieweiyang