olikraus/u8g2 #1164: Constructor problem for LCD display DEM 240064D FGH-PW (240x64) with UC1608

dannnek, 2020-04-18:
Hi Olikraus, I've a problem with my LCD display DEM 240064D FGH-PW (240x64) with UC1608 controller. During the testing U8g2Logo example from U8g2 library I couldn't run LCD with dedicated constructor:

**U8G2_UC1608_ERC24064_1_4W_SW_SPI u8g2 (U8G2_R2, / * clock = * / 14, / * data = * / 13, / * cs = * / 15, / * dc = * / 0, / * reset = * / 2);** 

instead that was working constructor dedicated for 240x120 display: 

**U8G2_UC1608_ERC240120_1_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ 14, /* data=*/ 13, /* cs=*/ 15, /* dc=*/ 0, /* reset=*/ 2);** 

Unfortunatelly for this setup the flip mode is not working correct.

olikraus, 2020-04-18:
Sure, I have created the new constructor U8G2_UC1607_DEM240064
There is a new beta release 2.28.3
Please test u8g2 flip mode example. Will it work with all rotations?
Once everything works, i suggest to use HW_SPI for max performance.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

dannnek, 2020-04-18:
Hi Olikraus, 
Thank you for your quick response.
I made  test with beta release. The results is similar, the display is not working correct. 
For constructor below: 
U8G2_UC1608_DEM240064_1_4W_SW_SPI u8g2(U8G2_R0, 14, 13, 15, 0, 2);
result:
https://photos.app.goo.gl/ZDjRozqB54XK2nzk6

when I changed U8G2_R0 to U8G2_R2 some data is displayed on the screen

result: 
https://photos.app.goo.gl/4fWQZeZG8uy4UEDW6

Additionally after u8g2.begin();  I had to set contrast  
u8g2.setContrast(155);

dannnek, 2020-04-18:
I tried one more time with constructor of ERC display 
U8G2_UC1608_ERC24064_1_4W_SW_SPI u8g2(U8G0_R0, 14, 13, 15, 0, 2); 
and the results is below: 
https://photos.app.goo.gl/RzDym5rvbLixb3V68
when I changed to U8G0_R2
https://photos.app.goo.gl/jBFKJb9zb93xfQ6DA
it's look like that the displayed area is shifted of a half.

olikraus, 2020-04-18:
ok, lets go step by step.
Please located file u8x8_d_uc1608.c on your harddisc

https://github.com/olikraus/u8g2/blob/60c3417d420b977a14308c6d6bad834ff538078f/csrc/u8x8_d_uc1608.c#L439

change the hex value 0x040 to something else. It can have values between 0x040 and 0x7f

Use this constructor only:
U8G2_UC1608_DEM240064_1_4W_SW_SPI u8g2(U8G2_R0, 14, 13, 15, 0, 2);

Goal is to have the complete logo visible. If you modify this value, then the logo should get shifted up and down. Which value would be needed to see the full logo?

dannnek, 2020-04-19:
Hi Olikraus,
I made the modyfications in u8x8_d_uc1608.c file according your suggestions and the results are listed below:
Optimal value of contrsat settings for the DEM display is U8X8_CA (0x081, 160).
1. With example U8g2Logo value of command 0x040 I changed to 0x07F and the results was better than earlier:
https://photos.app.goo.gl/n25rQxLG2vyvBsp39
but as you can see the displayed logo is not in defined position.

2. As a next step I changed the value of U8X8_C (0x090) to U8X8_C (0x094) and the results is very good.
https://photos.app.goo.gl/T79MZYtR8FKE6eme7 the logo is in defined position.
3. When I tried flip mode u8g2.setFlipMode (1); the display is not working correct
https://photos.app.goo.gl/zmw7eEFtN2WzrB1S6

I've also relized that for default values ​​of command 0x040 and 0x090 and u8g2.setFlipMode (1); display working fine but only with this settings.
I've changed the FlipMode to 0 and some results are below:
u8g2.setDisplayRotation (U8G2_R0); u8g2.setFlipMode (0); draw ( "R0, F0 ');
https://photos.app.goo.gl/un2Nm4fVT1PnoiZGA

 u8g2.setDisplayRotation (U8G2_R0); u8g2.setFlipMode (1); The draw ( "R0, F1");
https://photos.app.goo.gl/r4NPCJDWM7ppPWyHA

Generally everything with F1 works fine.

When I replaced U8X8_C (0x026) to U8X8_C (0x023) more data are visable on the screen in FlipMode 0 :
https://photos.app.goo.gl/VkZdr2rDTg5CXXJTA

olikraus, 2020-04-19:
It is great that you did a lot of testing, but now I am confused. Which command should have which value?
flip mode 0: 
0x040 --> ???
0x090 --> ???
flip mode 1:
0x040 --> ???
0x090 --> ???

dannnek, 2020-04-19:
Sorry for not precise information: 
Flip mode 0 - was working for:
0x040 --> 0x07F
0x090 --> 0x094
and   U8X8_CA(0x081, 160)

Flip mode 1 - was working fine with default values: 
0x040 --> 0x040
0x090 --> 0x090
and   U8X8_CA(0x081, 160)

olikraus, 2020-04-25:
Thanks for the update. I have created beta 2.28.6 which should include the changes.
Maybe you could test this once. Thanks!

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

dannnek, 2020-04-28:
Thanks for update. I made some test with the library, and almost everything looks good but one line is missed in flip mode 0, please look at the attached file. 

https://drive.google.com/file/d/1Hf560D0K5e_MmS56LNXhD3ldj8rOxYvD/view?usp=sharing

olikraus, 2020-05-03:
Thanks for the pictures and your tests. Meanwhile I think the problem can not be solved with the 0x90 command. And also the 0x40 command only allows shift of 63 pixel (where 64 pixel are required).

olikraus, 2020-05-03:
I think there is small bug in the code:
https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/csrc/u8x8_d_uc1608.c#L415
The default x offset should not be 1. Actually the x offset is used as y offset in your display. So the question is:
Will it help to set the x offset to 8 and revert 
https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/csrc/u8x8_d_uc1608.c#L457-L458
to 0x40 and 0x90?
Can you test this? Will this work?

olikraus, 2020-07-05:
not sure how to continue here... closing this for now

closed 2020-07-05 by olikraus