olikraus/u8g2 #1165: fntgrpiconic usage

ramenia, 2020-04-20:
Hi,
I am a windows user and got nothing from build process of fntgrpiconic. How can i use those icons at my project ?

olikraus, 2020-04-20:
WIndows? This project deals with Arduino Microcontrollers. I am not sure how to support here.
If your problem is not related to Arduino and Embedded Displays, you probably whant to look here:
https://github.com/iconic/open-iconic/

ramenia, 2020-04-20:
Haha, you get me wrong i am a windows user who develops embedded project :) just files like .sh or makefile are not for windows. anyway, how i could not found any guide for how to add fntgrpiconic icons to my project

closed 2020-04-20 by ramenia

olikraus, 2020-04-20:
> Haha, you get me wrong

Oh, excause

> I could not found any guide for how to add fntgrpiconic icons to my project, is there any ?

It is just a font. Use the "setFont()" . Weather.ino contains some examples:
https://github.com/olikraus/u8g2/blob/1e3f16790c62571a09a6a6ebaf06d11042b648fd/sys/arduino/u8g2_page_buffer/Weather/Weather.ino#L296-L315