olikraus/u8g2 #1166: u8g2.sendBuffer(); issue

xmodpt, 2020-04-20:
Hello mate it's me again.

I have made all the changes to my code as per your sujestion and all worked greate including creating menus and sub-menus.

My setup includes a stepper and a relay

When i execute the code with "u8g2.sendBuffer();"(@ line 253), my stepper goes funny and verry slowwwww
and i have menus, countdown clock and relay working properly.

If i take out "u8g2.sendBuffer();", the stepper start's to work with a few "gags" and a sec goes by but it works full speed as well as the relay but no countdown counter on display.

Can you shead some light to my issue?
here is my code...

´´´
/***********************************************************
Curing/Washing Box by Xmodpt
Last update @ 15/04/2020  Current Ver: 1.04

****** Version Control *******
1.00 - Countdown tested and working with relay 
1.01 - Added menu with 3 options
1.03 - Change to main menu, for 2 options and added a sub-menu for each of the 1st menu options
1.04 - Added EndStop Buzzer (using PIN7)
***********************************************************/
#include <Arduino.h>
/****** U8G2 LIB *********/
#include <U8g2lib.h>
/***** U8G2 settings *****/
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif
// Include the AccelStepper library:
#include <AccelStepper.h>
// Create a new instance of the AccelStepper class:
AccelStepper stepper = AccelStepper(MotorInterfaceType, 8, 9, 10, 11);

/**** U8G2 Display settings *****/
//U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

/********* Timer Settings ******************/

// 10000ul =   10 sec
// 30250ul =   30 sec
// 60500ul =   1 min
// 301500ul =  5 min
// 600000ul = 10 min
// 900000ul = 15 min
long int arrayTimer [4] = {301500ul, 600000ul, 900000ul, 10000ul}; // set of values 
#define pinRele 6  // Relay Pin Set

char
    szString[20];
byte
    mins, secs;
       
unsigned long
    timeTemp,
    timeNow,
    timeStart,
    timeElapsed,
    timeLeft,
    COUNTDOWN_TIME ;   

// End of constructor list

const char *string_list =
  "Curing\n"
  "Washing\n"
  "\n"
  ".:: About ::.";

const char *string_times =
  "05 min\n"
  "10 min\n"
  "15 min\n"
  "\n"
  ".:: Back ::.";
int buzzer = 7;
uint8_t current_selection = 1;
uint8_t current_selection_times = 1;

void setup(void) {
// ************** STEPPER SETTINGS ****************** 
     stepper.setMaxSpeed(1000);
     //stepper.move(800);  
     stepper.setSpeed(100); 

    pinMode(buzzer,OUTPUT);

// ************** RELAY SETTINGS ********************
     pinMode(pinRele, OUTPUT);
     digitalWrite(pinRele, HIGH);
// ************** Rotary encoder or buttons *********
     u8g2.begin(/*Select=*/ A2, /*Right/Next=*/ A4, /*Left/Prev=*/ A5, /*Up=*/ A0, /*Down=*/ A1, /*Home/Cancel=*/ A3); 
// ************** Font Type config for menu *********
     u8g2.setFont(u8g2_font_6x12_tr);
     u8g2.clearDisplay();
    // Serial.begin(9600);
    
}

void loop(void) {

    u8g2.setFont(u8g2_font_6x12_tr);
    current_selection = u8g2.userInterfaceSelectionList(
    "Curing & Washing OS",
    current_selection,
    string_list);
    
  //****************** CURING MENU *****************  
  if ( current_selection == 1 ) {
    u8g2.setFont(u8g2_font_6x12_tr);
    current_selection_times = u8g2.userInterfaceSelectionList(
    "Curing & Washing OS",
    current_selection,
    string_times);
            if ( current_selection_times == 1 ) { //5min selection Curing UV + Stepper 10rpm    
              timeStart = millis();
              mins = 1;
              secs = 1;
              
              COUNTDOWN_TIME = arrayTimer [3]; // Set for 5 min
              
                while (mins > 0 || secs > 0){
               DoCountdown();                
               relay_action();
               stepper_motor(); 
               
                } 
                buzzer_alarm();
                
            }
            if ( current_selection == 1 && current_selection_times == 2 ) { //10min selection Curing UV + Stepper 10rpm   
              timeStart = millis();
              mins = 1;
              secs = 1;

              COUNTDOWN_TIME = arrayTimer [1]; // Set for 10 min
                while (mins > 0 || secs > 0){
                DoCountdown();                
                relay_action();
                } 
                buzzer_alarm();
            }
              if ( current_selection_times == 3 ) { //15min selection Curing UV + Stepper 10rpm      
          timeStart = millis();
              mins = 1;
              secs = 1;

              COUNTDOWN_TIME = arrayTimer [2]; // Set for 15 min
                while (mins > 0 || secs > 0){
                DoCountdown();                
                relay_action();
                } 
                buzzer_alarm();
            }
    
  }
  //****************** CURING END ******************
  //**************** WASHING  MENU *****************
  if ( current_selection == 2 ) {
        u8g2.setFont(u8g2_font_6x12_tr);
    current_selection_times = u8g2.userInterfaceSelectionList(
    "Curing & Washing OS",
    current_selection,
    string_times);
 
               if ( current_selection_times == 1 ) { // 5min selection Washing Stepper 100rpm    
              timeStart = millis();
              mins = 1;
              secs = 1;
     
              COUNTDOWN_TIME = arrayTimer [0]; // Set for 5 min
                while (mins > 0 || secs > 0){
                DoCountdown();                
                relay_action();
                } 
                buzzer_alarm();
            }
              if ( current_selection_times == 2 ) { // 10min selection Washing Stepper 100rpm     
                timeStart = millis();
                mins = 1;
                secs = 1;
                COUNTDOWN_TIME = arrayTimer [1]; // Set for 10 min
                  while (mins > 0 || secs > 0){
                DoCountdown();                
                relay_action();
                } 
                buzzer_alarm();             
             } 
           if ( current_selection_times == 3 ) { //15min selection Washing Stepper 100rpm      
              timeStart = millis();
              mins = 1;
              secs = 1;
              COUNTDOWN_TIME = arrayTimer [2]; // Set for 15 min
                while (mins > 0 || secs > 0){
                DoCountdown();                
                relay_action();
                } 
                buzzer_alarm();
             }
         }
  //****************** WASHING MENU END *****************
  
  if ( current_selection == 4 ) {
    u8g2.userInterfaceMessage(
  "About:",
  "Curing & Washing OS",
  "design by Xmodpt",
  "ver: 1.04");
    }
   
}
void DoCountdown()
{
  
   //u8g2.clearBuffer();
    static unsigned long
        lastTimeNow = 0;
    static byte
        lastsecs = 1;
       
    timeNow = millis();
    timeElapsed = timeNow - timeStart;
   
    if( mins == 0 && secs == 0 )
        return;
       
    timeLeft = COUNTDOWN_TIME - timeElapsed;

    mins = (byte)(timeLeft / 60000ul);
    timeTemp = timeLeft - (mins * 60000);
    secs = (byte)(timeTemp / 1000ul);
    timeTemp = timeTemp - (secs * 1000ul);

    if( mins == 0 && secs == 0 )
    {
        buzzer_alarm();
        buzzer_alarm();
        buzzer_alarm();
       // u8g2.setFont(u8g2_font_helvB24_tf );
        u8g2.clearDisplay();
        u8g2.setCursor( 20, 45 );
        sprintf( szString, "DONE" );
        u8g2.print( szString );
   //     Serial.println( szString );
       // delay (2000);
     u8g2.sendBuffer();
    }
    else if( secs != lastsecs )
    { 
        //u8g2.setFont(u8g2_font_helvB24_tf );
        lastsecs = secs;
        u8g2.setCursor( 20,45 );
        sprintf( szString, "%02d:%02d", mins, secs );
   //     Serial.println( szString );
        u8g2.clearBuffer();
        u8g2.print( szString );
        
     }
   
   u8g2.sendBuffer(); // <-------------------------- this is my problem

  
}

//****** VOIDS ********//

//********** AccelStepper **************
void stepper_motor() { 

   if ( mins == 0 && secs == 0 ) { 
    
       stepper.stop();
      
   } else {
 
       stepper.runSpeed();
   }
}
//**************************************

void relay_action(){
  
  if ( mins == 0 && secs == 0 ) { 
    
      digitalWrite(pinRele, HIGH);
      
      
   } else {
       digitalWrite(pinRele, LOW);
      
   }

}

void buzzer_alarm() {
    // Sounds the buzzer at the frequency relative to the note C in Hz
    tone(buzzer,5000, 500);    
    // Waits some time to turn off
    delay(1100);
    //Turns the buzzer off
    noTone(buzzer); 
    // Sounds the buzzer at the frequency relative to the note D in Hz   
   return;
}
´´´

olikraus, 2020-04-20:
Will the problem also occur with U8G2_SH1106_128X64_NONAME_1_SW_I2C? (Software emulated I2C?)

xmodpt, 2020-04-20:
Hello

I will try that one and let you know

xmodpt, 2020-04-20:
If i use
```
U8G2_SH1106_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```
i get this error

```
exit status 1
no matching function for call to 'U8G2_SH1106_128X64_NONAME_1_SW_I2C::U8G2_SH1106_128X64_NONAME_1_SW_I2C(const u8g2_cb_t*, int)'
```

olikraus, 2020-04-20:
Obviously, because you need to add clock and data pin numbers, as specified here: 

https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1106-128x64_noname

You probably need to scroll down to your constructor.

olikraus, 2020-04-20:
U8G2_SH1106_128X64_NONAME_1_SW_I2C(rotation, clock, data [, reset])

xmodpt, 2020-04-20:
ok some results
if i use U8G2_SH1106_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, A5, A4);
i have NO countdown showing and the stepper still goes to slow

if i use U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, A5, A4);
I have COWNTDOW showing but it goes even slower...

xmodpt, 2020-04-20:
if i use U8G2_SH1106_128X64_NONAME_1_SW_I2C(rotation, clock, data [, reset]);
i get this error

´´´
exit status 1
expected ')' before ',' token
```

xmodpt, 2020-04-20:
Please note: with the SW version all display get's to much lag

xmodpt, 2020-04-20:
> Obviously, because you need to add clock and data pin numbers, as specified here:
> 
> https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1106-128x64_noname
> 
> You probably need to scroll down to your constructor.

I was looking in to your link and i can't get my head arround it.
my oLed has 4 pin and i am using SCA+SCL outputs from my UNO.... as allways and it worked so far

olikraus, 2020-04-20:
I assume 
```
U8G2_SH1106_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, A5, A4);
U8G2_SH1106_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, A5, A4);
```
are correct in your case.

The question is more, will your "stepper go funny".

So your answer is: The display is slow and the stepper is also slow, I would assume your controller (Uno?) not powerful enough to do these things in parallel.

Still there might be a kind of a conflict between the stepper lib and u8g2.
I do not know the stepper lib, so I can not say more than this :-(

xmodpt, 2020-04-20:
any sugestion???

olikraus, 2020-04-20:
Of course you should continue with "U8G2_SH1106_128X64_NONAME_1_HW_I2C", anything else, I do not know. There is a kind of a conflict between the libs. Maybe ask the author of the Servo lib.

xmodpt, 2020-04-20:
I will try with dif libs.

Thank you for your help and I will post any results

JamesGKent, 2020-05-13:
just had a quick look at the AccelStepper library:
```
    /// Constructor. You can have multiple simultaneous steppers, all moving
    /// at different speeds and accelerations, provided you call their run()
    /// functions at frequent enough intervals. 
```

so i think it simply boils down to the sendbuffer command takes too long and you aren't calling the run function on the stepper often enough. there isn't an easy way to solve this, you could try switching to one of the page constructors and calling the run between each page, but thats still likely to be too slow. otherwise you need some sort of concurrency like threads, or possibly one of the ticker libraries

closed 2020-07-05 by olikraus