olikraus/u8g2 #1167: missing bitmap display?

JamesGKent, 2020-04-22:
Hi @olikraus 

here:
https://github.com/olikraus/u8g2/blob/1e3f16790c62571a09a6a6ebaf06d11042b648fd/csrc/u8g2.h#L1392
seems to reference a bitmap display controller, which i'm guessing is like the null display but actually renders to memory, its also referenced right at the end of https://raw.githubusercontent.com/olikraus/u8g2/master/cppsrc/U8g2lib.h

but it doesn't seem to exist?

the file mentioned in the comments where it is supposed to reside (u8x8_d_bitmap.c) doesn't seem to exist at all.

Is this intentional?

olikraus, 2020-04-22:
Yes, the bitmap device was not intended for embedded controller and Arduino IDE as it references malloc() which often does not exist.
https://github.com/olikraus/u8g2/blob/1e3f16790c62571a09a6a6ebaf06d11042b648fd/sys/bitmap/common/u8x8_d_bitmap.c#L46

It should be possible to just include the device if you have the required includes (stdio, stdlib): 
https://github.com/olikraus/u8g2/blob/master/sys/bitmap/common/u8x8_d_bitmap.c

There is also a plain-c example:
https://github.com/olikraus/u8g2/blob/master/sys/bitmap/hello_world/main.c

JamesGKent, 2020-05-01:
Understood. Thank you.

closed 2020-05-01 by JamesGKent