olikraus/u8g2 #1168: u8g2 wiki fonts page

dingo27mobile, 2020-04-24:
Hello,

this is not primarily issue with library but with Github - I use u8g2 wiki often to search for fonts, and recently there are so many fonts that Github puts a "You have triggered an abuse detection mechanism." This is for page https://github.com/olikraus/u8g2/wiki/fntlistall. Also, images for bigger fonts do not load correctly.
Fonts do not have preview from line fntpic/u8g2_font_t0_12b_tf_short.png u8g2_font_t0_12b_tf UW ttyp0, i am not sure if it is only me or other people as well. I am sure before it worked ok.

After the mechanism enables, there is no way to look for any page on github.

I know this is a lot of work but would be helpful to split page in more articles. Thanks in advance.

BTW you lib is awesome.

olikraus, 2020-04-24:
For me it works (again?)

Maybe it is related to this: 
[image]

dingo27mobile, 2020-04-24:
Actually this is happening for me for few weeks... 

[image]

But if it works on your side there must be mistake on my side.

closed 2020-04-24 by dingo27mobile

olikraus, 2020-04-24:
Hmmm....