olikraus/u8g2 #1170: How can I detect that there is no "u8g2.firstPage" being executed?

JailsonBR, 2020-04-25:
Hello,

I'm doing an open system project, I'm trying to develop some pop-up windows, I managed to make these pop-ups when some u8g2.firstPage (); is running, however when there is no u8g2.firstPage (); being executed I cannot create any logic to call it.

I'm trying to figure out how I can detect that there is no "u8g2.firstPage ();" being executed at the moment, I tried to read through the buffer but the command u8g2.getBufferSize is not working, and even trying to getBufferTileHeight and getBufferTileWidth I can't use them to call another u8g2.firstPage (); if they are at 0.

My project can be followed in:
[https://github.com/JailsonBR/XPsys](https://github.com/JailsonBR/XPsys)
[https://xprojetos.net/xpsys-sistema-para-esp32-com-st7920/](https://xprojetos.net/xpsys-sistema-para-esp32-com-st7920/)

olikraus, 2020-04-25:
Create a function to ecapsulate the firstPage call and use a global variable;
```
int is_first_page_called = 0;
void myFirstPage() {
  is_first_page_called = 1;
  u8g2.firstPage();
}
```

if is_first_page_called is not 0, then firstPage was called.

JailsonBR, 2020-04-25:
Is there no way to read from the buffer? if it is empty does it return something?

olikraus, 2020-04-25:
There is a function to get the pointer to the buffer.

closed 2020-07-05 by olikraus