olikraus/u8g2 #1171: Just rendering to memory

YoHidden, 2020-04-26:
Hi,

I have a display with a manufacturer library which I have no internal access to. This library is customized to my board and just provide an initialization function and a full update display function.
In order to use the u82g library, should I setup the null device to just render to the memory buffer? This is, I do not need to control the display through SPI interface (already in the manufacturer API), but just a reference to the screen buffer in order to update it to the display.
Thanks.

olikraus, 2020-04-26:
You might use the u8x8 bitmap device. The was an issue regarding this recently.

YoHidden, 2020-04-26:
Thanks. Then I think that the example at "sys\bitmap\hello_world" can work for my case, can't it ?

Since I will initialize the display completely independent from u8g2 lib, do I still need to call the functions  _u8x8_InitDisplay(...)_ and _u8x8_SetPowerSave(...)_.

olikraus, 2020-04-26:
It should work. Initdisplay is still needed (I think)

YoHidden, 2020-04-26:
Thanks, it worked.
Now I just only have to re-arrange the mapping of the pixels in the buffer to match the pixel/byte order expected by my display lib. So, I should group pixels/byte in columns and not in lines. I mean, right now each byte groups the 8 pixels in 8 consecutive lines. But I need 1 byte to group 8 pixels in 8 consecutive columns.

olikraus, 2020-04-26:
Sounds good.

> Now I just only have to re-arrange the mapping of the pixels in the buffer
In
https://github.com/olikraus/u8g2/blob/master/sys/bitmap/common/u8x8_d_bitmap.c#L268
you could try to replace "u8g2_ll_hvline_vertical_top_lsb" with "u8g2_ll_hvline_horizontal_right_lsb".
Maybe this fits better.

YoHidden, 2020-04-26:
Perfect !!  It worked. Thanks a lot. You have made my day.

One last question. In my application, I will not use the u8x8... do I still need to allocate memory for the u8x8 bitmap ?  I mean to save some memory.

olikraus, 2020-04-27:
Hmmm you could try to transfer directly to your display instead of allocation of the bitmap.

YoHidden, 2020-04-27:
But that’s the problem. My lib only allows me to update the _full_ display at once from a buffer in memory. That’s the reason I needed to render directly to memory, so I first build the image in my memory an then it is sent over to update the display.
And that was my question about if the double allocation for u8g2 and u8x8 still needed if only going to use the first one (draw lines, circles, strings, ...).

olikraus, 2020-04-27:
U8x8 is the low level part of u8g2 lib.
U8g2 API inself is build on top of U8x8

U8x8 itself does not require any memory allocation. Instead U8x8 writes data directly to the display controller: So whatever U8x8 receives is written directly to the display.
U8x8 works with "tiles". One tile is kind of the minimal common data block on almost all the displays. It is a small quadratik area of 8x8 pixel (64 pixel all together, these are 8 bytes).
So: U8x8 can write this 8x8 "pixel icon" to a give place in the memory of a display controller.

As I said, U8g2 builds on top of it: U8g2 allocates memory to do the drawing more efficient. It can indeed allocate the complete bitmap, render everything in that bitmap and then transfers the content to U8x8. U8x8 in turn transfers everything to the display.

In case of the bitmap device, U8x8 can not do the transfer and as  a result, u8x8 bitmap device as to allocate the memory.

Indeed the memory is allocated twice. 
Mybe the u8x8 memory is not required. You could just receive the DRAW_TILE message and transfer the data to your display instead.

YoHidden, 2020-04-27:
Ok, I've removed the u8x8 buffer (some RAM saved) and changed the DRAW_TILE message, and it seems to still work ok (...no crashes ;-). Thanks.

But my display (and lib) can't stop surprising me. It seems now that, although the line bytes are ordered left-to-right, the pixels grouped in each line byte are reversed, this is, the left-hand pixel corresponds to the least-significant bit (lsb) in that byte.

I can copy the bitmap buffer to a new one reversing the bits so my display is happy, but is there any configuration in the u8g2 to assign the bits in this reversed order ?

YoHidden, 2020-04-27:
Well, I have just modified the function mentioned before ("u8g2_ll_hvline_horizontal_right_lsb"), so the internal _mask_ generated to manage the bits goes from 0x01 to the left, in place of going from 0x80 (128) to the right. And it seems to be working...

closed 2020-07-05 by olikraus