olikraus/u8g2 #1172: Template for new driver

nopbit, 2020-04-27:
Hello,

Thanks for this wonderful library :) I would like to add support for a screen which i removed from an old telephone. It is using chip on glass S1D15E06 with 160x100dots resolution(4 gray scale). Display uses 4 pins Serial Input, Serial Clock, Command/Data and CS pin. 

I created a simple test library with arduino and screen is working perfectly with it. Is there any driver template that i can change and add support for this display?

Best Regards
Mustafa

Datasheet for S1D15E06.
https://www.rockbox.org/wiki/pub/Main/DataSheets/epson_s1d15e06.pdf

olikraus, 2020-04-27:
This device might be a good candidate:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1608.c#L473

nopbit, 2020-04-28:
I edited u8x8_d_uc1608_dem240064_init_seq procedure according to my lcd driver. Now i want to learn how to send data to LCD ? My sample code is like below. I am sending data with row & column. I believe i must play with msg type U8X8_MSG_DISPLAY_DRAW_TILE right?

_My sample code:_

    uint8_t row = 0;
    uint8_t col = 0;
    uint8_t data=0;
    
    _lcd_cmd2(LCD_CMD_PAGE_ADDR,1+(row<<1));
    _lcd_cmd2(LCD_CMD_COL_ADDR,(6*col));
    _lcd_cmd1(LCD_CMD_DATA_WRITE);
     data = 0x0F;
     shiftOut(PIN_MOSI, PIN_SCK, MSBFIRST, b);

> https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1608.c#L473

olikraus, 2020-04-28:
There is basically no nied to use your own functions.
Instead there are several low level functions which allow hardware independent transfer to the display. They also include an abstraction layer, so that a display will work with multiple busses (I2C, 8-bit 4-wire SPI, 9-bit 3-wire SPI, parallel, etc).

The functions are listed here:
https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/csrc/u8x8.h#L597-L603

The first call should go to "u8x8_cad_StartTransfer". 
After this use any data transmit calls (u8x8_cad_SendCmd, u8x8_cad_SendArg, ...)
Last call should be "u8x8_cad_EndTransfer". 

"cad" means "command argument data".

StartTransfer/EndTransfer will clear/set the CS flag (SPI) or does similar action on I2C.

All in all, your code should look like this:
https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/csrc/u8x8_d_uc1608.c#L78-L101

nopbit, 2020-04-28:
during the init process, i couldn't able to switch D/C pin between command to data (Low Command, High Data). DC pin is always low. I checked it with logic analyzer. Is there any special command to active D/C pins output? Thanks in advance

a part from init process:
`  U8X8_CA(0x66, 0x0),           	/* (15) Display Mode, Parameter 0 (4 Gray Scale) 1 (Binary) */
  U8X8_CA(0x39, 0x43),            	/* (16) Gray Scale Pattern Set, Pattern */
  U8X8_C(0xBE|1),          	  	/* (2) Display Off Mode (0 VSS / 1 Vcc) */
  `

nopbit, 2020-04-29:
I forgot to mention: i tested with U8X8_UC1608_DEM240064_4W_SW_SPI and it is using u8x8_cad_001 as default. Do you have any idea what i am doing wrong?

olikraus, 2020-04-29:
There are actually three different kind of bytes which need to be transfered:
Commands
Arguments
Data
(thats where "cad" comes from)
An argument is an extra byte send after a command.
Data is really the bitmap data, which must be written to the display memory.
The DC (Data Command) or RS (Register Select) or CD (Command Data) line can carry only two information and needs to indicate what is what. Unfortunately each display controller has its one specification. Sometimes it is like this:

Commands: DC must be 0
Arguments: DC must be 1
Data: DC must be 1
In this case you have to use u8x8_cad_011 (you get the naming convention?)

However, in your case i think it is u8x8_cad_001, which means:
Commands: DC must be 0
Arguments: DC must be 0
Data: DC must be 1
This means, DC line will be set to 1 only if pure pixel data is transfered.

As mentioned above, there are basically three different functions:
 * u8x8_cad_SendCmd: Use this for command bytes
 * u8x8_cad_SendArg: Use this for arguments of command bytes
 * u8x8_cad_SendData: Use this for pixel data

These functions will automatically set the DC line according to the used CAD callback function.
So if you use u8x8_cad_001, then only u8x8_cad_SendData will set the DC line to 1. Others will clear the DC line.

The idea of all this is to write the driver code without knowing details on the DC line. Instead only the correct callback function for the DC line has to be selected.

nopbit, 2020-04-30:
Thanks for the support :),  i am able to create u8x8 graphictest run on display without any problem. I will try u8g2 with it.

nopbit, 2020-04-30:
I couldn't figure out how to choose hardware spi pins. I am using STM32F103C8T6 with arduino and this chip has 2 spi. For spi1 it is using 2 different ports. How can i choose which pin to use for MOSI and SCK ? Thanks

olikraus, 2020-05-03:
For many uC you can not choose MOSI and SCK. Instead those pins are hard wired in your microcontroller.

olikraus, 2020-05-17:
take code from pull request #1190 and create a device for this. I also think the constructor needs to be renamed.

nopbit, 2020-05-27:
I can rename it, do you have a suggestion for naming? Thanks

olikraus, 2020-06-05:
Excause my delay in working on this. My current personal situation does not allow me to spend much time on u8g2. 
In order to work with codebuild.c, the second keyword after u8x8 should be the controller name.

nopbit, 2020-06-22:
Sorry for delay too. I will make the changes at first chance. Can i compile it in windows with gcc? Do i need a special tool for codebuild.c

olikraus, 2020-06-22:
codebuild.c is plain c code, so any compiler should be valid.
Difficulties will only come from the different file IO handling compared to linux (CR vs CR/LF etc).

olikraus, 2020-12-25:
added constructor U8G2_S1D15E06_160100_1_4W_HW_SPI

closed 2020-12-25 by olikraus