olikraus/u8g2 #1174: doesnt support chip select on ERM19264-3 display

abusakib, 2020-04-28:
here i added the data sheet
https://www.buydisplay.com/download/manual/ERM19264-3_Series_Datasheet.pdf

olikraus, 2020-04-28:
It might still work if you do not connect cs0.

abusakib, 2020-04-28:
i connect cso and cs1

abusakib, 2020-04-28:
still facing the same problem

abusakib, 2020-04-28:
[image]
see the status

olikraus, 2020-04-28:
You may try different combinations.
How does your constructor look like?

closed 2020-07-05 by olikraus

m3mory, 2022-08-02:
for the ERM19264-3 Displays the CS Lines need to be changed, as there are only two CS lines available.
This worked for me:
if ( v.x < 8 )
      {
	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 4, NULL);  //6 -->4
	u8x8_ks0108_out(u8x8, &v, arg_ptr);
	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_END_TRANSFER, 7, NULL);
      }
      if ( v.x < 16 )
      {
	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 6, NULL); //5 --> 6
	u8x8_ks0108_out(u8x8, &v, arg_ptr);
	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_END_TRANSFER, 7, NULL);
      }
      if ( v.x < 24 )
      {
	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 5, NULL); //  3 -->5
	u8x8_ks0108_out(u8x8, &v, arg_ptr);
	u8x8->cad_cb(u8x8, U8X8_MSG_CAD_END_TRANSFER, 7, NULL);
      }

olikraus, 2022-08-02:
Thanks for the update :-)