olikraus/u8g2 #1175: Modify library so that Teensy devices can use i2c_t3 instead of Wire

St0RM53, 2020-04-28:
So i had made a project in the past using a Teensy 3.2 and oled connected to 2nd i2c interface. Today i wanted to use the library to do some tests and i found out my sketch wouldn't compile because i was using the latest library. 

Basically on Teensy devices if you want to use the 2nd i2c interface Wire1 doesn't work. You need to use i2c_t3 in place of Wire: https://github.com/nox771/i2c_t3

First issue is that Wire.h is included in U8x8lib.cpp
Since you need to use i2c_t3 in place there as well you need to account for that.

Secondly U8X8_HAVE_2ND_HW_I2C in doesn't work and i didn't dive deep enough to find exactly why. I see there is already an issue open regarding this: https://github.com/olikraus/u8g2/issues/779

Workaround for this case is to go to U8X8_HAVE_HW_I2C and replace all instances of "Wire." to "Wire1."

olikraus, 2020-04-29:
There is kind of a standard way to allow and enumerate multiple I2C devices. Unfortunately all none-Arduino boards invent their own mechanism. 
Why couldn't ESP and Teensy Board developer just follow the examples used in Due and Mega boards?

As a lib developer I would like to have a clear HAL which I can use. I do not want #ifdef's for each different board in my lib.

St0RM53, 2020-04-29:
Yes i understand that. It actually supports the Wire library as you can see here: https://www.pjrc.com/teensy/td_libs_Wire.html

The issue from what i remember is when you wanted to use the 2nd i2c interface (aka Wire1)..that might have been fixed since then but i thought to share my workaround at least so other people know.

If you want me to test let me know exactly how to implement the standard way you say as in my testing U8X8_HAVE_2ND_HW_I2C  didn't work.

olikraus, 2020-04-29:
I do not know whether there is a proper standard document, but all original Arduino libs define the constant WIRE_INTERFACES_COUNT.

If WIRE_INTERFACES_COUNT is equal to 1, then only a "Wire" object exists. 
If WIRE_INTERFACES_COUNT is equal to 2, then there are "Wire" and "Wire1" objects.
If WIRE_INTERFACES_COUNT is equal to 3, then there are "Wire", "Wire1", and "Wire2" objects.

This is checked in U8x8 (although only Wire and Wire1 are supported by u8x8/u8g2)
https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/cppsrc/U8x8lib.h#L110-L116

olikraus, 2020-04-29:
As an example, here is the code for the SAM based Wire:
https://github.com/arduino/ArduinoCore-sam/blob/eed66e7977b4037f49bbcd832437d4bcf72c0f3f/libraries/Wire/src/Wire.cpp#L377
and
https://github.com/arduino/ArduinoCore-sam/blob/eed66e7977b4037f49bbcd832437d4bcf72c0f3f/libraries/Wire/src/Wire.cpp#L415

St0RM53, 2020-05-03:
I can try it again but i don't think it has WIRE_INTERFACES_COUNT defined.
Seems to have been mentioned in the past here as well: https://github.com/olikraus/u8g2/issues/483

olikraus, 2020-05-03:
Not sure what else I can say.... Board software programmers should agree on a common standard.

atestani, 2020-07-21:
OK... if provision can't be made for Teensy i2c_t3 then can the library be fixed so if an SPI display is being used, collisions of the Wire and i2C_t2 libraries don't occur?  IOW, why does the u8g2 library do _anything_ with I2C if I am use an SPI display constructor (U8G2_SSD1309_128X64_NONAME2_1_4W_HW_SPI u8g(U8G2_R0, 10, 9, 8);)

In my specific case, I am using i2c_t3 for 2 external devices and 4-wire SPI for the display.   If I remove, the references and includes for the external devices, my test program compiles and runs fine.  If I have the external devices in, then I get the following and the build fails.  I am using VSCode/PlatformIO which sees the following dependencies (note the Wire.h for u8g2 - why is it there??) :
```
Dependency Graph
|-- <U8g2> 2.28.6
|   |-- <Wire> 1.0 
|   |-- <SPI> 1.0
|-- <extEEPROM-master>
|   |-- <i2c_t3>
|-- <EEPROM> 2.0
```

Here is the build/linking failure:
```
Linking .pio\build\teensy31\firmware.elf
.pio\build\teensy31\lib8fe\libWire.a(WireKinetis.cpp.o): In function 
`i2c0_isr':
WireKinetis.cpp:(.text.i2c0_isr+0x0): multiple definition of `i2c0_isr'
.pio\build\teensy31\libc14\libi2c_t3.a(i2c_t3.cpp.o):i2c_t3.cpp:(.text.i2c0_isr+0x0): first defined here
c:/users/al testani/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions    
.pio\build\teensy31\lib8fe\libWire.a(WireKinetis.cpp.o): In function 
`i2c1_isr':
WireKinetis.cpp:(.text.i2c1_isr+0x0): multiple definition of `i2c1_isr'
.pio\build\teensy31\libc14\libi2c_t3.a(i2c_t3.cpp.o):i2c_t3.cpp:(.text.i2c1_isr+0x0): first defined here
.pio\build\teensy31\lib8fe\libWire.a(WireKinetis.cpp.o):(.data.Wire1+0x0): multiple definition of `Wire1'
.pio\build\teensy31\libc14\libi2c_t3.a(i2c_t3.cpp.o):(.bss.Wire1+0x0): first defined here
c:/users/al testani/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol `Wire1' changed from 20 in .pio\build\teensy31\libc14\libi2c_t3.a(i2c_t3.cpp.o) to 108 in .pio\build\teensy31\lib8fe\libWire.a(WireKinetis.cpp.o)
.pio\build\teensy31\lib8fe\libWire.a(WireKinetis.cpp.o):(.data.Wire+0x0): multiple definition of `Wire'
.pio\build\teensy31\libc14\libi2c_t3.a(i2c_t3.cpp.o):(.bss.Wire+0x0): first defined here
c:/users/al testani/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Warning: size of symbol `Wire' changed from 20 in .pio\build\teensy31\libc14\libi2c_t3.a(i2c_t3.cpp.o) to 108 in .pio\build\teensy31\lib8fe\libWire.a(WireKinetis.cpp.o)
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\teensy31\firmware.elf] Error 1
=================== [FAILED] 
```

BTW u8glib doesn't have this problem.  I would like to upgrade to u8g2 but due to this problem I cannot.  

Any ideas?

olikraus, 2020-07-26:
> BTW u8glib doesn't have this problem. I would like to upgrade to u8g2 but due to this problem I cannot.

U8g2 did bypass the standard libs. This was one of the reasons to rewrite u8glib. I wanted to use standard Arduino libs. 

> why does the u8g2 library do anything with I2C

It does not do anything, but code needs to be included because it might be used. Moreover there is an issue in the Arduino Wire lib, which not only declares but also defines a wire object. 

You can fully disable I2C and SPI support in U8g2 by commenting the statements below in your local copy:

https://github.com/olikraus/u8g2/blob/0966f30f7daeb9249f1a4d2042460932a0b2ac48/cppsrc/U8x8lib.h#L56-L61

atestani, 2020-07-26:
Thanks... That is just what I was looking for.  Unfortunately, I am not having much luck getting it to work. 

If I comment out one or both of the #defines, I still get both libraries included as follows
```
Dependency Graph
|-- <EEPROM> 2.0
|-- <U8g2> 2.28.6
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <extEEPROM-master>
|   |-- <i2c_t3>
```
I am using PlatformIO and do not have the u8g2 library "installed" but rather have the entire library in my local /lib folder.  PIO seems to be finding everything properly including the u8x8lib.h file I modified.  I also checked thru various *.cpp files and cannot find any place that is calling Wire.  

I'm continuing to try to find why Wire is still being included but thus far I haven't found it.

commit 97b654cf (2022-01-12): Add support for i2c-Library of Teensy T3/LC fixes #1175
commit ef515e6c (2022-01-12): Merge pull request #1408 from jogi-k/Add_Teensy_T3_Support

Add teensy t3 support fixes #1175
closed 2022-01-12 by olikraus