olikraus/u8g2 #1176: JLX19264G-925 spi display support

PhoenixMarie, 2020-04-29:
Hello, is it possible to use your library with this type of display? http://www.jlxlcd.cn/html/zh-detail-1038.html  (JLX19264G-925 , spi connection). Thank you in advance.

olikraus, 2020-04-29:
The metioned controller seems to be ST7525
The support for this controller can be checked here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

So the answer is: No, this controller is not supported at the moment.

closed 2020-07-05 by olikraus

5B4ALR, 2026-09-20:
Hello, and thanks for your terrific job! 
Is ST7525 ever supported by any version of your library? I found a code that makes it show stuff on the display, but is useless for my very little knowledge, unless it is supported by a normal library..
Thanks,
Stelios.