olikraus/u8g2 #1177: ld7032 60x32 constructors

bturner1273, 2020-04-29:
Okay so your hardware spi for ld7032 definitely works, I just got it going but I was having trouble with every other constructor for the ld7032. Also I was wondering what 2ND means in a constructor. 

I am using an arduino micro with the seed studio ocell .5 inch oled. And followed some of this tutorial (till it specified pins that implied using the software spi constructor that didn't work for me)
[https://www.seeedstudio.com/document/pdf/OCELL_use_guide2.pdf](url)

I didn't take much time to look into the code but I did see there was that issue (#865) you had previously with Majestix1970 and I couldn't get the sw spi or hardware or software i2c working. Could it be related?

bturner1273, 2020-04-29:
nevermind I am stupid

closed 2020-04-29 by bturner1273

olikraus, 2020-04-29:
2nd refers to a second hardware which is there on some boards.