olikraus/u8g2 #1178: SSD1309 128x64 NONAME0 not working 8080 mode

JamesGKent, 2020-05-01:
Hi @olikraus 
I have a display: https://uk.farnell.com/midas/mcot128064hv-bm/oled-display-blue-2-42-inch/dp/2522949

Which is using the SSD1309 controller. It is supposed to support I2C, 4 wire SPI, 6800 and 8080 parallel interfaces.

I've used this on something previously where I was able to drive it using I2C and that worked fine, however with what i'm not working on I can't use HW I2C so I was hoping to use the 8080 parallel interface as I believe it should be faster than SW I2C or SW SPI and thats what the previous screen on my intended device was.

I'm currently testing using an arduino pro micro (32u4) to try and get it working and I can't. so far all i get is a black screen.
I have 13V from a lab PSU for the OLED PSU, and 5V from the same PSU (common ground) for the logic supply.
I have a 1M resistor fitted between IREF and 0V as per the datasheet, I also have the recommended capacitors fitted on the supply rails and VCOMH.
Both mode select pins are connected to 5V which should put it in 8080 mode.

the constructor I'm using is:
`U8G2_SSD1309_128X64_NONAME0_F_8080 u8g2(U8G2_R0, 9, 8, 7, 6, 5, 4, 3, 2, 10, 16, 14, 15);`

the RD pin is connected to 5V

I've tried to have a look for other people using this display in this mode but I haven't seen much evidence, is this something you've been able to test?

JamesGKent, 2020-05-01:
just to add, SW I2C works, as does 6800 mode with the following constructors:
```
// SW I2C										 scl, sda, reset
//U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(U8G2_R0, 9, 8, 15);

// 6800 parallel							  // D0                 -D7, enable, cs, dc, reset
U8G2_SSD1309_128X64_NONAME0_F_6800 u8g2(U8G2_R0, 9, 8, 7, 6, 5, 4, 3, 2, 10,     16, 14, 15);
```

for the 6800 mode I swapped R/W of the display for E/RD, and connected the R/W pin to 0V.

still puzzled why 8080 mode doesn't seem to work.

olikraus, 2020-05-03:
Hmm  good question. I think you modified the mode pins also..
Maybe there is still a wiring issue.

JamesGKent, 2020-05-04:
I did also modify the mode pins as per the datasheet. unfortunately I'm out of time to look into this when I have the display working as needed. I suspect this is an issue with the display/wiring and not the library so I will close this issue.

closed 2020-05-04 by JamesGKent

bijx-ly, 2024-08-16:
How to implement the functions u8x8_gpio_and_delay() and u8x8_byte_8bit_8080mode() when using 8080 protocol