olikraus/u8g2 #1183: Data format for ST7586S based display (YMC240160)

Schnulli87, 2020-05-05:
Hello,
I'm trying to get a ST7586S based display up and running, as I have a few of them here.
The model number is YMC240160.

I tested both available display constructors:
The ERC240160 gets the display height correct, but the width has issues.
The s028hn118a at some point got the width correct, but the height was incorrect and shifted.
At the moment I can no longer get output with the s028.. constructor at all. The first positive results seem to be caused by not fully powering down the circuit after each reprogramming. Somehow, the display seems to store some data/settings, even if the HW reset pin is toggled.

So, to continue, I used the ERC.. constructor and modified the settings to have the image not mirrored or upside down.
Unfortunately, I cannot get the display to continuously display the image from left to right.
The ST7586S packs 3 pixels into 1 Byte in monochrome mode (2+x+2+x+2 Bit).
The ERC240160 additionally seems to skip every 3rd output line to the display, so an empty pixel is inserted in the data sent to the display. The result on the YMC240160 looks like this:

[image]
The line should be diagonal and the rightmost column should be black.
Detail:
[image]

To fix this, I used the data "translation" of the s028hn118a, which does not skip internal outputs and should be what I need.
Unfortunately, the output is the following:
[image]

I tried to change the way that ptr and c are modified, but it seems, I do not fully understand the correct function and use of all the variables/functions (ptr, c and u8x8_cad_SendData).
The best I could achieve, are blocks of 24 columns correctely displayed, but each block repeats the last previous 8 columns:
[image]

How do I need to call u8x8_cad_SendData (I was too stupid to find its sourcecode; what is the 2nd parameter?) and how do I have to modify ptr and c correctly?
The original function of s028hn118a is this:
```
while (c > 0) {
  input = (((uint32_t)ptr[0] << 16) | ((uint32_t)ptr[1] << 8) | (uint32_t)ptr[2]);
  for (i=0; i<8; i++)
  {
    byte = 0;
    if (input & 0x800000)          // if bit 23
        byte = byte | 0xC0;  //set pixel 1
    if (input & 0x400000)          // if bit 22
        byte = byte | 0x18;  //set pixel 2
    if (input & 0x200000)          // if bit 22
        byte = byte | 0x3;  //set pixel 3
    output[i] = byte;
    input <<= 3;
  }
  u8x8_cad_SendData(u8x8, 8, output);
  ptr += 3;
  c -= 3;
}
```
The output is as described above. The picture with 24 correct columns only has the modification:
```
ptr +=2;
c-=1;
```
Other values for ptr and c counting result in worse output.

Thank you for your help!

olikraus, 2020-05-10:
Can you please send a picture with both different constructors, the ERC240160 and the S028HN118A?
Please note that you need to enable U8g2 16 bit mode for the S028HN118A (you can also keep the 16 bit mode enabled for the ERC240160): https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/doc/faq.txt#L163

Schnulli87, 2020-05-13:
Sure, no problem.
I put all the original files back in place and made some pictures:

The original constructor:
`U8G2_ST7586S_ERC240160_F_4W_HW_SPI u8g2(U8G2_R0, 5, 4, 16 );`

[image]

With mirrored image:
`U8G2_ST7586S_ERC240160_F_4W_HW_SPI u8g2(U8G2_MIRROR, 5, 4, 16 );`

[image]

Now with the S028HN118A constructor:
`U8G2_ST7586S_S028HN118A_F_4W_HW_SPI u8g2(U8G2_R0, 5, 4, 16 );`
(I have added another vertical line in the last column to show that the whole width is addressed)

[image]

Now using
`U8G2_ST7586S_S028HN118A_F_4W_HW_SPI u8g2(U8G2_MIRROR, 5, 4, 16 );`
and additionally
`u8g2.setFlipMode(true);`
results in an almost perfect display (ony the active rows seem to be wrong):

[image]

By modifying the scan direction registers in the u8x8_d_st7586s_s028hn118a.c , I can correct the mirrored display without using `U8G2_MIRROR ` and `u8g2.setFlipMode(true)`.
What I can not achieve is the missing display area at the top.
Copying the start/end COL values from the ERC240160 constructor only results in the image being shifted up in the active area. But the rows at the top remain inactive and the missing rows at the bottom are not moved into view, but filled with garbage:

[image]

olikraus, 2020-05-17:
I think the missing part is the actual height:
https://github.com/olikraus/u8g2/blob/9ecb87c4b62e91302006d278fc3498263a6c567d/csrc/u8x8_d_st7586s_s028hn118a.c#L207-L212
tile_height should be 20
and
pixel_height should be 160
Moreover tile_width should be 30 and pixel_width should be 240
Will this help?

Which exact value did you use for the scan direction register?

Schnulli87, 2020-05-19:
Scan register must be set to 0x80. 0x00 is Y inverted,
Thanks for the hint with the hardware information. This (together with the correct start and end values for the scan registers) correctly moves the image to the upper left corner.
But: the lower part of the display is still missing. It seems, the data for this area is simply not transferred to the display. By testing with the rotated constructor (`U8G2_R0`), the image is correctly rotated and the lower part of the image becomes visible. But the lower area of the display still shows nothing. By modifying the display starting line (cmd 0x37), the empty area can be moved across the display, but it never gets rendered.
The missing area is 24 or 25 pixels high (difficult to count without any pixel showing).
Here is my current S028H.. driver, modified for the YMC240160 display (rename to .c).
modified are the HW definitions, BIAS and Duty settings, as well as the scan areas and directions.
Did I miss to change something, that defines how much data is sent to the display?`
[u8x8_d_st7586s_s028hn118a.txt](https://github.com/olikraus/u8g2/files/4651342/u8x8_d_st7586s_s028hn118a.txt)

If you need photos of any setting/combination, let me know.

Schnulli87, 2020-05-19:
5 minutes later and a big "WTF?" is on my face:
Because I was annoyed, that I couldn't get this running on my own, I wanted to try one last idea for today, that seemed to be absolutely nonsense:
Instead of modifying the S028H.. driver, I decided to additionally adapt the ERC240 driver.
So, I changed the start/end column values, the scan direction information and copied the data translation routine from the s028h 1:1 into the ERC driver file.
I changed the constructor to use the ERC instead of the S028H and voilá: 100% of the image is shown.
Is there anywhere else in the library, where information about the display is stored? Because I cannot find a functional difference between both driver files (also checked with Notepad++ compare).

[image]

...and here is the modified ERC driver, if you want to include it into the library:
[u8x8_d_st7586s_erc240160.txt](https://github.com/olikraus/u8g2/files/4651546/u8x8_d_st7586s_erc240160.txt)

olikraus, 2020-05-22:
Wow, sieht gut aus. 

Will add this to the lib, when i find some time for this. :-/

Thanks!

en4rab, 2020-06-25:
Schnulli87 sorry i missed this when you posted it and congratulations on sorting it out.
The difference between the two drivers is to do with how the displays are wired, the st7586s is a bit odd in how it works with each column address is adressing 3 pixels, so if you set the column address to 0x00 and wrote a byte of data that would  be setting pixels 0 1 and 2.
The s028hn118a uses all 384 horizontal pixels so its connected as youd expect, however the ERC240160 only uses 240 horizontal pixels and instead of using the first 240 pixels the manufacturer left every 3rd pixel disconnected so the data to be written has to be packed differently to take this into account.

olikraus, 2020-12-26:
Implemented...

closed 2020-12-26 by olikraus