olikraus/u8g2 #1184: Stuck pixel using u8g2

wanrulez, 2020-05-08:
Hello,

I'm a bit newbie.
Im starting using this library since my oled monochrome couldn't detect i2c address.
It work for displaying, however there is stuck pixel.
Could you please help me.

[image]

olikraus, 2020-05-10:
Did you use a SSD1306 constructor? This is probably wrong, this looks like a SH1106 display.

wanrulez, 2020-05-11:
Owh thank you, but how do you identified it was SH1106 display? Sry im newbie.. Thank you very much

olikraus, 2020-05-11:
:-D Because of the pixel columns on the right. The difference between SSD1306 and SH1106 are exactly those pixel columns. There are many discussions on this problem, especially, because those displays are sold as SSD1306 but are actually SH1106, e.g. here:
https://www.eevblog.com/forum/beginners/problem-with-ssd1306-oled/

wanrulez, 2020-05-14:
Ok thnx you very much..

closed 2020-05-14 by wanrulez